Koenkk/zigbee2mqtt #622: How to change to DEV and how to update DEV?

dcboy87, 2018-11-22:
Hi,

first, thanks to all of you for creating and maintaining this great project!

I have a second Raspberry Pi (Zero W) and wanted to try DEV branch, but i'm a rookie in Linux and everything that has to do with it. 

I figured out how to clone the DEV branch with GIT, but now i wanted to update my pi to the latest commits. I followed the update instructions in the "get startet" guide, but i got a similar message like #74 

error: Your local changes to the following files would be overwritten by merge:
	configuration.yaml
Please commit your changes or stash them before you merge.
Aborting

How can i update now?

i think, it should be mentioned somewhere in the Wiki how to change to DEV and how to update this branch later.

Koenkk, 2018-11-23:
The instructions from: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version can be used for this.

WagnerBeccard, 2018-11-26:
I`m not shure at all! if I do this the right way - But this is how I got the Dev brance on my Pi 

1: First make a fork  
[image]

2: Create a new pull request on your fork. 
[image]

3: chose your new master branch first - afterwards the DEV branch from here.
[image]

4: merge the pull request 

5: Afterwards follow the install - but change the url to your fork. 
For example: 
```
# Clone zigbee2mqtt repository
sudo git clone https://github.com/dcboy87/zigbee2mqtt.git /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt
```
To update your fork repeat step 2-4 and update as in the instrucktions.

Koenkk, 2018-11-26:
Cloning the repo is as simple as: `git clone https://github.com/koenkk/zigbee2mqtt.git -b dev`