Koenkk/zigbee2mqtt #623: Repeater and connected through repeater sensors always linkquality = 0

MikeVRT, 2018-11-22:
zigbee2mqtt: 2.0.0
CC253x firmwares: both last for today

CC2530 repeater allways has linkquality = 0. 
And all devices connected throught repeater reports linkquality = 0.
Same time all devices connected through coordinator (CC2531) has non zero linkquality.

log sample (hall_thp - near coordinator, bedroom_thp - near repeater):
```
2018-11-22 21:00:27 - info: MQTT publish, topic: 'zigbee2mqtt/hall_thp', payload: '{"temperature":23.4,"linkquality":31,"pressure":987,"humidity":24.39,"battery":99,"voltage":3005}'
2018-11-22 21:01:19 - info: MQTT publish, topic: 'zigbee2mqtt/repeater_cc2530', payload: '{"state":false,"linkquality":0}'
2018-11-22 21:01:50 - info: MQTT publish, topic: 'zigbee2mqtt/bedroom_thp', payload: '{"temperature":20.15,"linkquality":0,"humidity":31.42,"pressure":987.63,"battery":99,"voltage":3015}'
```

MikeVRT, 2018-11-22:
p.s. by the way - in home assistant repeater has state "Disconnected" (in log - "state":false), but all works.

Koenkk, 2018-11-23:
@kirovilya do you know the cause of this?

kirovilya, 2018-11-23:
Earlier (about 3 months ago), the firmware for the router did not correctly report neighboring devices. But the latest firmware does it well. http://ptvo.info/wp-content/uploads/2018/09/cc2530_1.2.2a.44539_firmware.zip

1) Try to power off the cc2530 router and then turn it on again.
2) Reflash the cc2530 router and connect it again.

MikeVRT, 2018-11-23:
Firmware used last (yes, from cc2530_1.2.2a.44539_firmware.zip, router-cc2530-std.hex)

MikeVRT, 2018-11-23:
> 1. Try to power off the cc2530 router and then turn it on again.
@kirovilya, state after that true in log, linkquality still zero...

`info: MQTT publish, topic: 'zigbee2mqtt/repeater_cc2530', payload: '{"state":true,"linkquality":0}'`

MikeVRT, 2018-11-23:
And another bad info - losed devices (was on repeater) "reports" old data - on not binary sensors we can understand it by no data changes at all, on binary sensors - we will never know it - on water leak or smoke detector for example :( In my case I lose binary window/door sensor. And for understand that it losed I must open window for check, but in log I see:

`info: MQTT publish, topic: 'zigbee2mqtt/cabinet_window', payload: '{"battery":99,"voltage":3015,"linkquality":0,"contact":true}'`

dossidr, 2018-11-23:
I can confirm the same problem. The router does not connect to the coordinator. All devices behind are on link quality == 0. There were no further messages in the zigbee2mqqt window.
The version used is: 0.2.0 / router software router-cc2530-diag.hex from http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/. Only the 1st connection is established between coordinator CC2531 stick and router CC2530 Webee module.

MikeVRT, 2018-11-23:
> after ~4 hours all still working, but reports again:

I found very interesting thing in log:

```
2018-11-23 22:16:21 - info: MQTT publish, topic: 'zigbee2mqtt/repeater_cc2530', payload: '{"state":true,"linkquality":0}'
2018-11-23 22:18:44 - info: MQTT publish, topic: 'zigbee2mqtt/repeater_cc2530', payload: '{"state":false,"linkquality":5}'
```
When repeater ends report true state it starting reports linkquality non zero :D

closed 2019-01-30 by stale[bot]