Koenkk/zigbee2mqtt #624: cannot connect xiaomi mijia pir RTcq01LM

alinelena, 2018-11-22:
I cannot connect anymore a xiami mijia pir.

the sensor connects as expected to the xiaomi gateway but not to my system with cc2531.

i can connect without issues xiaomi aqara pirs, temperature sensors and door sensors from xiaomi, also ikea bulbs.

there is no trace into the log about this device (debug level is on)
but starting npm with debug seems to show something but i do not understand.

```
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +0ms
  serialport:poller Polling for "readable" +4ms
  serialport:poller received "readable" +42ms
  serialport:bindings read +42ms
  serialport:unixRead Starting read +42ms
  serialport:unixRead Finished read 32 bytes +1ms
  serialport:main binding.read finished +43ms
  cc-znp { sof: 254,
  cc-znp   len: 27,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 1030,
  cc-znp      srcaddr: 43663,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 52,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 14028448,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 7,
  cc-znp      data: <Buffer 18 13 0a 00 00 18 01> },
  cc-znp   fcs: 26,
  cc-znp   csum: 26 } +44ms
  serialport:main _read reading +2ms
  serialport:bindings read +3ms
  serialport:unixRead Starting read +2ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 1030, srcaddr: 43663, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 52, securityuse: 0, timestamp: 14028448, transseqnumber: 0, len: 7, data: <Buffer 18 13 0a 00 00 18 01> } +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +0ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +0ms
  serialport:unixRead waiting for readable because of code: EAGAIN +0ms
  serialport:poller Polling for "readable" +3ms

```

Koenkk, 2018-11-22:
Did you follow the pairing procedure from: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi ?

alinelena, 2018-11-23:
Yes I did. changed the battery too just in case.

What I find strange is that the device worked before.

Alin

Koenkk, 2018-11-23:
Sometimes it helps to

- stop zigbee2mqtt
- press the reset button on the cc2531 (the one close the usb port)
- start zigbee2mqtt
- retry pairing procedure.

alinelena, 2018-11-23:
thanks I will try it tomorrow... although attached a map of the network... things seem to be funnier... 
i see the sensor in the network... i wonder if by chance the ikea bulbs "steal it"

[image]

alinelena, 2018-11-24:
ok tried... shut down all the ikea bulbs too to avoid interference
no luck... 

however seems somehow still remembers
```
11/24/2018, 10:41:43 AM - info: unknown (0x00158d0002239c9e): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
11/24/2018, 10:41:43 AM - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002239c9e', payload: '{"occupancy":true,"linkquality":34,"battery":"100.00","voltage":3005}'
11/24/2018, 10:46:25 AM - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002239c9e', payload: '{"occupancy":true,"linkquality":34,"battery":"100.00","voltage":3005}'
```

somehow i think still remembers the device form old times...

Koenkk, 2018-11-25:
Can you remove all entries of `0x00158d0002239c9e` from the database.db and try again?

alinelena, 2018-11-25:
yes I did that... got a clean graph... but still no luck... I plan to set a clean cc2531 and try to pair it on that... this drives me banana... since I know it worked. busy week ahead but i will report back

Koenkk, 2018-11-29:
Great!

closed 2018-11-29 by Koenkk

alinelena, 2018-11-29:
not so great since that does not explain why does not pair on the other cc2531 but other devices do.
I will remove it it on this and try to repair and report