Koenkk/zigbee2mqtt #626: MQTT state topic for "permit_join"

ophilips, 2018-11-23:
I'm trying to create an easier integration for zigbee2mqtt in home assistant as "MQTT lock".  It also works without customization in homekit and is quite easy to control and setup because it is just one entry in your configuration file. Everything works so far https://gist.github.com/ophilips/8146ff802119bf6eb8ae5915851ad6d4 but sometimes it switches to the wrong state after a restart. I tried to set the retain flag and set the _state_topic_ to the same topic as _command_topic_ but it would be more reliable to get the real state.

Koenkk, 2018-11-23:
I propose that zigbee2mqtt published it's config to `zigbee2mqtt/bridge/config` with a payload like

```js
{
"log_level": "info",
"permit_join": false
```

What do you think?

ophilips, 2018-11-24:
That would be perfect I think. Then we could extract and use all values with a simple template like '{{value_json.permit_join}}'

Koenkk, 2018-12-01:
Implemented!

ophilips, 2018-12-01:
Hey Koen, I just checked it with the latest development build and found that the config gets written to 
**'zigbee2mqtt/zigbee2mqtt/bridge/config'** and not **'zigbee2mqtt/bridge/config'** as you wrote earlier.

MQTT publish: topic 'zigbee2mqtt/zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":false}'

Otherwise it was working MQTT Sensor

andreasbrett, 2018-12-01:
I can confirm. I just pulled the latest code and for me this info is posted to "zigbee/zigbee/bridge/config" (I use a shorter MQTT base topic).

https://github.com/Koenkk/zigbee2mqtt/blob/266e3f9a5b2156f8fcfd64e3292d698223566e9e/lib/extension/bridgeConfig.js#L143

Should be the issue.

Koenkk, 2018-12-01:
My bad! Should be fixed now.

andreasbrett, 2018-12-01:
It is. Works like a charm now!

ophilips, 2018-12-01:
Perfect. It's working now. I also updated my gist to control and see the the status of zigbee2mqtt in one single config entry: https://gist.github.com/ophilips/8146ff802119bf6eb8ae5915851ad6d4

Koenkk, 2018-12-01:
Great, can this be closed?

ophilips, 2018-12-02:
Ok. You can close the issue now. The only remaining thing is that home assistant template {{value_json.permit_join}} gives a True/False with capital letters instead of true/false so that publish topic and subscribe topic don‘t match. I had to use either {{value_json.permit_join|lower}} to force lowercase or send True/False.

Koenkk, 2018-12-02:
ok, thanks!

closed 2018-12-02 by Koenkk