Koenkk/zigbee2mqtt #627: Hue Bulb Color Temp

jcf6288, 2018-11-23:
Request the color_temp parameter be converted internally from 0-454 to 0-100.  My hue bulbs are expected the 0-454 range but that makes things difficult for a dimmer in openhab.  Thanks

Koenkk, 2018-11-23:
We need to somehow distinguish the mireds value (0-454) of the % value (0-100). Would it be possible to add an extra attribute to this message? Or provide the value as: {"color_temp": "80%"}?

jcf6288, 2018-11-23:
I would say to add a variable similar to RGB and HSV,  I was thinking that if a percentage is sent then it would just multiply it by 454 to  not interfere with the established format.  I don't know how to do this but it would make sense for it to recognize both a number and a percent and do the math internally before sending it on.  Is that possible?

Koenkk, 2018-11-27:
brightness and color_temp are now accepted in %, can you test this by checking out the dev branch?

jcf6288, 2018-11-27:
how do I install the dev branch?

Koenkk, 2018-11-27:
How are you running zigbee2mqtt? (via https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge)

jcf6288, 2018-11-27:
On  stretch lite via a RPI2

jcf6288, 2018-11-27:
I figured it out, what is the correct syntax for the message.  I have tried
{
  "color_temp": 0,
}
and
{
  "color_temp": 0%,
}

jcf6288, 2018-11-28:
I installed the dev branch and my Hue bulb would go online them immediately offline.

jcf6288, 2018-11-28:
For a color bulb is it possible to accept the format
{color: { rgb: "12,43,112" }}

Koenkk, 2018-11-29:
Currently 

```js
{
  "color": {
    // RGB color
    "r": 46,
    "g": 102,
    "b": 193
  }
}
```

is supported, is this ok for you?

jcf6288, 2018-11-29:
It works but requires a proxy item and rule to convert to a usable format.  For openhab it would simplify things if it could accept either the format mentioned above or and HSB value.

Koenkk, 2018-11-29:
This is supported know in the dev branch.

For domoticz and home assistant we have instructions on how to integrate with zigbee2mqtt.

Would you mind doing this for openhab? (https://github.com/Koenkk/zigbee2mqtt#integrations)

jcf6288, 2018-11-29:
I could give it a shot, I will get to work and send you a preview when it is ready.  I will test out the dev branch and let you know.

jcf6288, 2018-11-29:
I set up a new instance of Zigbee2MQTT with the dev installed and my hue bulbs will show connected then go offline.  I cycle power to them and they go online then immediately offline.  Any ideas?

jcf6288, 2018-11-30:
If I send a RGB such as {"color": {"r":25,"g":0,"b":0}} and the light is off, shouldn't it turn on to a brightness of 10?

Koenkk, 2018-11-30:
No, that is only the case for brightness (because this is supported by ZCL, not for colortemp and color). To also turn it on send: `{"state": "ON", "color": {"r":25,"g":0,"b":0}}`

jcf6288, 2018-11-30:
Understand all any ideas why I am having disconnect issues with the dev version?

Koenkk, 2018-11-30:
Can you provide the log? (maybe zigbeem2qtt crashes)

jcf6288, 2018-11-30:
I figured out the disconnect issue, it was my fault.  Color Temp works great.  
For color is this the correct format added?
{color: { rgb: "12,43,112" }}

Koenkk, 2018-11-30:
yes, that format is supported.

jcf6288, 2018-12-01:
The new format is working sort of, if I send
{"color":{ "rgb": "5,0,0" }} then {"color":{ "rgb": "255,0,0" }}
the output is the exact same, shouldn't the first be dimmer?

jcf6288, 2018-12-01:
[New Text Document.txt](https://github.com/Koenkk/zigbee2mqtt/files/2636169/New.Text.Document.txt)

jcf6288, 2018-12-01:
[RGB0,0,0.txt](https://github.com/Koenkk/zigbee2mqtt/files/2636171/RGB0.0.0.txt)

Koenkk, 2018-12-03:
I think this is due to the xy color conversion (not a bug), see: https://stackoverflow.com/questions/29834556/why-there-is-no-brown-or-grey-in-the-cie-xy-color-space

jcf6288, 2018-12-06:
After playing with the dev for a few days is it possible to instead accept
{
"color_temp_percent": XX
}
{
"brightness_percent": XX
}

This would significantly improve and simplify the integration with openhab.

Koenkk, 2018-12-06:
Supported in the dev branch.

jcf6288, 2018-12-07:
Everything is working great, the last thing would be to accept an HSB value but that isn't necessary to make it work.

Koenkk, 2018-12-07:
Why? Does openhab send both RGB and HSB values?

jcf6288, 2018-12-07:
I can send either, with other bindings if you change the color with RGB the brightness follows so 255,0,0 would be 100 percent brightness on the red channel. 2.55,0,0 would be 1 percent brightness.  The difference with Zigbee2MQTT is the user would have to send a brightness command then a color command unless there is a way to have Zigbee2MQTT interpret RGB with a Brightness value, if that makes sense.

jcf6288, 2018-12-08:
So after digging around I found this
`// sRGB luminance(Y) values
const double rY = 0.212655;
const double gY = 0.715158;
const double bY = 0.072187;

// Inverse of sRGB "gamma" function. (approx 2.2)
double inv_gam_sRGB(int ic) {
    double c = ic/255.0;
    if ( c <= 0.04045 )
        return c/12.92;
    else 
        return pow(((c+0.055)/(1.055)),2.4);
}

// sRGB "gamma" function (approx 2.2)
int gam_sRGB(double v) {
    if(v<=0.0031308)
        v *= 12.92;
    else 
        v = 1.055*pow(v,1.0/2.4)-0.055;
    return int(v*255+0.5); // This is correct in C++. Other languages may not
                           // require +0.5
}

// GRAY VALUE ("brightness")
int gray(int r, int g, int b) {
    return gam_sRGB(
            rY*inv_gam_sRGB(r) +
            gY*inv_gam_sRGB(g) +
            bY*inv_gam_sRGB(b)
    );
}`

Would it be possible to send this 
{"color":{ "rgb": "255,0,0" }}

have it calculate the brightness and send both in sequence?  brightness_percent then color?

jcf6288, 2018-12-08:
https://stackoverflow.com/questions/596216/formula-to-determine-brightness-of-rgb-color is the forum I pulled it from.

Koenkk, 2018-12-09:
What about

```js
{"color":{ "rgb_brightness": "255,0,0" }}
```

I don't want to change the current behaviour for all RGB commands.

jcf6288, 2018-12-09:
That would work just fine

What about

jcf6288, 2018-12-15:
Did you implement this?

{"color":{ "rgb_brightness": "255,0,0" }}

Also the new Hue app updates the hue bulbs so that when power returns they can be set to remain off,  is that something that can be implemented?

Koenkk, 2018-12-15:
Not yet, could you perhaps make a pull request for adding the functions to https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js

closed 2019-02-20 by stale[bot]