Koenkk/zigbee2mqtt #629: config problem after moving to mosquitto

nick9one1, 2018-11-23:
I had this working correctly with the built in home assistant MQTT, but recently moved to mosquitto. 

Since then I haven't been able to get it working. I also have a sonoff basic tasmota that works fine. 

configuration.yaml
```
# MQTT
mqtt:
discovery: true
broker: 192.168.1.125
username: "nick9one1"
password: "x254965s"
```

mosquitto config
```
{
  "logins": [
    {
      "username": "HA_user",
      "password": "password"
    }
  ],
  "anonymous": false,
  "customize": {
    "active": false,
    "folder": "mosquitto"
  },
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem"
}
```

zigbee2mqtt config
```
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": true,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://192.168.1.125",
  "serial_port": "/dev/ttyACM0",
  "mqtt_user": "HA_user",
  "mqtt_pass": "password"
}
```

mosquitto log
```
1542998445: New connection from 192.168.1.125 on port 1883.
[INFO] found HA_user on local database
1542998445: New client connected from 192.168.1.125 as 5b2090d3-4d59-4e69-8ddf-e4e8a7fb0483 (c1, k60, u'HA_user').
1542998449: New connection from 192.168.1.12 on port 1883.
[INFO] found HA_user on local database
1542998449: New client connected from 192.168.1.12 as Driveway_Lights (c1, k15, u'HA_user').
1542998454: New connection from 172.30.32.1 on port 1883.
[INFO] found HA_user on local database
1542998455: New client connected from 172.30.32.1 as mqttjs_c2e3f9db (c1, k60, u'HA_user').
```

zigbee2mqtt log
```
  zigbee2mqtt:info 2018-11-23 18:40:54 Connecting to MQTT server at mqtt://192.168.1.125
  zigbee2mqtt:info 2018-11-23 18:40:55 zigbee-shepherd ready
  zigbee2mqtt:info 2018-11-23 18:40:55 Connected to MQTT server
  zigbee2mqtt:info 2018-11-23 18:40:55 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-11-23 18:40:55 MQTT publish, topic: 'homeassistant/switch/0x00158d000251ad06/switch/config', payload: '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x00158d000251ad06/set","state_topic":"zigbee2mqtt/0x00158d000251ad06","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d000251ad06","unique_id":"0x00158d000251ad06_switch_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-11-23 18:40:55 MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d0002536e38/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d0002536e38","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d0002536e38","unique_id":"0x00158d0002536e38_contact_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-11-23 18:40:55 MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d0001e7b9fc/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d0001e7b9fc","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d0001e7b9fc","unique_id":"0x00158d0001e7b9fc_contact_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-11-23 18:40:55 MQTT publish, topic: 'zigbee2mqtt/0x00158d000251ad06', payload: '{"state":"OFF","linkquality":63}'
  zigbee2mqtt:info 2018-11-23 18:40:55 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002536e38', payload: '{"battery":31,"voltage":2965,"linkquality":31,"contact":true}'
  zigbee2mqtt:info 2018-11-23 18:40:55 MQTT publish, topic: 'zigbee2mqtt/0x00158d0001e7b9fc', payload: '{"contact":true,"linkquality":39,"battery":99,"voltage":3005}'
  zigbee2mqtt:info 2018-11-23 18:41:05 Zigbee publish to '0x00158d000251ad06', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2018-11-23 18:41:35 Zigbee publish to '0x00158d000251ad06', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms

```

Koenkk, 2018-11-23:
What switch do you have?

nick9one1, 2018-11-23:
its a Xiaomi Aquara qbkg04lm

nick9one1, 2018-11-23:
Actually, I think mosquitto might not be the issue. I've just restored a backup that I know was working, and it has the same problem. 

I also have two xiaomi door switches whose state are no longer detected by home assistant - I noticed this a few days ago but forgot to look into it.

nick9one1, 2018-11-23:
I've installed and tried the dev branch with the same config - works as it should. 
Going back to 0.2.0 causes it to stop working.

Koenkk, 2018-11-23:
This was indeed an issue in 0.2.0 which has been fixed (original: https://github.com/Koenkk/zigbee2mqtt/issues/593)

closed 2018-11-23 by Koenkk

Electronlibre2012, 2019-09-09:
Hello,

i have the same problem, i am in 0.98.4 hassio, when i used mosquitto for zigbee2mqtt, my 2 gateway doesnt respond/connect to HA anymore...

@nick9one1 please, how do you resolved it? you install the dev branch of what?

help please, i am a noob