Koenkk/zigbee2mqtt #630: CC2530 Webee Modul

dossidr, 2018-11-23:
[Zigbee_CC2530_Webee.pdf](https://github.com/Koenkk/zigbee2mqtt/files/2611744/Zigbee_CC2530_Webee.pdf)

I can confirm that the Webee module CC2530 is functional with the firmware from http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/.
It works as a coordinator and as a router.
The * .pdf attached contains the information how to connect the module and flash it. Also listed is how it needs to be connected as a coordinator with a CP2102 UART.
But there is a problem in router operation. Although it is detected by a CC2531 USB module, it does not provide any cyclic data. Likewise, the sensor behind it is no longer recognized by the coordinator. It seems already up to date to give this problem messages. Is there already a solution?

Koenkk, 2018-11-23:
What do you mean with cyclic data?

dossidr, 2018-11-25:
With cyclic I meant the timestamp which I see in the "zigbee2mqqt" log.
But I see it as solved.
Because I receive the data from the router. Unfortunately, I do not quite understand it. What does it mean if the router has the data: 

zigbee2mqtt: info 2018-11-25 14:40:28 MQTT publish, topic: 'zigbee2mqtt / 0x00124b0012022e95', payload: '{"state": true, "linkquality": 31, " description ":" 0 / 0x00124B001B6DF639 "," type ":" COORD "," rssi ": 30} '
  zigbee2mqtt: info 2018-11-25 14:40:28 MQTT publish, topic: 'zigbee2mqtt / 0x00124b0012022e95', payload: '{"state": true, "linkquality": 28, "description": "51316 / 0x00158D00022885A4", "type": "CHILD_RFD", "rssi": 35} '
  zigbee2mqtt: info 2018-11-25 14:40:28 MQTT publish, topic: 'zigbee2mqtt / 0x00124b0012022e95', payload: '{"state": true, "linkquality": 28, "description": "33911 / 0x00158D00023E3010", "type": "CHILD_RFD", "rssi": 99} '

Router = 0x00124B001B6DF639
Sensor 1 = 0x00158D00022885A4
Sensor 2 = 0x00158D00023E3010
I don't see data inside mqtt(json) string. Im not understand where can find these Data from Senor 1 and Sensor 2. What is the Point include Data from Sensor about the Way with Router?
example: "description": "51316 / 0x00158D00022885A4"
The Sensor 1 and 2 itself send these Information.
MQTT publish, topic: 'zigbee2mqtt/0x00158d00022885a4', payload: '{"temperature":18.23,"linkquality":0,"humidity":39.55,"pressure":987.47,"battery":99,"voltage":3005}'

MQTT publish, topic: 'zigbee2mqtt/0x00158d00023e3010', payload: '{"temperature":19.17,"linkquality":28,"humidity":43.02,"battery":55,"voltage":2995}'

 best regards

khzd, 2018-11-25:
Hi
This what I see with a couple of CC2531, normal behaviour
I think the CC2531 shows the enddevices linked to it and you got the information in the log file from each device independently from the route the took :)
It happen when I shut down one of the CC2531.
Cheers

Koenkk, 2018-11-25:
@ptvoinfo could your clarify what this data means?

dossidr, 2018-11-25:
Solved. 
Now the state what i see in fhem is true. Router set Child_rfd type switched between coord state and return to child_rfd. I want check permanently raw zigbee2mqtt log and check to learn understand communication. Currently im respect your development state software bridge and firmware to be use ti hardware zigbee network. Im using an not listed hardware modul in your wiki cc2530 webee. Only webee listed as cc2530 + 2591 modul. Thats the mistake im resolved my issues. Did you can added this module to the wiki and other user can find additional information about that different modules with cc2530 chip? Thanks for the feedback. Best regards.... denny

broekd, 2018-12-13:
> [Zigbee_CC2530_Webee.pdf](https://github.com/Koenkk/zigbee2mqtt/files/2611744/Zigbee_CC2530_Webee.pdf)
>
> I can confirm that the Webee module CC2530 is functional with the firmware from [...]

Hoi denny,
I have been struggeling for a little while to get the other CHDTech cc2530 to work. I have read your documentation and want to try your setup. Could you provide me the exact components. I have noted the following so far :
CC2530 Z-001 : http://www.h-2technik.com/webshop/h-2technik_webshop/shop/index.php?rt=product/product&product_id=124
USB - UART : can give me a link where you ordered this

In advance already thanks for your help and the effort you put into this. 

regards,
Dirk

dossidr, 2018-12-16:
Hi Dirk,
sorry for the latest response. Im using this module from link below. http://www.h-2technik.com/webshop/h-2technik_webshop/shop/index.php?rt=product/product&product_id=124 You can order the module in this shop. PLS connect your CC2530 to an arduino. TX / RX / 3,3 V / GND. Very important is when you are using the arduino as UART Bridge, you must connect GND / RESET PIN on the Arduino. This PIN direct connect you need to be run the arduino as UART interface. Now you connect to the Network you should see the RX or TX LED blink. Thats youre first base Connect between CC2530 and UART Interface. After that you should made your Setup in your *.yaml configuration file [/dev/ttyACM0 or /dev/serial/by-id/........). Now you can start your zigbee2mqtt bridge.
As router im using an USB Stick CC2531 with these fw https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/CC2531/bin/CC2531_router_2018_09.zip. 

Did you have check the right pinning on your module? 

best regards denny

broekd, 2018-12-16:
Denny thanks for your reply.

Iam trying to set it up somewhat differently. I currently have a CC2531-USB connected to a raspberry pi 3 running hassio. Range is however a problem.
So iam going to switch to a CC2530 - Webee

closed 2019-02-21 by stale[bot]