Koenkk/zigbee2mqtt #631: CC2531 USB device incompatibility

schmidf, 2018-11-24:
Hello,
I am using an old laptop (Sony VAIO Pro 13) for zigbee2mqtt, homeassistant, and as a backup/file server. The device has two external USB 3.0 ports, one of which is used for the CC2531 stick.

When I plug certain USB devices into the second USB port of the laptop, the Zigbee communication becomes very unresponsive. For example with a USB 3.0 Gbit NIC connected via a USB hub, switching all lights (one Philips Hue Go and three Osram Lightify) on or off can take several seconds and sometimes one of the bulbs times out without changing state. Typical errors in these cases are `Error: AF data request fails, status code: 183. APS no ack.` or `Error: AF data request fails, status code: 233. MAC no ack.` In this situation, pairing new devices also becomes almost impossible. The more devices are connected to the USB hub, the worse the CC2531 seems to work. With an additional USB 3.0 HDD (powered externally) connected, all Zigbee commands time out (incoming messages from sensors are still received).

Here is the lsusb output for the system with the three external devices marked:
```Bus 003 Device 002: ID 8087:8000 Intel Corp.
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 031: ID 1058:25a3 Western Digital Technologies, Inc. <-- USB HDD
Bus 002 Device 030: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet <-- Gbit NIC
Bus 002 Device 029: ID 0bda:0411 Realtek Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:5727 Realtek Semiconductor Corp.
Bus 001 Device 014: ID 0bda:5411 Realtek Semiconductor Corp.
Bus 001 Device 002: ID 0451:16a8 Texas Instruments, Inc. <-- CC2531
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
The weird thing is that the CC2531 is not even on the same internal USB hub as the USB HDD and NIC.

I am running zigbee2mqtt version 0.2.0 (commit #b0d3c2f) on Debian Stretch and Firmware version CC2531ZNP-Prod_20181024. The error is reproducible with firmware 20180815 and with a second CC2531 stick.

Has anyone seen similar problems and are there any ideas how to debug this?

Koenkk, 2018-11-25:
I doubt if this is an USB issue. The `183` and `233` errors come from the CC2531 stick and they mean that the zigbee devices not be reached. This could be due to range/interference (e.g. wifi).

azacan, 2019-03-19:
Just in case helps anyone, I have had a similar issue using a Rock64 board, the solution was to put some distance between the USB2 device (CC2531 stick) and the USB3 disk, I used an USB2 extender to connect the CC2531 to the board. This solved my issues. 
I found the info here. https://forum.pine64.org/showthread.php?tid=7124

Morphy99, 2020-08-03:
I've had a similar problem with a USB 3.0 SSD adaptor like [this](https://www.ebay.co.uk/itm/USB-3-0-to-Mini-PCIE-mSATA-SSD-SATA-Adapter-Card-External-Caddy-Converter-A/273997069904) plugged into a RPi4 and a CC2530 connected to the serial GPIO.
It would break my network every time I plugged it in. The issue didn't occur when I had the CC2530 mounted on the opposite side of the USB ports. After mounting in a new case the other side by the USB and extending the cable from the GPIO to the CC2530 (USB shielded cable) the interference started. I've resorted to wrapping the SSD in electrical tape (urgh) and then copper shielding tape. This seems to have done the trick although next time I'll be buying a SSD adaptor in an enclosure!
[image]