Koenkk/zigbee2mqtt #632: IKEA TRADFRI Outlet is crashing zigbee2mqtt if include_device_information is set

dr1rrb, 2018-11-24:
Using the latest zigbee2mqtt docker image (0.2.0), I was able to successfully pair my new TRADFRI outlet, but as soon as I try to turn it on or off, zigbee2mqtt crashes:

```
  zigbee2mqtt:debug 2018-11-24 21:04:31 Received MQTT message on 'zigbee2mqtt/0x000d6ffffe263428/set' with data 'ON'
  zigbee2mqtt:info 2018-11-24 21:04:31 Zigbee publish to '0x000d6ffffe263428', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
/app/node_modules/q/q.js:155
                throw e;
                ^

TypeError: this.getDeviceInfoForMqtt is not a function
    at DevicePublish.mqttPublishDeviceState (/app/lib/controller.js:470:42)
    at zigbee.publish (/app/lib/extension/devicePublish.js:136:34)
    at callback_ (/app/lib/zigbee.js:209:13)
    at /app/node_modules/q/q.js:2055:17
    at runSingle (/app/node_modules/q/q.js:137:13)
    at flush (/app/node_modules/q/q.js:125:13)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.2.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-11-24T21_04_31_235Z-debug.log
```

I search in code for `getDeviceInfoForMqtt` and found it here: https://github.com/Koenkk/zigbee2mqtt/blob/b0d3c2f96459fa78b657f0431399d5f7f06ba75a/lib/controller.js#L470

**WORKAROUND:** If you set `include_device_information: false`, it works like a perfectly!

Koenkk, 2018-11-25:
Does this issue also happen with the `latest-dev` docker image? (https://hub.docker.com/r/koenkk/zigbee2mqtt/tags/)

dr1rrb, 2018-11-26:
Unfortunately yes, the bug is still present, however the call stack is slightly different:
```
 zigbee2mqtt:debug 2018-11-26 03:34:21 Received MQTT message on 'zigbee2mqtt/0x000d6ffffe263428/set' with data 'ON'
  zigbee2mqtt:info 2018-11-26 03:34:21 Zigbee publish to '0x000d6ffffe263428', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
/app/node_modules/q/q.js:155
                throw e;
                ^

TypeError: this.getDeviceInfoForMqtt is not a function
    at DevicePublish.publishDeviceState (/app/lib/controller.js:227:42)
    at zigbee.publish (/app/lib/extension/devicePublish.js:137:34)
    at callback_ (/app/lib/zigbee.js:192:13)
    at /app/node_modules/q/q.js:2055:17
    at runSingle (/app/node_modules/q/q.js:137:13)
    at flush (/app/node_modules/q/q.js:125:13)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.2.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-11-26T03_34_21_679Z-debug.log
```

Koenkk, 2018-11-28:
Fixed in the dev branch.

closed 2018-11-28 by Koenkk

salopette, 2018-11-28:
I have Crash in the Dev branch

Nov 28 20:57:02 raspberrypi npm[581]:   zigbee2mqtt:info 2018-11-28 20:57:02 MQTT publ                                                                                         ish: topic 'zigbee2mqtt/LED-1', payload '{"state":"ON","device":{"ieeeAddr":"0x000b57f                                                                                         ffed8438d","friendlyName":"LED-1","type":"Router","nwkAddr":13039,"manufId":4476,"manu                                                                                         fName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb G                                                                                         U10 W 400lm","status":"offline"},"brightness":2}'
Nov 28 20:57:06 raspberrypi npm[581]:   zigbee2mqtt:info 2018-11-28 20:57:06 Zigbee pu                                                                                         blish to '0x000b57fffed8438d', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0}                                                                                          - null
Nov 28 20:57:06 raspberrypi npm[581]: /opt/zigbee2mqtt/node_modules/zigbee-shepherd-co                                                                                         nverters/converters/toZigbee.js:65
Nov 28 20:57:06 raspberrypi npm[581]:                 if (value.includes('%')) {
Nov 28 20:57:06 raspberrypi npm[581]:                           ^
Nov 28 20:57:06 raspberrypi npm[581]: TypeError: value.includes is not a function
Nov 28 20:57:06 raspberrypi npm[581]:     at Object.convert (/opt/zigbee2mqtt/node_mod                                                                                         ules/zigbee-shepherd-converters/converters/toZigbee.js:65:27)
Nov 28 20:57:06 raspberrypi npm[581]:     at Object.keys.forEach (/opt/zigbee2mqtt/lib                                                                                         /extension/devicePublish.js:119:41)
Nov 28 20:57:06 raspberrypi npm[581]:     at Array.forEach (<anonymous>)
Nov 28 20:57:06 raspberrypi npm[581]:     at DevicePublish.onMQTTMessage (/opt/zigbee2                                                                                         mqtt/lib/extension/devicePublish.js:111:27)
Nov 28 20:57:06 raspberrypi npm[581]:     at results.extensions.filter.map (/opt/zigbe                                                                                         e2mqtt/lib/controller.js:121:27)
Nov 28 20:57:06 raspberrypi npm[581]:     at Array.map (<anonymous>)
Nov 28 20:57:06 raspberrypi npm[581]:     at Controller.onMQTTMessage (/opt/zigbee2mqt                                                                                         t/lib/controller.js:121:14)
Nov 28 20:57:06 raspberrypi npm[581]:     at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt                                                                                         .js:81:18)
Nov 28 20:57:06 raspberrypi npm[581]:     at emitThree (events.js:136:13)
Nov 28 20:57:06 raspberrypi npm[581]:     at MqttClient.emit (events.js:217:7)
Nov 28 20:57:06 raspberrypi npm[581]: npm ERR! code ELIFECYCLE
Nov 28 20:57:06 raspberrypi npm[581]: npm ERR! errno 1
Nov 28 20:57:06 raspberrypi npm[581]: npm ERR! zigbee2mqtt@0.2.0 start: `node index.js                                                                                         `
Nov 28 20:57:06 raspberrypi npm[581]: npm ERR! Exit status 1
Nov 28 20:57:06 raspberrypi npm[581]: npm ERR!
Nov 28 20:57:06 raspberrypi npm[581]: npm ERR! Failed at the zigbee2mqtt@0.2.0 start s                                                                                         cript.
Nov 28 20:57:06 raspberrypi npm[581]: npm ERR! This is probably not a problem with npm                                                                                         . There is likely additional logging output above.
Nov 28 20:57:06 raspberrypi npm[581]: npm ERR! A complete log of this run can be found                                                                                          in:
Nov 28 20:57:06 raspberrypi systemd[1]: zigbee2mqtt.service: Main process exited, code                                                                                         =exited, status=1/FAILURE
Nov 28 20:57:06 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered failed state                                                                                         .
Nov 28 20:57:06 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-                                                                                         code'.
Nov 28 20:57:07 raspberrypi systemd[1]: zigbee2mqtt.service: Service hold-off time ove                                                                                         r, scheduling restart.
Nov 28 20:57:07 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Nov 28 20:57:07 raspberrypi systemd[1]: Started zigbee2mqtt.
Nov 28 20:57:08 raspberrypi npm[821]: > zigbee2mqtt@0.2.0 start /opt/zigbee2mqtt
Nov 28 20:57:08 raspberrypi npm[821]: > node index.js
Nov 28 20:57:09 raspberrypi npm[821]:   zigbee2mqtt:info 2018-11-28 20:57:09 Logging t                                                                                         o directory: 'data/log/2018-11-28.20-57-09'
[... cut ...]
Nov 28 20:57:11 raspberrypi npm[821]:   zigbee2mqtt:info 2018-11-28 20:57:11 zigbee-sh                                                                                         epherd ready
Nov 28 20:57:11 raspberrypi npm[821]:   zigbee2mqtt:info 2018-11-28 20:57:11 Connected                                                                                          to MQTT server
Nov 28 20:57:11 raspberrypi npm[821]:   zigbee2mqtt:info 2018-11-28 20:57:11 MQTT publ                                                                                         ish: topic 'zigbee2mqtt/bridge/state', payload 'online'

JLFN, 2018-12-06:
> include_device_information: false

Where did you put that information?