Koenkk/zigbee2mqtt #633: FLS-PP

csu-csu, 2018-11-25:
HI!

I would like to config a dresden-elektronik FLS-PP controller. 
I made a costum devices.js on my hass.io but It seems there is a special character within the modelID, so it's not working. 
[image]

Any idea how can I handle it?

khzd, 2018-11-25:
Hi
Did you try to delete this special caractère in the DB ?
I had the same issue for a Xiaomi device

Koenkk, 2018-11-25:
Can you post your database.db?

csu-csu, 2018-11-25:
First of all I was able to make it work!
@khzd, You gave me a good idea! I saw what is the special character  in the DB and I write it to the devices.js. And it's work! Otherways it was: "\u0000"

@Koenkk here is my databese if you want to use.

thanks both of you...
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/2613488/database.zip)

khzd, 2018-11-26:
EZ :)

closed 2018-11-26 by csu-csu