Koenkk/zigbee2mqtt #636: Unable to pair RTCGQ01LM

tb205gti, 2018-11-27:
zigbee2mqtt 0.1.6
npm v6.4.1
node v8.12.0

I'm completely unable to pair a new Xiaomi motion sensor, RTCGQ01LM. I do get the incoming device message, and I can see in the log that it tries to connect, but every single time I get a timeout error.

I can perfectly pair other xiaomi sensors (motion/ambience, door and temp/humid), but not this single RTCGQ01LM. I have almost tried everything now, and it will just not work. 

What do I need to do to make it work?

Koenkk, 2018-11-27:
Have you followed the pairing procedure from: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi ?

tb205gti, 2018-11-28:
Yes, tried everything :/

Koenkk, 2018-11-28:
Can you try replacing the battery?

tb205gti, 2018-11-28:
Also tried that :/

Koenkk, 2018-11-28:
Can you remove all the entries of the device from  `database.db` and try again? (stop zigbee2mqtt before doing)

tb205gti, 2018-11-28:
if I make a backup of the database.db and the configurations.yaml - will I then be able to restore to the current setup if needed?

MaximKraev, 2019-01-28:
Anyone got it working? I am having the same issue

akuky, 2019-02-18:
I successfully installed  zigbee2mqtt ( bridge?)  on RPI3B+ and have Hassio and Mosquitto broker on Ubuntu 18.04 2 Server (docker) } 
Have the "Xiaomi Smart Human Body Sensor" SKU YTC4016CN , paired with zigbee2mqtt as "RTCGQ01LM"
From the log: MQTT publish: topic 'homeassistant/sensor/0x00158d0002717c27/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/MotionSensor","json_attributes_topic":"zigbee2mqtt/MotionSensor","name":"MotionSensor_battery","unique_id":"0x00158d0002717c27_battery_zigbee2mqtt","device" {"identifiers":"zigbee2mqtt_0x00158d0002717c27","name":"MotionSensor","sw_version":"Zigbee2mqtt 1.1.1","model":"MiJia human body movement sensor (RTCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}' 

No problems on Motion detection but no Values or log entries(?) for the battery
Any suggestions? ( The link above is 404)
[image]
Thanks

akuky, 2019-02-18:
Yay - got battery voltage in the log - now i just need to figure out how to display it ;)
I unplugged the keyboard for RPI and restarted "npm start" - weird

closed 2019-04-26 by stale[bot]