Koenkk/zigbee2mqtt #638: Unable to pair... anything.

jonathanmusto, 2018-11-28:
So, I'm setting zigbee2mqtt up with the ultimate aim of replacing Smartthings, which I've fallen out of love with of late.

I bought all the kit from Aliexpress and flashed the stick with the latest firmware without any problems. I'm running zigbee2mqtt inside a docker container (along with Eclipse MQTT and HASS).  All services are talking to each other fine, everything starts up fine, I get the online message from zigbee2mqtt in mqtt, and the stick has a green light.  But, I'm unable to pair anything I've tried so far, which has been a Xaoimi Door/Window Sensor (1 MiJia and 1 Aqara).

With the MiJa, i held the reset button for 4-5s until it flashed, then kept on hitting the reset every second or so. Nothing shows up in the log as pairing or trying to pair.  I also tried changing the battery.

With the Aqara, same thing. Hold the button for 4-5s until it blinks 3 times, then wait.  But, nothing showing up in the logs.  I tried both devices for around 1h last night, including holding them centimeters away from the USB stick.

Is there any additional logging I can switch on or check? I enabled debug mode, but that didn't seem to have much effect:

https://pastebin.com/3X2k3ddp

jonathanmusto, 2018-11-28:
Other devices I have are Tradfri Bulbs then some Smartthings sensors, the ones that come with the kit (outlet, motion, presence, contact).  I'll try a Tradfri bulb when I get home tonight.

Koenkk, 2018-11-28:
First make sure your old network is offline (power off all zigbee devices), otherwise device rejoin the old network.

jonathanmusto, 2018-11-28:
> First make sure your old network is offline (power off all zigbee devices), otherwise devices rejoin the old network.

Thanks, so the Mijia device was previously connected to Smartthings, but the Aqara sensor was a new device straight out of the box.  That raises an interesting question though, since my Smartthings network is still online with circa 20 zigbee devices, could that be interfering with the pairing process somehow? I'll try switching off ST hub.

Koenkk, 2018-11-28:
Yes it interferes, you should turn of the hub + all devices (e.g. bulbs).

jonathanmusto, 2018-12-03:
> Yes it interferes, you should turn of the hub + all devices (e.g. bulbs).

Ok, I turned off my ST hub, removed the batteries. switched off all my bulbs / sockets. I still can't get anything to pair, anything else I can do to troubleshoot?

Koenkk, 2018-12-03:
Do you see any output when running with `DEBUG=* npm start` and starting the pairing procedure?

jonathanmusto, 2018-12-03:
> Do you see any output when running with `DEBUG=* npm start` and starting the pairing procedure?

I'm running in docker-compose, is that the same as adding log_level: debug to the config?

jonathanmusto, 2018-12-03:
2018-12-3 15:01:08 - info: Currently 0 devices are joined:
2018-12-3 15:01:08 - warn: `permit_join` set to  `true` in configuration.yaml.
2018-12-3 15:01:08 - warn: Allowing new devices to join.
2018-12-3 15:01:08 - warn: Set `permit_join` to `false` once you joined all devices.
2018-12-3 15:01:08 - info: Zigbee: allowing new devices to join.
2018-12-3 15:01:08 - info: Connecting to MQTT server at mqtt://mqtt:1883
2018-12-3 15:01:08 - info: zigbee-shepherd ready
2018-12-3 15:01:08 - info: Connected to MQTT server
2018-12-3 15:01:08 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-12-3 15:01:08 - debug: Soft reset timeout disabled
2018-12-3 15:06:08 - debug: Saving state to file /app/data/state.json

jonathanmusto, 2018-12-03:
Maybe I got a faulty CC2531?

jonathanmusto, 2018-12-03:
Wait, I just noticed the following before the Currently 0 devices are joined line:

  zigbee2mqtt:debug 2018-12-3 15:01:08 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ed23d2","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1543849268,"joinTimeLeft":0}

joinTimeLeft: 0?

Does that mean nothing can join?

jonathanmusto, 2018-12-03:
Ok, so I got it working by.... plugging the sniffer into a different USB port!

Koenkk, 2018-12-05:
Ok, great!

closed 2018-12-05 by Koenkk

bonjour81, 2019-12-05:
Hi! just wanted to tell got same issue (cannot pair anything) and same work around: moving the CC2531 to a different USB port solve the issue... (weird)

doz10us, 2022-03-22:
Same issue, same solution worked.

aviogit, 2023-01-23:
Same for me. This bug is amazing, a few days ago, my Xiaomi Aqara Cube ceased to send traffic, so I changed the battery (that was still ok btw), and tried to pair it again. Nothing. After a few days of debugging the cube (just add `-e DEBUG=zigbee-herdsman*`   in your docker command line) and a few Aqara buttons that behaved the same way (debugging nothing mostly, because of no output, and all other Aqara devices worked ok as usual), I found this thread. Then I spent another couple of hours trying to pair the cube (that was complaining about routing problems in the zigbee protocol), changing the USB port after every attempt because it was stopping to communicate altogether, and eventually I was successful.