Koenkk/zigbee2mqtt #640: Aqara Wall Switch (Double Rocker) not working in branch DEV

andreasbrett, 2018-11-28:
After switching to DEV branch for some other feature my Aqara Wall Switches now don't fire MQTT publishes anymore when pressing them.

Running zigbee2mqtt in debug mode I see that the press is registered but not processed further. Looks like something broke from master -> dev.

Debug is at https://pastebin.com/DDbq4aH7

Koenkk, 2018-11-29:
This has been fixed in the dev branch, can you confirm?

andreasbrett, 2018-11-29:
There hasn't been any new commit since my testing 8 hours ago. I already was on the dev branch for zigbee2mqtt.

andreasbrett, 2018-11-29:
What I reverted was:

https://github.com/Koenkk/zigbee-shepherd-converters/blob/4cc2abbf0079e8b40e8bd103d787f296f3e0e4af/converters/fromZigbee.js#L503-L505

https://github.com/Koenkk/zigbee-shepherd-converters/blob/4cc2abbf0079e8b40e8bd103d787f296f3e0e4af/devices.js#L128-L130

Koenkk, 2018-11-29:
Can you provide the whole log?

andreasbrett, 2018-12-01:
Problem is: when I switch to the dev branch for zigbee2mqtt, clean all node modules and run npm install I won't get the dev branch for zigbee-shepherd (at least looked like to me). So what I did afterwards is to remove the zigbee-shepherd folder in node_modules and clone it directly from github (and of course switching to dev).

andreasbrett, 2018-12-01:
Yes, now it works. zigbee-shepherd-converts latest version gets pulled during npm install. My wall switches work again with zigbee2mqtt on dev.

closed 2018-12-01 by andreasbrett