Koenkk/zigbee2mqtt #641: motion sensor not sending MQTT

trekker25, 2018-11-29:
Just starting with zigbee2mqtt.

2 joined devices:
- motion sensor Aqara
- Hue dimmer

the dimmer is sending nicely updates to MQTT, but the motion sensor notices updates but does not send them to mqtt?

this is in the log:

2018-11-29 16:08:43 - info: MQTT publish, topic: 'zigbee2mqtt/0x0017880104ec5bb3', payload: '{"battery":100,"linkquality":68,"action":"down-press","brightness":50}'
2018-11-29 16:08:44 - debug: Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'RWL021' (0x0017880104ec5bb3)
2018-11-29 16:08:44 - info: MQTT publish, topic: 'zigbee2mqtt/0x0017880104ec5bb3', payload: '{"battery":100,"linkquality":76,"action":"on"}'
2018-11-29 16:08:44 - debug: Received zigbee message of type 'cmdOffWithEffect' with data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}' of device 'RWL021' (0x0017880104ec5bb3)
2018-11-29 16:08:44 - info: MQTT publish, topic: 'zigbee2mqtt/0x0017880104ec5bb3', payload: '{"battery":100,"linkquality":92,"action":"off"}'
2018-11-29 16:08:49 - debug: Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":32}}' of device 'undefined' (0x00158d000272b736)
2018-11-29 16:08:49 - debug: Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'undefined' (0x00158d000272b736)
2018-11-29 16:08:56 - debug: Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":32}}' of device 'undefined' (0x00158d000272b736)
2018-11-29 16:08:56 - debug: Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'undefined' (0x00158d000272b736)

Why is this data not passed through?

trekker25, 2018-11-29:
Was my first device i added and it blinked correctly. What could have gone wrong?

Seems the device is reporting correctly but something in Zigbee2mqtt can’t handle it correctly?

Koenkk, 2018-11-30:
When the pairing procedures it not completed the device has a modelId of `undefined`. A modelId which zigbee2mqtt doesn't know.

palitu, 2018-12-07:
I came here for a similar issue.

I restart the zigbee2mqtt container nightly, and after that happens, HASS will not register the motion sensors updates.  If i restart HASS, it starts registering the changes again.  

This behaviour only started occurring after i updated to HASS v83.

I can see the occupancy changes in z2m as expected, but i cannot see the changes in HASS.  I haven't inspected the MQTT bus yet though.

Koenkk, 2018-12-07:
@palitu did you update from zigbee2mqtt 0.2 to the dev? This version has a breaking change which adds `_occupancy` behind the sensor name.

palitu, 2018-12-09:
@Koenkk  i dont think so, it is a lastest docker release image.  All of the MQTT messages are going as expected, when using mosquitto_sub, but HA is not updating.

`mosquitto_pub -h 192.168.1.20 -t zigbee2mqtt/motion_toilet -m '{"occupancy":"On","linkquality":92,"battery":99,"voltage":3025}'`

should be the same as what Z2M is sending, but i get no updates.  Do you know how i check mqtt discovered devices in HA?  or how to debug them?  Everything else seems to work

palitu, 2018-12-09:
Ok this is weird - When i restart Z2M, i get the following in HASS:
`2018-12-09 14:47:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to zigbee2mqtt/Motion_Toilet`

However, teh next messages that come through mosquitto_sub is:
`zigbee2mqtt/Motion_Toilet {"occupancy":false,"linkquality":78,"battery":99,"voltage":3025}`

and nothing comes through the HASS logs.  I am receiving other messages from xiaomi plugs and other things.

palitu, 2018-12-09:
> That's really strange, I would recommend opening an issue at home
> assistant (as I don't see what wrong here either).

Yeah, I have started A post on the hass forums, as everything is working on
your side!

It is an odd issue

palitu, 2018-12-14:
This was a bug in the way that HA restored devices that were offline.
It is fixed in the 84.1.

thanks :)

commit 7507ad6d (2019-09-29): Remove caching for `to_side` and `from_side`. (#2048)

Stop caching `to_side` and `from_side` which only have meaning during the `flip90` action of the MFKZQ01LM Smart Cube.
Fixes #641
closed 2019-02-19 by stale[bot]