Koenkk/zigbee2mqtt #642: RGBgenie Support Info

mcsSolutions, 2018-11-29:
RGBgenie is a handheld 3 zone remote and dimmer.  I was able to pair it with zigbee2mqtt and was able to update devices.js and toZigbee.js with the following to get MQTT traffic on battery status.

    generic_battery_percent: {
        cid: 'genPowerCfg',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            return {battery: precisionRound(msg.data.data['batteryPercentageRemaining'], 2)};
        },
    },

    // RGBgenie
    {
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        vendor: 'RGBgenie',
        description: '3 Zone remote and dimmer',
        supports: 'onoff dim scene control',
        fromZigbee: [fz.generic_battery_percent],
        toZigbee: [],
    },
Where I am stopped is with the primary function of the 12 buttons on the remote.  When I pressed the 10 individual buttons the zigbee traffic was identical except timing and there was no WARN message about something without converter.  I would expect each button to produce a unique payload.

When I pressed the All Off button then a very long stream of debug resulted, but no WARN message again.  The debug and related information was pasted at https://pastebin.com/S8mbttRr

Don't know what my next step should be .  The install instructions contain the following info:

RGBgenie is Zigbee 3.0 compliant.  It is an end point device that can perform all available operations within the standard.  It is able to bind up to 30 lighting devices for on/off and dimming control.  It is compatible with universal ZigBee gateway products and universal Zigbee lighting devices. ...
The remote can pair with lighting devices that support Touch-Link commissioning.

My objective is to get MQTT message for each button push.  I do not have a desire to link the remote with other ZigBee devices.

mcsSolutions, 2018-12-01:
Summary
----------
Progress with recognizing a button push for CmdOff and CmdOn which have existing converters.
Same CmdOff/CmdOn messages produced with "All Off" & "1 Off" / "All On" & ""1 On" buttons .
Other 4 buttons produce no response when pressed.
At Zigbee2MQTT startup a 5 retry configuration sequence with the remote eventually errors-out.

Detail
--------

The On and Off buttons on the remote now produce messages that have existing converters so I an update devices.js so the battery and these two buttons will produce MQTT messages.

All Off or 1 Off button
  zigbee2mqtt:debug 2018-12-1 11:39:02 Received zigbee message of type 'cmdOff' with data '{"cid":"genOnOff","data":{}}' of device 'ZGRC-KEY-013' (0x000d6f000fc74fda)
  zigbee2mqtt:warn 2018-12-1 11:39:02 No converter available for 'ZGRC-KEY-013' with cid 'genOnOff', type 'cmdOff' and data '{"cid":"genOnOff","data":{}}'

All On or 1 On button
  zigbee2mqtt:debug 2018-12-1 11:15:30 Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'ZGRC-KEY-013' (0x000d6f000fc74fda)
  zigbee2mqtt:warn 2018-12-1 11:15:30 No converter available for 'ZGRC-KEY-013' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'

I am also able to use the button labeled "1" to produce the same result for the On and Off positions.  The other 4 buttons produce no messages when pressed in either the On or Off position.

Each start of zigbee2MQTT results in a messages sequence repeated 5 times.  It appears the remote is trying to configure.  At this time the only Zigbee device that is powered in the remote.  The snippit of the ending of the startup sequence is

  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +43ms
  zigbee-shepherd:request REQ --> ZDO:bindReq +9s
  cc-znp:SREQ --> ZDO:bindReq, { dstaddr: 31483, srcaddr: '0x000d6f000fc74fda', srcendpoint: 1, clusterid: 6, dstaddrmode: 3, addr_short_long: '0x00124b0018e1a280', dstendpoint: 1 } +11ms
  serialport:main _write 28 bytes of data +10s
  serialport:bindings write 28 bytes +10s
  serialport:unixWrite Starting write 28 bytes offset 0 bytesToWrite 28 +10s
  serialport:unixWrite write returned null 28 +1ms
  serialport:unixWrite wrote 28 bytes +0ms
  serialport:unixWrite Finished writing 28 bytes +0ms
  serialport:main binding.write write finished +4ms
  serialport:poller received "readable" +10s
  serialport:bindings read +697ms
  serialport:unixRead Starting read +10s
  serialport:unixRead Finished read 6 bytes +4ms
  serialport:main binding.read finished +703ms
  cc-znp { sof: 254,
  cc-znp   len: 1,
  cc-znp   type: 'SRSP',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'bindReq',
  cc-znp   payload: { status: 0 },
  cc-znp   fcs: 69,
  cc-znp   csum: 69 } +714ms
  serialport:main _read reading +7ms
  serialport:bindings read +17ms
  serialport:unixRead Starting read +8ms
  cc-znp:SRSP <-- ZDO:bindReq, { status: 0 } +4ms
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +25ms
  zigbee-shepherd:request REQ --> ZDO:bindReq +9s
  cc-znp:SREQ --> ZDO:bindReq, { dstaddr: 31483, srcaddr: '0x000d6f000fc74fda', srcendpoint: 1, clusterid: 6, dstaddrmode: 3, addr_short_long: '0x00124b0018e1a280', dstendpoint: 1 } +2ms
  serialport:main _write 28 bytes of data +9s
  serialport:bindings write 28 bytes +9s
  serialport:unixWrite Starting write 28 bytes offset 0 bytesToWrite 28 +10s
  serialport:unixWrite write returned null 28 +1ms
  serialport:unixWrite wrote 28 bytes +0ms
  serialport:unixWrite Finished writing 28 bytes +1ms
  serialport:main binding.write write finished +3ms
  serialport:poller received "readable" +10s
  serialport:bindings read +938ms
  serialport:unixRead Starting read +10s
  serialport:unixRead Finished read 6 bytes +5ms
  serialport:main binding.read finished +951ms
  cc-znp { sof: 254,
  cc-znp   len: 1,
  cc-znp   type: 'SRSP',
  cc-znp   subsys: 'ZDO',
  cc-znp   cmd: 'bindReq',
  cc-znp   payload: { status: 0 },
  cc-znp   fcs: 69,
  cc-znp   csum: 69 } +959ms
  serialport:main _read reading +5ms
  serialport:bindings read +22ms
  serialport:unixRead Starting read +11ms
  cc-znp:SRSP <-- ZDO:bindReq, { status: 0 } +4ms
  serialport:unixRead waiting for readable because of code: EAGAIN +2ms
  serialport:poller Polling for "readable" +27ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.bind('genOnOff', coordinator, cb)' with result 'Error: Timed out after 10000 ms' +0ms
  zigbee2mqtt:error 2018-12-1 11:18:50 Failed to configure 0x000d6f000fc74fda 0x000d6f000fc74fda

mcsSolutions, 2018-12-03:
Full debug at https://pastebin.com/99r4Xn7m for the first case described below.

Pressed S1 button immediately before starting zigbee2mqtt
After initial startup messages a few second pause occurred.  
Pressed S1 button again
Error "cannot read property ..." occurred per the following. No subsequent response to any button except the All On, All Off, 1 On and 1 Off.
Repeated process but waited for the completion of the 5 cycles and original Error timeout occurred.  Same response behavior with only 4 of the button states reported.

  serialport:main binding.read finished +2s
  cc-znp { sof: 254,
  cc-znp   len: 3,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'dataConfirm',
  cc-znp   payload: { status: 0, endpoint: 1, transid: 1 },
  cc-znp   fcs: 199,
  cc-znp   csum: 199 } +2s
  serialport:main _read reading +7ms
  serialport:bindings read +8ms
  serialport:unixRead Starting read +8ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 1 } +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +4ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1 +2ms
  serialport:unixRead Finished read 30 bytes +8ms
  serialport:main binding.read finished +10ms
  cc-znp { sof: 254,
  cc-znp   len: 25,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 6,
  cc-znp      srcaddr: 31483,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 63,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 16668798,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 5,
  cc-znp      data: <Buffer 08 01 0b 06 82> },
  cc-znp   fcs: 163,
  cc-znp   csum: 163 } +7ms
  serialport:main _read reading +9ms
  serialport:bindings read +19ms
  serialport:unixRead Starting read +11ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 31483, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 63, securityuse: 0, timestamp: 16668798, transseqnumber: 0, len: 5, data: <Buffer 08 01 0b 06 82> } +6ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +17ms
  serialport:unixRead waiting for readable because of code: EAGAIN +21ms
  serialport:poller Polling for "readable" +51ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +3ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.report('genOnOff', 'onOff', 0, 1000, 0, cb)' with result 'TypeError: Cannot read property 'status' of undefined' +4s
  zigbee2mqtt:error 2018-12-3 09:56:36 Failed to configure 0x000d6f000fc74fda 0x000d6f000fc74fda

Koenkk, 2018-12-05:
Can you try with

```js
{
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        vendor: 'RGBgenie',
        description: '3 Zone remote and dimmer',
        supports: 'onoff dim scene control',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);

            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.bind('genLevelCtrl', coordinator, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

mcsSolutions, 2018-12-05:
When pressing 3 ON key shortly after startup while it likely was in its 2nd polling interval before timeout a continuous stream pasted at https://pastebin.com/8C056cpn appeared.  Ctl-C to stop it.

When pressing each key after the initial 5 cycle timeout resulted in a response to each key.  End of log pasted at https://pastebin.com/6fk8zkcu

While I believe it is the same as prior I also pasted the initial startup until the 10000 ms message was received.  No key presses.  It is at https://pastebin.com/wcaRjzDV

Koenkk, 2018-12-07:
You say that the on buttons are working now, but do you see any `No converter available for ` produced by these buttons? I only see one for `genPowerCfg`.

mcsSolutions, 2018-12-07:
The All on key and 1 on key produce the same no-converter message such as the below from startup

  zigbee2mqtt:info 2018-12-7 14:30:24 zigbee-shepherd ready
  zigbee2mqtt:info 2018-12-7 14:30:24 Connected to MQTT server
  zigbee2mqtt:info 2018-12-7 14:30:24 MQTT publish, topic: 'zigbeeWA/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-12-7 14:30:24 MQTT publish, topic: 'zigbeeWA/0x000d6f000fc74fda', payload: '{"battery":82,"linkquality":81}'
{ groupid: 0,
  clusterid: 6,
  srcaddr: 31483,
  srcendpoint: 1,
  dstendpoint: 1,
  wasbroadcast: 0,
  linkquality: 44,
  securityuse: 0,
  timestamp: 11814241,
  transseqnumber: 0,
  len: 3,
  data: { '0': 1, '1': 84, '2': 1 },
  zclMsg:
   { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 0 },
     manufCode: 0,
     seqNum: 84,
     cmdId: 'on',
     payload: {} } }
cmd { groupid: 0,
  clusterid: 6,
  srcaddr: 31483,
  srcendpoint: 1,
  dstendpoint: 1,
  wasbroadcast: 0,
  linkquality: 44,
  securityuse: 0,
  timestamp: 11814241,
  transseqnumber: 0,
  len: 3,
  data: { '0': 1, '1': 84, '2': 1 },
  zclMsg:
   { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 0 },
     manufCode: 0,
     seqNum: 84,
     cmdId: 'on',
     payload: {} } }
  zigbee2mqtt:warn 2018-12-7 14:31:13 No converter available for 'ZGRC-KEY-013' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'
  zigbee2mqtt:warn 2018-12-7 14:31:13 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:error 2018-12-7 14:31:14 Failed to configure 0x000d6f000fc74fda 0x000d6f000fc74fda

The All Off key and 1 off key produces similar

2018-12-7 14:28:49 No converter available for 'ZGRC-KEY-013' with cid 'genOnOff', type 'cmdOff' and data '{"cid":"genOnOff","data":{}}'

2018-12-7 14:29:50 No converter available for 'ZGRC-KEY-013' with cid 'genOnOff', type 'cmdOn' and data '{"cid":"genOnOff","data":{}}'

The other keys usually produce no response, but I did capture in the  prior post paste's the 3 ON key during startup where it looks as if the remote is looking for devices to which it can pair to control.

Koenkk, 2018-12-09:
Ok, can you try if more buttons report `No converter available` when running with:
```js
{
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        vendor: 'RGBgenie',
        description: '3 Zone remote and dimmer',
        supports: 'onoff dim scene control',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep1 = shepherd.find(ieeeAddr, 1);
            const ep2 = shepherd.find(ieeeAddr, 2);
            const ep3 = shepherd.find(ieeeAddr, 3);
            const ep4 = shepherd.find(ieeeAddr, 4);

            const actions = [
                (cb) => ep1.bind('genOnOff', coordinator, cb),
                (cb) => ep1.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep2.bind('genOnOff', coordinator, cb),
                (cb) => ep2.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep3.bind('genOnOff', coordinator, cb),
                (cb) => ep3.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep4.bind('genOnOff', coordinator, cb),
                (cb) => ep4.bind('genLevelCtrl', coordinator, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

mcsSolutions, 2018-12-09:
Run time error with the latest update to devices.js.  

 root@MQTT:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@0.2.0 start /opt/zigbee2mqtt
node index.js

                Zigbee2mqtt requires node version >=8.11 10, you are running v8.14.0!

  zigbee2mqtt:info 2018-12-9 12:31:38 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-12-09.12-31-38'
  zigbee2mqtt:info 2018-12-9 12:31:39 Starting zigbee2mqtt version 0.2.0 (commit #b0d3c2f)
  zigbee2mqtt:info 2018-12-9 12:31:39 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-12-9 12:31:41 zigbee-shepherd started
  zigbee2mqtt:info 2018-12-9 12:31:41 Coordinator firmware version: '20180815'
  zigbee2mqtt:info 2018-12-9 12:31:41 Currently 1 devices are joined:
  zigbee2mqtt:info 2018-12-9 12:31:41 0x000d6f000fc74fda (0x000d6f000fc74fda): ZGRC-KEY-013 - RGBgenie 3 Zone remote and dimmer (EndDevice)
/opt/zigbee2mqtt/node_modules/q/q.js:155
                throw e;
                ^

ReferenceError: device is not defined
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/devices.js:1520:21)
    at Controller.configureDevice (/opt/zigbee2mqtt/lib/controller.js:131:29)
    at devices.forEach (/opt/zigbee2mqtt/lib/controller.js:54:30)
    at Array.forEach (<anonymous>)
    at zigbee.start (/opt/zigbee2mqtt/lib/controller.js:52:29)
    at shepherd.start (/opt/zigbee2mqtt/lib/zigbee.js:59:17)
    at /opt/zigbee2mqtt/node_modules/q/q.js:2055:17
    at runSingle (/opt/zigbee2mqtt/node_modules/q/q.js:137:13)
    at flush (/opt/zigbee2mqtt/node_modules/q/q.js:125:13)
    at _combinedTickCallback (internal/process/next_tick.js:132:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.2.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-12-09T20_31_41_456Z-debug.log
root@MQTT:/opt/zigbee2mqtt#

Koenkk, 2018-12-10:
Sorry, made a mistake:

```js
{
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        vendor: 'RGBgenie',
        description: '3 Zone remote and dimmer',
        supports: 'onoff dim scene control',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep1 = shepherd.find(ieeeAddr, 1);
            const ep2 = shepherd.find(ieeeAddr, 2);
            const ep3 = shepherd.find(ieeeAddr, 3);
            const ep4 = shepherd.find(ieeeAddr, 4);

            const actions = [
                (cb) => ep1.bind('genOnOff', coordinator, cb),
                (cb) => ep1.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep2.bind('genOnOff', coordinator, cb),
                (cb) => ep2.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep3.bind('genOnOff', coordinator, cb),
                (cb) => ep3.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep4.bind('genOnOff', coordinator, cb),
                (cb) => ep4.bind('genLevelCtrl', coordinator, cb),
            ];

            execute(ep1, actions, callback);
        },
    },
```

mcsSolutions, 2018-12-10:
Data detail at https://pastebin.com/mphfPrHZ

Progress made with the 4 buttons that control on/off/dim.  One of these buttons need to be pressed shortly after zigbee2mqtt is started.  All 4 will then will be recognized subsequently.  No progress with the two scene button S1 and S2.  No matter when they are pressed there is no feedback from zigbee2mqtt.

The on/off/dim buttons report cid:genOnOff or cid:genLevelCtrl depending upon short vs. long press.  
The screndpoint: parameter being 1,2,3, or 4 depending upon button
Type:  being cmdOff or cmdOn for the cid:genOnOff case for on/off control
Movemode:  being 0 or 1 for up/down for the cid:genLevelCtrl case

The long press yields a single feedback.  It seems that one should be able to hold the dim control down and repeated messages produced to cause the end point to ramp up and down rather than expecting the remote user to pulse the button, but not certain of the remote design.

The All On and All off buttons use cid:genOnOff control, but the screndpoint: parameter is fixed at a  4.  The contents of 4-off and All-off messages are identical except the increment of the sequence number in the 6 data bytes.

Remaining issues:
1. Do not know how to distinguish the "All" from the "4" button for on/off control.
2. Scene control buttons S1 and S2 do not report
3. Dim control produces a single message no matter how long the button pushed

Koenkk, 2018-12-11:
Let's fix the scene buttons next,

```js
{
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        vendor: 'RGBgenie',
        description: '3 Zone remote and dimmer',
        supports: 'onoff dim scene control',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep1 = shepherd.find(ieeeAddr, 1);
            const ep2 = shepherd.find(ieeeAddr, 2);
            const ep3 = shepherd.find(ieeeAddr, 3);
            const ep4 = shepherd.find(ieeeAddr, 4);

            const actions = [
                (cb) => ep1.bind('genOnOff', coordinator, cb),
                (cb) => ep1.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep2.bind('genOnOff', coordinator, cb),
                (cb) => ep2.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep3.bind('genOnOff', coordinator, cb),
                (cb) => ep3.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep4.bind('genOnOff', coordinator, cb),
                (cb) => ep4.bind('genLevelCtrl', coordinator, cb),
                (cb) => ep1.bind('genScenes', coordinator, cb),
            ];

            execute(ep1, actions, callback);
        },
    },
```

If the scenes buttons don't work after this, please try different `ep` for `(cb) => ep1.bind('genScenes', coordinator, cb),`. Possible eps are `ep1`, `ep2`, `ep3`, `ep4`.

mcsSolutions, 2018-12-11:
No response to either S1 or S2 buttons.  Tried changing the following line to ep2, ep3 and ep4 with restart in each case.
                (cb) => ep1.bind('genScenes', coordinator, cb),

Also tried changing both the above line and the below line so the ep# were the same for both.       
            execute(ep1, actions, callback);

The other buttons continued to function as the prior iteration.

Koenkk, 2018-12-12:
Can you try the same but then with `(cb) => ep1.bind('lightingColorCtrl', coordinator, cb),`

mcsSolutions, 2018-12-12:
Tried each of the eight bind/execute for ep# with no response from either S1 or S2 buttons.

Koenkk, 2018-12-13:
@mcsSolutions you would need to sniff the zigbee packets in order to find out what these buttons transmit (https://blog.jimmo.id.au/zigbee-packet-capture-724cf9346b47)

mcsSolutions, 2019-01-11:
I setup a sniffer and captured traffic on the RGBGenie.  I started with removing database.db, state.json and editing configuration.yaml to remove all devices.  I reset the RGBGenie to factory.   I started zigbee2mqtt and whsniff/wireshark. I then tried to pair RGBGenie per pairing instructions in the manual. 

While much traffic was collected the pairing process never resulted in any change to .db or .yaml files and subsequent button pushes on RGBGenie saw no sniffer traffic.  I repeated the process with the same result.  

The wireshark traffic and pertinent zigbee2mqtt files are available at [http://mcsSprinklers.com/RGBGeniePairing2.zip](url).

When I first started with zigbee2mqtt the RGBGenie was able to pair, but don't know why it is no longer able to.  With all the traffic in wireshark it does look like it is trying, but zigbee2mqtt does not complete the handshake.

Koenkk, 2019-01-11:
@mcsSolutions without getting it to pair with zigbee2mqtt, we cannot continue. Could you perhaps try figuring out why you could do this previously?

mcsSolutions, 2019-01-12:
I did a restart of the computer where zigbee2mqtt is installed and added a bridge and was able to get RGBGenie pairing.  I am  not familiar with wireshark to filter for the message with the pairing encryption key so not able to view the data when not encrypted.  The entire session including other device pairing was included in the wireshark capture at [http://mcsSprinklers.com/RGBGenie3.zip](url)

The RGBGenie device ID ends in 4FDA.  The coordinator in A280.  The router in F2C8.  During the session I also tried to pair Aqara Water Leak detector 0C6F as it was something easy to pair.

At record starting 5900 was the press of the 1-On button.  At record 5949 was press of the S1 button.  In both cases wireshark reflected the communication, but nothing appeared in the zigbee2mqtt console and of course no MQTT message.

The console for most of this RGBGenie activity is shown below.

root@MQTT:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.0.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2019-1-11 15:43:00 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-01-11.15-42-59'
  zigbee2mqtt:info 2019-1-11 15:43:01 Starting zigbee2mqtt version 1.0.1 (commit #b0d3c2f)
  zigbee2mqtt:info 2019-1-11 15:43:01 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-1-11 15:43:02 zigbee-shepherd started
  zigbee2mqtt:info 2019-1-11 15:43:02 Coordinator firmware version: '20180815'
  zigbee2mqtt:info 2019-1-11 15:43:02 Currently 5 devices are joined:
  zigbee2mqtt:info 2019-1-11 15:43:02 0x00158d0002b23edc (0x00158d0002b23edc): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
  zigbee2mqtt:info 2019-1-11 15:43:02 0xb0ce18140001de1b (0xb0ce18140001de1b): E11-N1EA - Sengled Element Plus Color (A19) (EndDevice)
  zigbee2mqtt:info 2019-1-11 15:43:02 0x000d6f000fc74fda (0x000d6f000fc74fda): ZGRC-KEY-013 - RGBgenie 3 Zone remote and dimmer (EndDevice)
  zigbee2mqtt:info 2019-1-11 15:43:02 0x00124b0008c1f2c8 (0x00124b0008c1f2c8): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
  zigbee2mqtt:info 2019-1-11 15:43:02 0x00158d0002130c6f (0x00158d0002130c6f): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
  zigbee2mqtt:warn 2019-1-11 15:43:02 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-1-11 15:43:02 Allowing new devices to join.
  zigbee2mqtt:warn 2019-1-11 15:43:02 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-1-11 15:43:02 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-1-11 15:43:02 Connecting to MQTT server at mqtt://192.168.0.30
  zigbee2mqtt:info 2019-1-11 15:43:02 zigbee-shepherd ready
  zigbee2mqtt:info 2019-1-11 15:43:02 Connected to MQTT server
  zigbee2mqtt:info 2019-1-11 15:43:02 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-1-11 15:43:02 MQTT publish: topic 'zigbee2mqtt/0x00158d0002b23edc', payload '{"angle_x":3,"angle_y":0,"angle_z":87,"angle_x_absolute":87,"angle_y_absolute":90,"linkquality":0,"unknown_data":2818048,"battery":42,"voltage":2985}'
  zigbee2mqtt:info 2019-1-11 15:43:02 MQTT publish: topic 'zigbee2mqtt/0xb0ce18140001de1b', payload '{"state":"ON","brightness":5,"color":{"x":0.6,"y":0.3},"color_temp":250}'
  zigbee2mqtt:info 2019-1-11 15:43:02 MQTT publish: topic 'zigbee2mqtt/0x000d6f000fc74fda', payload '{"battery":82,"linkquality":81}'
  zigbee2mqtt:info 2019-1-11 15:43:02 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":63,"description":"34935/0x000D6F000FC74FDA","type":"CHILD_RFD","rssi":31}'
  zigbee2mqtt:error 2019-1-11 15:43:53 Failed to configure 0xb0ce18140001de1b 0xb0ce18140001de1b
  zigbee2mqtt:error 2019-1-11 15:43:53 Failed to configure 0x000d6f000fc74fda 0x000d6f000fc74fda
  zigbee2mqtt:info 2019-1-11 15:43:58 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":60,"description":"34935/0x000D6F000FC74FDA","type":"CHILD_RFD","rssi":31}'
  zigbee2mqtt:info 2019-1-11 15:43:58 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":60,"description":"0/0x00124B0018E1A280","type":"COORD","rssi":18}'
  zigbee2mqtt:info 2019-1-11 15:43:58 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":60,"description":"21962/0x00158D0002130C6F","type":"CHILD_RFD","rssi":117}'
  zigbee2mqtt:info 2019-1-11 15:43:58 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":60,"description":"34935/0x000D6F000FC74FDA","type":"CHILD_RFD","rssi":31}'
  zigbee2mqtt:warn 2019-1-11 15:44:04 Message without device!
  zigbee2mqtt:warn 2019-1-11 15:44:04 Message without device!
  zigbee2mqtt:info 2019-1-11 15:44:04 Device incoming...
  zigbee2mqtt:info 2019-1-11 15:44:04 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2019-1-11 15:44:58 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":68,"description":"34935/0x000D6F000FC74FDA","type":"CHILD_RFD","rssi":31}'
  zigbee2mqtt:info 2019-1-11 15:44:58 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":65,"description":"0/0x00124B0018E1A280","type":"COORD","rssi":22}'
[... cut ...]
  zigbee2mqtt:info 2019-1-11 15:45:58 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":68,"description":"34935/0x000D6F000FC74FDA","type":"CHILD_RFD","rssi":23}'

On a subsequent button push to request battery status the following was produced in console.  The MQTT message was delivered,. On prior press of the 1-ON and S1 buttons there was no console feedback and no MQTT message.
  
  zigbee2mqtt:info 2019-1-11 16:14:47 MQTT publish: topic 'zigbee2mqtt/0x000d6f000fc74fda', payload '{"battery":66,"linkquality":76}'

All related files are included in the referenced zip.

Koenkk, 2019-01-13:
The device functions similar as the TRADFRI five button remote control.

- Can you add the device here: https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/coordinatorGroup.js#L8

- Follow: https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/getting_started/pairing_devices.md#ikea-tradfri-remote-control-e1524 (in your case the group id is: `0x7701`)

mcsSolutions, 2019-01-14:
I added the coordinatorGroup.js to my setup.  Changed 1524 to 7701 in the file, but don't know if it makes any difference or not.  Also added the coordinator line in the configuration.  After restarting zigbee2mqtt there was no mqtt or console response to any button press on the RGBGenie.  

What should I try next?

Koenkk, 2019-01-14:
Are you on the latest dev branch? Otherwise that file will not be used.

mcsSolutions, 2019-01-14:
I copied the dev branch to overwrite my install on RPi except the /Data folder.  Configuration.yaml was editted per the second link above.  It is shown below.  I then ran npm install and npm start.  I continue to obtain nothing in the console or mqtt related to RGBGenie when any button is pushed.  I do get feedback for router and a vibration sensor that I used to confirm it is otherwise working.  It is also shown below.

homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.0.30'
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d0002b23edc':
    friendly_name: '0x00158d0002b23edc'
    retain: false
  '0x00124b0008c1f2c8':
    friendly_name: '0x00124b0008c1f2c8'
    retain: false
  '0x00158d0002130c6f':
    friendly_name: '0x00158d0002130c6f'
    retain: false
  '0x000d6f000fc74fda':
    friendly_name: '0x000d6f000fc74fda'
    retain: false
    coordinator_group: 0x7701
  '0xb0ce18140001de1b':
    friendly_name: '0xb0ce18140001de1b'
    retain: false
  '0x00124b0018e1a280':
    friendly_name: '0x00124b0018e1a280'
    retain: false

  zigbee2mqtt:info 2019-1-14 12:41:54 MQTT publish: topic 'zigbee2mqtt/0x00158d0002b23edc', payload '{"angle_x":3,"angle_y":0,"angle_z":87,"angle_x_absolute":87,"angle_y_absolute":90,"linkquality":63,"unknown_data":720896,"battery":31,"voltage":2965,"action":"vibration","last_seen":"2019-01-14T20:41:54.276Z"}'
  zigbee2mqtt:info 2019-1-14 12:42:10 MQTT publish: topic 'zigbee2mqtt/0x00158d0002b23edc', payload '{"angle_x":3,"angle_y":0,"angle_z":87,"angle_x_absolute":87,"angle_y_absolute":90,"linkquality":63,"unknown_data":720896,"battery":31,"voltage":2965,"action":"drop","last_seen":"2019-01-14T20:42:10.131Z"}'
  zigbee2mqtt:info 2019-1-14 12:42:18 MQTT publish: topic 'zigbee2mqtt/0x00158d0002b23edc', payload '{"angle_x":4,"angle_y":1,"angle_z":86,"angle_x_absolute":86,"angle_y_absolute":89,"linkquality":42,"unknown_data":720896,"battery":31,"voltage":2965,"last_seen":"2019-01-14T20:42:18.084Z"}'
  zigbee2mqtt:info 2019-1-14 12:42:36 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":49,"description":"59251/0x00158D0002B23EDC","type":"CHILD_RFD","rssi":1,"last_seen":"2019-01-14T20:42:36.895Z"}'
  zigbee2mqtt:info 2019-1-14 12:42:36 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":49,"description":"0/0x00124B0018E1A280","type":"COORD","rssi":8,"last_seen":"2019-01-14T20:42:36.955Z"}'
  zigbee2mqtt:info 2019-1-14 12:42:36 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":49,"description":"21962/0x00158D0002130C6F","type":"CHILD_RFD","rssi":1,"last_seen":"2019-01-14T20:42:36.960Z"}'
  zigbee2mqtt:info 2019-1-14 12:42:37 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":49,"description":"34935/0x000D6F000FC74FDA","type":"CHILD_RFD","rssi":170,"last_seen":"2019-01-14T20:42:37.001Z"}'
  zigbee2mqtt:info 2019-1-14 12:42:37 MQTT publish: topic 'zigbee2mqtt/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":49,"description":"59251/0x00158D0002B23EDC","type":"CHILD_RFD","rssi":44,"last_seen":"2019-01-14T20:42:37.006Z"}'

Koenkk, 2019-01-15:
Can you provide the complete log?

mcsSolutions, 2019-01-15:
Turns out that the battery was dead. The only thing I have done with it is the testing for zigbee2mqtt so the battery life must not be very good.

What I did today is installed new battery.  Did a factory reset of the remote.  Paired the device.  This was around record 4600 in the first/larger wireshark file and about 12:15 PM in the log-1.txt.  Looking at the console I received the not-recognized message and then observed that devices.js was a download and not the one that included the prior work.

I placed my latest devices.js on the RPi and restarted zigbee2mqtt and whsniff.  Pressed the button pair to request battery status.  Console abort resulted.  The console-2.txt and the other -2 files are included in http://mcsSprinklers.com/RGBGenie4.zip.  In the devices.js I have further edits than was last posed on this thread in an attempt to inlcude the battery reporting as well as get mqtt for the previously recognized buttons.  I may have made some error in trying to follow other patterns.

Koenkk, 2019-01-16:
Did you also make the changes mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/642#issuecomment-453861395 ?

mcsSolutions, 2019-03-21:
Don’t know the next step.  Wireshark sniffer traffic was captured and posted.  Don’t know how to use what was captured to make any updates to get the device to be operational with zigbee2mqtt.

Koenkk, 2019-03-21:
Sorry for the late response, somehow I missed it.

Many things has been changed in the meantime and https://github.com/Koenkk/zigbee2mqtt/issues/642#issuecomment-453861395 isn't needed anymore.

Let's start fresh:
- Make sure you are on zigbee2mqtt 1.2.1
- And are running at least the `20190223` firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin

Do any `No converter available..` messages appear when pressing the buttons?

mcsSolutions, 2019-03-23:
Flashed latest firmware and updated to latest zigbee2mqtt
Normal operation after restart until RGB Genie button is presssed.  Initially message that device not supported.  This is expected because devices.js was not yet edited.  After adding definition for it then npm aborts.  Two different abort locations depending upon if a fromZigbee converter was specified or not.  Log of activity follows...

On button
----------

zigbee2mqtt:info 2019-3-23 12:50:55 MQTT publish: topic 'zigbeeWA/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":49,"description":"0/0x00124B0018E1A280","type":"COORD","rssi":3}'
  zigbee2mqtt:warn 2019-3-23 12:51:50 Device with modelID 'ZGRC-KEY-013' is not supported.
  zigbee2mqtt:warn 2019-3-23 12:51:50 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-3-23 12:51:51 Device with modelID 'ZGRC-KEY-013' is not supported.
  zigbee2mqtt:warn 2019-3-23 12:51:51 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-3-23 12:51:51 Device with modelID 'ZGRC-KEY-013' is not supported.
  zigbee2mqtt:warn 2019-3-23 12:51:51 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-3-23 12:51:51 Device with modelID 'ZGRC-KEY-013' is not supported.
  zigbee2mqtt:warn 2019-3-23 12:51:51 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-3-23 12:51:51 Device with modelID 'ZGRC-KEY-013' is not supported.
  zigbee2mqtt:warn 2019-3-23 12:51:51 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-3-23 12:51:53 Device with modelID 'ZGRC-KEY-013' is not supported.
  zigbee2mqtt:warn 2019-3-23 12:51:53 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 2019-3-23 12:51:53 Device with modelID 'ZGRC-KEY-013' is not supported.
  zigbee2mqtt:warn 2019-3-23 12:51:53 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Add to devices.js
-----------------

	// RGBGenie

   {
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        description: 'RGBenie remote control',
        supports:
            'on off, brightness up/down, scene select', 
        vendor: 'RGBGenie',
        fromZigbee: [fz.generic_battery, fz.generic_state, fz.generic_state_change, fz.light_brightness],
        toZigbee: [],
    },

Results in abort when On button pressed
---------------------------------------

/opt/zigbee2mqtt/lib/extension/deviceReceive.js:79
                return c.cid === cid && c.type === search;
                         ^

TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:79:26)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:74:52)
    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:139:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:139:14)
    at Zigbee.onMessage (/opt/zigbee2mqtt/lib/zigbee.js:185:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:134:14)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-23T19_58_48_871Z-debug.log

Changed devices.js to remove converter reference
------------------------------------------------

[... 65 lines cut ...]
TypeError: Cannot read property 'filter' of undefined
    at DeviceReceive.onZigbeeMessage (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:74:52)
    at extensions.filter.forEach (/opt/zigbee2mqtt/lib/controller.js:139:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/opt/zigbee2mqtt/lib/controller.js:139:14)
    at Zigbee.onMessage (/opt/zigbee2mqtt/lib/zigbee.js:185:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:134:14)
    at emitMany (events.js:147:13)
    at ZShepherd.emit (events.js:224:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-23T20_32_59_942Z-debug.log
root@MQTT:/opt/zigbee2mqtt#

remove RGBGenie definition from devices.js.  After restart operation returns to same as initially attempt with ZGRC-KEY-013 not being supported
------------------------------------------------------------------------------------------------------------------------------------------------

Koenkk, 2019-03-25:
You should define the converters as empty arrays:

```js
    {
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        description: 'RGBenie remote control',
        supports:  'on off, brightness up/down, scene select', 
        vendor: 'RGBGenie',
        fromZigbee: [],
        toZigbee: [],
    },
```

mcsSolutions, 2019-03-25:
Much more progress.
No converters available for scene cmdRecall so

added the following to fromZigbee.js 
------------------------------------------

    RGBGenie_scene: {
        cid: 'genScenes',
        type: 'cmdRecall',
        convert: (model, msg, publish, options) => {
           return {
                scene: {
                    groupid: msg.data.data.groupid,
                    sceneid: msg.data.data.sceneid,
                },
            };
        },
    },

added to devices.js:
----------------------
    // RGBGenie

    {
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        description: 'RGBenie remote control',
        supports:
            'on off, brightness up/down, scene select', 
        vendor: 'RGBGenie',
        fromZigbee: [fz.generic_battery, fz.genOnOff_cmdOn, fz.genOnOff_cmdOff,
                     fz.E1743_brightness_up, fz.E1743_brightness_down, fz.E1743_brightness_stop,
                     fz.RGBGenie_scene,
                    ],
        toZigbee: [],
    },

Remaining Issues
------------------
#1 There are four brightness buttons  using any of the four results in the same MQTT message.  Before updating devices.js with brightness conversion direction the following was obtained when brightness #4 and brightness #2 were pressed.  i.e. the same data without anything to distinguish button #4 vs. button #2.  What is the best way to coax more info so the converter can be updated to include the button id?

Button #4
  zigbee2mqtt:warn 2019-3-25 10:22:23 No converter available for 'ZGRC-KEY-013' with cid 'genLevelCtrl', type 'cmdMoveWithOnOff' and data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":50}}'
  zigbee2mqtt:warn 2019-3-25 10:22:23 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 2019-3-25 10:22:24 No converter available for 'ZGRC-KEY-013' with cid 'genLevelCtrl', type 'cmdStopWithOnOff' and data '{"cid":"genLevelCtrl","data":{}}'
  zigbee2mqtt:warn 2019-3-25 10:22:24 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Button #2
 zigbee2mqtt:info 2019-3-25 10:23:04 MQTT publish: topic 'zigbeeWA/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 2019-3-25 10:23:31 No converter available for 'ZGRC-KEY-013' with cid 'genLevelCtrl', type 'cmdMoveWithOnOff' and data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":50}}'
  zigbee2mqtt:warn 2019-3-25 10:23:31 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:warn 2019-3-25 10:23:31 No converter available for 'ZGRC-KEY-013' with cid 'genLevelCtrl', type 'cmdStopWithOnOff' and data '{"cid":"genLevelCtrl","data":{}}'
  zigbee2mqtt:warn 2019-3-25 10:23:31 Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

#2 The MQTT payload always contains the scene payload even though the scene button was not pressed.  It should only deliver scene content following a button push.  Likely a result of how I defined the converter for the scene.

#3 When the scene 1 or scene 2 button is pressed there are 4 identical MQTT messages published.  THey each look like: {"battery":68,"linkquality":36,"scene":{"groupid":0,"sceneid":1}}'

Koenkk, 2019-03-27:
Can you update to the latest dev branch, remove all converters from `fromZigbee` (use `fromZigbee:[],`). And provide the `Received zigbee message` and `No converter available` message for each button (please clearly annotate from which one it came). For this you need to set `log_level: debug` (http://www.zigbee2mqtt.io/configuration/configuration.html)

mcsSolutions, 2019-03-28:
I am not clear on how to update to the Dev branch.  What I did is from Github selected dev as the branch.  Downloaded the zip, unzipped and copied all the files to the /opt/zigbee2mqtt folder on the RPi that has the CC2531.  Edited the yaml and devices.js file as specified in prior post.

When I do npm start the log file is created, but no zigbee traffic is recognized.  This included from other working devices as well as the RGBGenie.

What is the process to update to Dev branch?

The log that was produced is

2019-3-28 15:54:35 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-28.15-54-35'
2019-3-28 15:54:35 - debug: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-03-23.13-47-20'
2019-3-28 15:54:36 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
2019-3-28 15:54:36 - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2019-3-28 15:54:36 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-3-28 15:54:36 - info: Starting zigbee2mqtt version 1.2.1 (commit #4048cb8)
2019-3-28 15:54:36 - info: Starting zigbee-shepherd
2019-3-28 15:54:39 - info: zigbee-shepherd started
2019-3-28 15:54:39 - info: Coordinator firmware version: '20190223'
2019-3-28 15:54:39 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b001936a82b","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190223},"startTime":1553813679,"joinTimeLeft":0}
2019-3-28 15:54:39 - info: Currently 6 devices are joined:
2019-3-28 15:54:39 - info: 0x00158d0002b23edc (0x00158d0002b23edc): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
2019-3-28 15:54:39 - info: 0xb0ce18140001de1b (0xb0ce18140001de1b): E11-N1EA - Sengled Element Plus Color (A19) (EndDevice)
2019-3-28 15:54:39 - info: 0x000d6f000fc74fda (0x000d6f000fc74fda): ZGRC-KEY-013 - RGBGenie RGBenie remote control (EndDevice)
2019-3-28 15:54:39 - info: 0x00124b0008c1f2c8 (0x00124b0008c1f2c8): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
2019-3-28 15:54:39 - info: 0x00158d0002130c6f (0x00158d0002130c6f): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
2019-3-28 15:54:39 - info: 0x000d6f001099e6c1 (0x000d6f001099e6c1): 3326-L - Iris Motion and temperature sensor (EndDevice)
2019-3-28 15:54:39 - warn: `permit_join` set to  `true` in configuration.yaml.
2019-3-28 15:54:39 - warn: Allowing new devices to join.
2019-3-28 15:54:39 - warn: Set `permit_join` to `false` once you joined all devices.
2019-3-28 15:54:39 - info: Zigbee: allowing new devices to join.
2019-3-28 15:54:39 - info: Connecting to MQTT server at mqtt://192.168.0.30
2019-3-28 15:54:39 - debug: Configuring 0x000d6f001099e6c1 (0x000d6f001099e6c1) ...
2019-3-28 15:54:39 - info: zigbee-shepherd ready
2019-3-28 15:54:39 - info: Connected to MQTT server
2019-3-28 15:54:39 - info: MQTT publish: topic 'zigbeeWA/bridge/state', payload 'online'
2019-3-28 15:54:39 - info: MQTT publish: topic 'zigbeeWA/0x00158d0002b23edc', payload '{"angle_x":11,"angle_y":-7,"angle_z":-77,"angle_x_absolute":79,"angle_y_absolute":97,"linkquality":55,"unknown_data":720896,"battery":91,"voltage":2985}'
2019-3-28 15:54:39 - info: MQTT publish: topic 'zigbeeWA/0xb0ce18140001de1b', payload '{"state":"ON","brightness":5,"color":{"x":0.17,"y":0.7},"color_temp":250,"color_mode":1}'
2019-3-28 15:54:39 - info: MQTT publish: topic 'zigbeeWA/0x000d6f000fc74fda', payload '{"battery":68,"linkquality":36,"scene":{"groupid":0,"sceneid":1}}'
2019-3-28 15:54:39 - info: MQTT publish: topic 'zigbeeWA/0x00124b0008c1f2c8', payload '{"state":false,"linkquality":55,"description":"46158/0x00124B001BC74A87","type":"CHILD_FFD_RXI","rssi":0}'
2019-3-28 15:54:39 - info: MQTT publish: topic 'zigbeeWA/0x00158d0002130c6f', payload '{"battery":97,"voltage":2995,"linkquality":65}'
2019-3-28 15:54:39 - info: MQTT publish: topic 'zigbeeWA/0x000d6f001099e6c1', payload '{"occupancy":false,"tamper":true,"battery_low":false,"linkquality":57,"temperature":20.76}'
2019-3-28 15:54:39 - info: MQTT publish: topic 'zigbeeWA/bridge/config', payload '{"version":"1.2.1","commit":"4048cb8","coordinator_firmware":20190223,"log_level":"debug","permit_join":true}'
2019-3-28 15:54:40 - debug: Mounted the cieApp (epId 11)
2019-3-28 15:54:58 - warn: Failed to configure 0x000d6f001099e6c1 (0x000d6f001099e6c1) ('Error: AF data request fails, status code: 183. APS no ack.') (attempt #1)
2019-3-28 15:54:58 - warn: This can be ignored if the device is working properly
2019-3-28 15:59:36 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-3-28 16:01:15 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-3-28 16:01:15 - info: MQTT publish: topic 'zigbeeWA/bridge/state', payload 'offline'
2019-3-28 16:01:15 - info: Disconnecting from MQTT server
2019-3-28 16:01:15 - info: zigbee-shepherd stopped

mcsSolutions, 2019-04-04:
Started fresh with factory reset of RGBGenie.  

setup zigbee2mqtt  using "git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt -b dev", followed by "npm install"

Ran four test cases.  The logs of three are at http://mcsSprinklers.com/RGBGenie.zip.  The forth had no applicable log data.  In this fourth case I waited several minutes before pressing button after "npm start".  Other  three a button was pressed soon after startup.  It did not matter which button initially pressed.   'devInterview' interview messages repeated after being pressed.  After they stopped a button push resulted in log entries, but generally no further button pushes resulted in log entries.

devices.js with RGBGenie appended, the .yaml, and the .db from data folder were also included with the three annotated logs.

Manual for RGBGenie is at https://rgbgenie.com/wp-content/uploads/2018/12/ZB-5001-user-manual.pdf

Koenkk, 2019-04-08:
Sorry for the delay, what is the difference between the attempts? Why do some button works and in another attempt they don't?

Let's fill in this overview:
---------------------------
- All on: works
- All off: works

**group 1**
  - on:
  - off
  - dim up
  - dim down

**group 2**
  - on:
  - off
  - dim up
  - dim down

**group 3**
  - on:
  - off
  - dim up
  - dim down

**group 4**
  - on:
  - off
  - dim up
  - dim down

**group 5**
  - on:
  - off
  - dim up
  - dim down

mcsSolutions, 2019-04-08:
I believe the unit is in some learning mode with 'devInterview' being broadcast.  Once any button is pushed the mode is exited and subsequent buttons are not recognized.  The multiple attempts showed it did not matter which button was first pushed.  If no buttons were pushed before 'devInterview' stopped being broadcast then no log/console info was produced with any button push.

 It seems to enter the 'devInterview' with each zigbee2mqtt start.  The manual describes pairing with other zigbee devices as I think that is its primary design function to operate independent of hub.  I do not know what 'devInterview' is in the zigbee 3 schema.

 I was able to get relatively full functionality before I did the factory reset on the unit and before I set fromZigbee to [].  The problem was that each brightness control button produced the same mqtt message as if the unit only had one up/down/on/off button.

Koenkk, 2019-04-10:
I think this is happening because the `configure` is missing, please try with:

```js
{
        zigbeeModel: ['ZGRC-KEY-013'],
        model: 'ZGRC-KEY-013',
        vendor: 'RGBgenie',
        description: '3 Zone remote and dimmer',
        supports: 'onoff dim scene control',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

mcsSolutions, 2019-04-10:
Started over with new flash, no db, no devices in yaml, reset/re-pair rgbgenie.  Log at http://mcsSprinklers.com/RGBGenie_4_10_2019.zip contains pairing and initial handshake.  No buttons pushed for several minutes.

The log annotated with each button push.

Observations:
1, S1 and S1 button resulted in no log entries
2. Same log entry for each button 1, 2, 3 and 4 (i.e. no specific button identification)

Also included in zip is yaml and db from end of test.

Koenkk, 2019-04-11:
@mcsSolutions as far as I can see there is a difference, you see that the endpoint it get the message from is different (1 - 4, it's and the end of the log lines).

This converter: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L977 shows how to take the endpoint ID from the message.

mcsSolutions, 2019-04-11:
I am able to get the button number in the MQTT payload, but it reserves a position in the MQTT payload as each button is pressed.  What I want is a MQTT message where the payload only contains the event of a specific button being pressed to the On or OFF state.  I do not understand the syntax well enough to accomplish this.

Another problem I continue to have is the lack of distinction between the All On/All Off buttons and the button 1 On/button1 Off.  Pressing All On and Button 1 On produce the same MQTT message.  The prior posting at http://mcsSprinklers.com/RGBGenie_4_10_2019.zip shows that both button id 1 is reported for both All and Button 1.

Two converters I added to fromZigbee.js are:

ZGRC013_cmdOn: {
        cid: 'genOnOff',
        type: 'cmdOn',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                const payload = {};
                payload[`button_${button}`] = 'on';
                return payload;
            }
        },
    },
    ZGRC013_cmdOff: {
        cid: 'genOnOff',
        type: 'cmdOff',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                const payload = {};
                payload[`button_${button}`] = 'off';
                return payload;
            }
        },
    },

Example of pressing the Button 2, 3 and 4  ON buttons

2019-4-11 11:18:45 - info: MQTT publish: topic 'zigbee/bridge/state', payload 'online'
2019-4-11 11:18:45 - info: MQTT publish: topic 'zigbee/0x000d6f000fc74fda', payload '{"battery":71,"linkquality":0}'
2019-4-11 11:18:45 - info: MQTT publish: topic 'zigbee/bridge/config', payload '{"version":"1.2.1","commit":"32db1fd","coordinator_firmware":20190223,"log_level":"debug","permit_join":true}'
2019-4-11 11:19:14 - debug: Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'ZGRC-KEY-013' (0x000d6f000fc74fda) with groupID undefined of endpoint 2
2019-4-11 11:19:14 - info: MQTT publish: topic 'zigbee/0x000d6f000fc74fda', payload '{"battery":71,"linkquality":34,"button_2":"on"}'
2019-4-11 11:23:53 - debug: Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'ZGRC-KEY-013' (0x000d6f000fc74fda) with groupID undefined of endpoint 3
2019-4-11 11:23:53 - info: MQTT publish: topic 'zigbee/0x000d6f000fc74fda', payload '{"battery":71,"linkquality":0,"button_2":"on","button_3":"on"}'
2019-4-11 11:24:45 - debug: Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'ZGRC-KEY-013' (0x000d6f000fc74fda) with groupID undefined of endpoint 4
2019-4-11 11:24:45 - info: MQTT publish: topic 'zigbee/0x000d6f000fc74fda', payload '{"battery":71,"linkquality":0,"button_2":"on","button_3":"on","button_4":"on"}'

Koenkk, 2019-04-12:
Let's first try to distinguish the all / 1 on/off messages. Could you provide a wireshark sniff log of both of these buttons? (Please provide the message ID, most left column, of the messages)

mcsSolutions, 2019-04-12:
Process was to reset RGBGenie to factory, reflash CC2531.  remove .db and edit .yaml to remove devices section. Start wireshark, start zigbee2mqtt.  Pair RGBGenie.  Pairing Device Announcement is at record 188 in Wireshark.  Interview proceeded and some random buttons pushed.  Interview stopped and controlled button push with the following collected in Wireshark record #'s
1268  All On
1287 All lOff
1300 Button 1 On
1316 Button 1 Off
1328 Button 2 On
1343 Button 2 Off

Wireshark, log, devices.js and fromZigbee at http://mcsSprinklers.com/RBGGenie_Pair_Interview_All_1_2.zip

Koenkk, 2019-04-13:
I have no clue how bulbs are distinguishing between group 1 and all off, 1268 and 1300 are exactly the same message.

My guess is that when paired it will send it to a specific bulb only, something which is not possible when paired to only the coordinator.

mcsSolutions, 2019-04-13:
How about the syntax so only the last button push is reported?

Koenkk, 2019-04-15:
You can use the following converters:
```js
ZGRC013_cmdOn: {
        cid: 'genOnOff',
        type: 'cmdOn',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                return {click: `${button}_on`}
            }
        },
    },
    ZGRC013_cmdOff: {
        cid: 'genOnOff',
        type: 'cmdOff',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                return {click: `${button}_off`}
            }
        },
    },
    ZGRC013_brightness_up: {
        cid: 'genLevelCtrl',
        type: 'cmdMove',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                return {click: `${button}_down`}
            }
        },
    },
    ZGRC013_brightness_down: {
        cid: 'genLevelCtrl',
        type: 'cmdMoveWithOnOff',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                return {click: `${button}_up`}
            }
        },
    },
    ZGRC013_brightness_stop: {
        cid: 'genLevelCtrl',
        type: 'cmdStopWithOnOff',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                return {click: `${button}_stop`}
            }
        },
    },
    ZGRC013_scene: {
        cid: 'genScenes',
        type: 'cmdRecall',
        convert: (model, msg, publish, options) => {
            return {click: `scene_${msg.data.data.groupid}_${msg.data.data.sceneid}`};
        },
    },
```

(it might be that you need to clear `data/state.json` once if you still see the old payloads).

mcsSolutions, 2019-04-16:
Progress with still some questions

The up/down buttons produce a Payload field in the Zigbee Cluster Library Frame that contains direction of motion.  I captured it in Wireshark at http://mcsSprinklers.com/RGBGenie_Up_Down_7_19_48_56.zip.  Records 7 and 48 for the two directions.  I tried to update fromZigbee.js but did not have success with the following:

    ZGRC013_brightness_down: {
        cid: 'genLevelCtrl',
        type: 'cmdMoveWithOnOff',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            const direction = msg.data.data['Payload'];
            if (button) {
                return {click: `${button}_${direction}`}
            }
        },
    },

What shows up in the console/log (below) is "undefined" for direction.  (i.e. '{"battery":69,"linkquality":0,"click":"4_undefined"}')

  zigbee2mqtt:debug 2019-4-16 09:40:00 Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 2019-4-16 09:40:34 Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'ZGRC-KEY-013' (0x000d6f000fc74fda) with groupID undefined of endpoint 4
  zigbee2mqtt:info 2019-4-16 09:40:34 MQTT publish: topic 'zigbee/0x000d6f000fc74fda', payload '{"battery":69,"linkquality":0,"click":"4_on"}'
  zigbee2mqtt:debug 2019-4-16 09:40:38 Received zigbee message of type 'cmdMoveWithOnOff' with data '{"cid":"genLevelCtrl","data":{"movemode":0,"rate":50}}' of device 'ZGRC-KEY-013' (0x000d6f000fc74fda) with groupID undefined of endpoint 4
  zigbee2mqtt:info 2019-4-16 09:40:38 MQTT publish: topic 'zigbee/0x000d6f000fc74fda', payload '{"battery":69,"linkquality":0,"click":"4_undefined"}'
  zigbee2mqtt:debug 2019-4-16 09:40:42 Received zigbee message of type 'cmdStopWithOnOff' with data '{"cid":"genLevelCtrl","data":{}}' of device 'ZGRC-KEY-013' (0x000d6f000fc74fda) with groupID undefined of endpoint 4
  zigbee2mqtt:info 2019-4-16 09:40:42 MQTT publish: topic 'zigbee/0x000d6f000fc74fda', payload '{"battery":69,"linkquality":0,"click":"4_stop"}'
  zigbee2mqtt:debug 2019-4-16 09:45:00 Saving state to file /opt/zigbee2mqtt/data/state.json
 
Second question relates to the above log snippet.  The up/down button produce log message of "with groupID undefined of endpoint 4".  Should something be done to avoid it.

The third question is the startup operation.  Each time I restart zigbee2mqtt I get the "devInterview" sequence after I press the first RGBGenie button.  This takes a few minutes.  It also requires that the RGBGenie be close to the CC2531.  If it is a couple rooms away then button presses are not recognized after startup.  After "devInterview" completes then RGBGenie can be moved away and works normally.  The "devInterview" sequence was previously captured in the post above four days ago at  http://mcsSprinklers.com/RBGGenie_Pair_Interview_All_1_2.zip

Koenkk, 2019-04-18:
**1**
For the direction:
```js
ZGRC013_brightness_down: {
    cid: 'genLevelCtrl',
    type: 'cmdMoveWithOnOff',
    convert: (model, msg, publish, options) => {
        const button = msg.endpoints[0].epId;
        const direction = msg.data.data.movemode == 0 ? 'up' : 'down';
        if (button) {
            return {click: `${button}_${direction}`}
        }
    },
},
```

**2**
This isn't a problem, it's debug logging

**3**
Do you have `permit_join: true`?

mcsSolutions, 2019-04-19:
I started from scratch again and now things are working as expected after startup.  The reporting is as desired with direction and with the button #.  The All on and All off report the same as the 1 on and 1 off, but that was already known.  The anomalies I noticed are:
1.  When one of the two scene button is pushed the same MQTT message is repeated four times.  If other buttons are pressed then only one message is produced.  
2. The scene group also is retained in subsequent MQTT reports even if the reported click is not a scene.  These are not showstoppers, but just not seem quite correct.  I put a log showing the behavior at http://mcsSprinklers.com/RGBGenie_4_18_19.zip

devices.js added for RGBGenie
---------------------------------

    // RGBGenie

    {
      zigbeeModel: ['ZGRC-KEY-013'],
      model: 'ZGRC-KEY-013',
      vendor: 'RGBgenie',
      description: '3 Zone remote and dimmer',
      supports: 'onoff dim scene control',
      fromZigbee: [fz.generic_battery, 
                 fz.ZGRC013_brightness_onoff, fz.ZGRC013_brightness, fz.ZGRC013_brightness_stop,
                 fz.ZGRC013_cmdOn, fz.ZGRC013_cmdOff,
                 fz.ZGRC013_scene,
                ],
      toZigbee: [],
      configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const device = shepherd.find(ieeeAddr, 1);
        const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
        const actions = [
            (cb) => device.bind('genOnOff', coordinator, cb),
            (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
        ];

        execute(device, actions, callback);
      },
    },

fromZigbee.js for RGBGenie added
--------------------------------------

    ZGRC013_cmdOn: {
        cid: 'genOnOff',
        type: 'cmdOn',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                return {click: `${button}_on`}
            }
        },
    },
    ZGRC013_cmdOff: {
        cid: 'genOnOff',
        type: 'cmdOff',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                return {click: `${button}_off`}
            }
        },
    },
    ZGRC013_brightness: {
        cid: 'genLevelCtrl',
        type: 'cmdMove',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            const direction = msg.data.data.movemode == 0 ? 'up' : 'down';
            if (button) {
                return {click: `${button}_${direction}`}
            }
        },
    },
    ZGRC013_brightness_onoff: {
        cid: 'genLevelCtrl',
        type: 'cmdMoveWithOnOff',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            const direction = msg.data.data.movemode == 0 ? 'up' : 'down';
            if (button) {
                return {click: `${button}_${direction}`}
            }
        },
    },
    ZGRC013_brightness_stop: {
        cid: 'genLevelCtrl',
        type: 'cmdStopWithOnOff',
        convert: (model, msg, publish, options) => {
            const button = msg.endpoints[0].epId;
            if (button) {
                return {click: `${button}_stop`}
            }
        },
    },
    ZGRC013_scene: {
        cid: 'genScenes',
        type: 'cmdRecall',
        convert: (model, msg, publish, options) => {
            return {click: `scene_${msg.data.data.groupid}_${msg.data.data.sceneid}`};
        },
    },

Koenkk, 2019-04-19:
1. This is also what the buttons seems to send
2. Should be fixed by removing your `data/state.json` and restarting zigbee2mqtt (probably comes from an old converter).

mcsSolutions, 2019-05-19:
I do not expect to do any more work with characterization of the RGBGenie and will use the April 18 devices.js and fromZigbee.js updates.  The only shortcoming is that the All On/Off buttons and the 1 On/Off button produce the same mqtt message.  In essence the All On/Off should be considered not operational.

Will these be added to the mainline or do I continue to add edits moving forward?

Koenkk, 2019-05-19:
Thanks, this device is now supported in the dev branch and will be supported in zigbee2mqtt 1.5.

closed 2019-05-19 by Koenkk