Koenkk/zigbee2mqtt #645: Docker on Ubuntu

57k, 2018-11-30:
Hello everybody,

tried to install zigbee2mqtt via docker (docker/Dockerfile.amd64) on Ubuntu getting the following result

The command '/bin/sh -c apk add --update --no-cache make gcc g++ python linux-headers udev nodejs=8.11.4-r0 npm=8.11.4-r0 git &&     npm install --unsafe-perm &&     apk del make gcc g++ python linux-headers udev git' returned a non-zero code: 2

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.1 LTS
Release:        18.04
Codename:       bionic

$ docker version
Client:
 Version:           18.09.0
 API version:       1.39
 Go version:        go1.10.4
 Git commit:        4d60db4
 Built:             Wed Nov  7 00:49:01 2018
 OS/Arch:           linux/amd64
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          18.09.0
  API version:      1.39 (minimum version 1.12)
  Go version:       go1.10.4
  Git commit:       4d60db4
  Built:            Wed Nov  7 00:16:44 2018
  OS/Arch:          linux/amd64
  Experimental:     false

If I run the docker container via the existing docker image ...

docker run \
   -it \
   -v $(pwd)/data:/app/data \
   --device=/dev/ttyACM0 \
   koenkk/zigbee2mqtt

the container starts without issues. Anyone in the community who is experiencing the same behavior ?

Thanks, 
Jens

Koenkk, 2018-11-30:
This is because the specified nodejs/npm version are not available anymore. https://github.com/Koenkk/zigbee2mqtt/commit/0e41dd97ea1ba29a8c7455ec4a67d0922652db2f fixes this.

57k, 2018-11-30:
ok - thanks !

closed 2018-11-30 by 57k