Koenkk/zigbee2mqtt #647: Improving range with CC2531 USB sniffer

CypherMK, 2018-12-02:
Hi,

I have run into some range issues with my CC2531 USB sniffer. But I managed to improve it dramatically with a simple solution. 

My RPI with the CC2531 USB sniffer is located in the meter cupboard (meterkast in Engels?). Also my router is located there. Sometimes my IKEA bulbs wouldn't respond properly because of range issues. Also tried putting the pi outside of the meter cupboard, but range didn't improve.

Now I tried to use a usb extension cable for my CC2531 USB snifferand located it 40 cm away from my RPI. But still in the meter cupboard. And now my range issues are gone. So maybe this simple solution works for other people too.

Koenkk, 2018-12-02:
Do we have an explanation why this happens? Does the Pi interfere with the CC2531 signal?

quarcko, 2018-12-03:
I also came up with same idea and can confirm that range is much better than directly in PI.
In my case i also have aeotec z-wave dongle in pi, so just moved one father away from another.

CypherMK, 2018-12-03:
Good, I'm not the only one. Maybe add this as recommended to the readme?

Koenkk, 2018-12-03:
Yes!

johnnyletrois, 2018-12-03:
My server is in the basement, and my Zigbee devices are in the attic master. I use a Ethernet-to-usb adapter to put the usb dongle in the attic. Range is still really bad, but I have five powered tradfri routers and one flashed cc2531 router to extend range throughout the attic and floor below. The router flashed cc2531 is nearly worthless as a range extender tho, but the bulbs and outlet work well.

mihalski, 2018-12-04:
I have both the aeotec z-wave dongle and the CC2531 in a tiny hub about 15cm from the Pi and I do not use the onboard wifi. I've not done any proper tests of the link quality but it seems to work well so far.

alco28, 2018-12-20:
it's mostly because a router, cupboard or especialy a RPI 3 generate a lot of interference on the zigbee frequency bands (800-900 & 2.4Ghz) especially a cupboard has a lot of electric and magnetic fields and will effect that tiny antenna on the USB stick. that antenna is also very low DB. you can google or youtube a lot about this like: https://www.lifewire.com/rf-interference-wireless-home-automation-devices-817680 and this is also nice: https://support.metageek.com/hc/en-us/articles/201320430-WiFi-and-ZigBee-Optimization-Video

I'm not a M2Z user or C2531 owner yet.. But I will design a retrofit antenna when I have one!

clockbrain, 2019-02-18:
This project to fit an external antenna to cc2531 looks straightforward. I'll give it a go once I get the bits I need.

[https://hackaday.io/project/163505-cc2531-usb-adapter-antenna-mod](https://hackaday.io/project/163505-cc2531-usb-adapter-antenna-mod)

closed 2019-04-26 by stale[bot]