Koenkk/zigbee2mqtt #648: Zigbee2MQTT port for Windows

mcsSolutions, 2018-12-02:
I recognize that there are instructions for use of Windows to program the sniffer, but is there a port of zigbee2mqtt that runs on Windows?

ptvoinfo, 2018-12-03:
@mcsSolutions Zigbee2MQTT runs on Windows without any problems. You may install NodeJS, Zigbee2MQTT and all depended packages.

mcsSolutions, 2018-12-03:
All the install instructions I have seen are based upon a Linux install.  Things like sudo does not work from the Windows command prompt.  Seems like the Wiki should have a Windows install to parallel the Linux one.  Just as it does for the Windows flash of the USB dongle.

mcsSolutions, 2018-12-05:
I gave it a shot at trying to install on Windows using as best I could the pattern defined for Linux.
First issue is the CC2531 device identification.  It shows in Windows Device Manager as an "Other Device" "TI CC2531 USB CDC".  I assume then that configuration.yaml will use this text rather than the "ttyACM0".  I suspect this will not be correct and zigbee2mqtt will expect something like "COM5", but no device driver for Windows has been installed for this device.

A fresh install of node.js resulted in version 10.14.1 and npm 6.4.1.  This should be good as it is a higher version than what is installed on Linux.

/opt/ folder does not exist on Windows so to make it as close as possible I did the git clone into C:\opt\zigbee2mqtt.  I already had git installed, but I suspect if I did not then this would be a prerequisite part of the instructions.

I installed from an administrator account so chown equivalent is not needed

After npm install and npm start the expected issue with the port showed up with the following in the command window:
Error while starting zigbee-shepherd, attempting to fix ...
PM failed to start
Opening TI CC2531 USB CDC: File not found,

So it seems the essential problem with the Windows install is the method to identify the USB Dongle in configuration.yaml.

mcsSolutions, 2018-12-05:
It shows up in Device Manager as an Other Device because there is no driver installed that makes the USB device look like a serial port.  Is there a place to get such a driver?

ptvoinfo, 2018-12-05:
I've installed this driver with Z-Stack. You may try the following file:
http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=swrc088&fileType=zip
driver\usb_cdc_driver_cc2531.inf
Right click -> Install

mcsSolutions, 2018-12-05:
Able to install the driver via Control Panel and now have a COM15 port.
Issue now is privledges to access port.
I opened command prompt with right-click on cmd.exe, run as administrator.  Logged into administer account on Win 7 Ultimate.
cd c:/opt/zigbee2mqtt
npm start
Receive Error: Opening COM15: Access denied

Is there a way to run npm directly so it has the elevated privileges to access the port?

mcsSolutions, 2018-12-06:
Turns out that a PC restart cleared up the access denied message.  Success was then achieved. ptvo, thanks for all the pointers.

 I documented my steps at https://forums.homeseer.com/forum/lighting-primary-technology-plug-ins/lighting-primary-technology-discussion/mcsmqtt-michael-mcsharry/1264779-zigbee2mqtt-on-windows.  It would be nice if this or similar information was available for others via the Wiki.

clockbrain, 2018-12-06:
I'd like to second those thanks to @ptvoinfo and @mcsSolutions.

I got it working on Windows 10 quite easily thanks to all the tips here. Had to install node.js and git but I didn't have to install any special drivers, or even reboot - worked straight up.

Now I had somewhere I can experiment with new device pairing etc without disrupting my operational Pi setup, Pro tip: when purchasing the CC2531 get 2 of them - one for prod and one for dev.

ptvoinfo, 2018-12-07:
Also, if you have Hyper-V on your Windows system, you may install and use a docker container. I've tried it too, but installing NodeJS is more easy :).

closed 2019-02-12 by stale[bot]