Koenkk/zigbee2mqtt #650: How to view my current pan_id & channel?

quarcko, 2018-12-03:
Just a question for devs:

I started my journey with this module on all default settings,
did not specified any 'advanced' options, and really don't see any need now, but...

i saw some posts where people mention that USB CC dongle sometimes (power cycle?)
looses and refuses to reuse pan id?? it is my worst nightmare now, 
especially since pairing each Xiaomi device takes like 20-30 tries (for EACH device!!).

So... i want to know, how can i view my current pan_id and channel, just to write it down
somewhere, and in case of emergency i could reuse it in 'advanced' configuration...
Thanks! :)

h4nc, 2019-02-01:
@Koekk does everyone get the same panID with the standard config? 
Would you recommend changing the panID.

I have to re-pair my devices because I changed my network key. If it's more secure to change the panID, it would be better to know that before re-pairing all devices.

Koenkk, 2019-02-01:
@h4nc yes, by default `0x162a`. Using the same panID isn't less secure, it's used to operate multiple zigbee networks on the same channel.

closed 2019-02-01 by Koenkk

h4nc, 2019-02-01:
Thats weird I didn't change something at the panID and I don't habe `0x162a`.

I got `0x1a63`. Should I change it to `0x162a`.

Koenkk, 2019-02-01:
When a panID is already occupied a new one will be picked (this cannot be forced, it's part of the zigbee stack on the CC2531).

h4nc, 2019-02-01:
Ok I updated my coordinator some days ago, so this explains why it went from`0x1a62` too `0x1a63`.

But above you mentioned `0x162a` as default. docs say `0x1a62`. Only the letter a is at an other place. Perhaps a typing mistake?

Koenkk, 2019-02-01:
sorry, yes that was a typo

h4nc, 2019-02-01:
Ok, thanks. So my last question on this one, as I'm still not sure about that.

Should I change it back to `0x1a62` from `0x1a63`?

And from your information above it is not a security risk, that I published my `panID` here, right?
Security is what the `network_key` provides, right?

Just want to know all of that before I re-pair all those devices. Hopefully the last time for a while.

Koenkk, 2019-02-01:
Changing the panid requires repairing of devices.

You can view panID as something similar to the channel, it's just a configuration setup and has nothing to do with security.

`network_key` is what provides the security indeed.

h4nc, 2019-02-01:
@Koenkk Thanks for your patiency and clearing that up for me. 

By the way (don't want to open an issue for that): Is there a way to reset the cc2530 router to re-pair it? Or do I have to reflash the device to get it connected again?

Koenkk, 2019-02-01:
@h4nc you have to reflash it.

h4nc, 2019-02-01:
I suspected that. Thanks

emc02, 2020-11-13:
is it possible to make the Pan ID static and to get perfect clones? I would need that for a big project to pair devices once and use them on differnet places but with same networks without repairing them. Thanks!