Koenkk/zigbee2mqtt #652: Does the zigbee network map work correctly?

mihalski, 2018-12-04:
Does the zigbee network map work as it should? Here is mine:
```
digraph G { node[shape=record]; "0x00124b001202328b" [label="{0x00124b001202328b|Coordinator|No model information available|online}"]; "0x00158d000123df04" [label="{study_button|EndDevice|Xiaomi Aqara wireless switch (WXKG11LM)|online}"]; "0x00158d000200d0c4" [label="{bedroom_motion|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"]; "0x00158d00017206c0" [label="{office_door|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"]; "0x00158d0001720704" [label="{bedroom_door|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"]; "0x00158d0002476ba9" [label="{bedroom_temp|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"]; "0x00158d0002476ba9" -> "0x00124b001202328b" [label="94"] "0x00158d00023f5036" [label="{bathroom_temp|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"]; "0x00158d0001872b69" [label="{living_room_button|EndDevice|Xiaomi MiJia wireless switch (WXKG01LM)|online}"]; "0x00158d0002476b3c" [label="{outside_temp|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"]; "0x00158d0001a668e8" [label="{master_bedroom_button|EndDevice|Xiaomi Aqara wireless switch (WXKG11LM)|online}"]; "0x00158d0001a668e8" -> "0x00124b001202328b" [label="45"] "0x00158d00019dee01" [label="{spare_bedroom_button|EndDevice|Xiaomi MiJia wireless switch (WXKG01LM)|online}"]; "0x00158d00019dee01" -> "0x00124b001202328b" [label="71"] "0x00158d000216085f" [label="{master_bedroom_fan|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"]; "0x00158d000216085f" -> "0x00124b001202328b" [label="62"] "0x00158d0001148b7e" [label="{office_cube|EndDevice|Xiaomi Mi smart home cube (MFKZQ01LM)|online}"]; "0x00158d0002437899" [label="{office_temp|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"]; "0x00158d00015da618" [label="{master_bedroom_motion|EndDevice|Xiaomi MiJia human body movement sensor (RTCGQ01LM)|online}"]; "0x00158d00015da618" -> "0x00124b001202328b" [label="72"] "0x00158d0001fa6453" [label="{spare_bedroom_booklight|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"]; "0x00158d0001fa6453" -> "0x00124b001202328b" [label="69"] "0x00158d000204658a" [label="{master_bedroom_door|EndDevice|Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online}"]; "0x00158d000200e303" [label="{office_motion|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"]; "0x00158d00015da4c3" [label="{spare_bedroom_motion|EndDevice|Xiaomi MiJia human body movement sensor (RTCGQ01LM)|online}"]; "0x00158d00015da604" [label="{stairs_motion|EndDevice|Xiaomi MiJia human body movement sensor (RTCGQ01LM)|online}"]; "0x00158d0001a2b36b" [label="{office_storage_motion|EndDevice|Xiaomi MiJia human body movement sensor (RTCGQ01LM)|online}"]; "0x00158d00026d387e" [label="{study_backlight|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"]; "0x00158d00026d387e" -> "0x00124b001202328b" [label="7"] }
```
I'm confused by the lack of links with signal quality in the diagram to the online (and working) zigbee devices. Additionally, should a router be able to connect to another router if it is closer than the co-ordinator? Because in my scenario `study_backlight` would be much better off connecting via `master_bedroom_fan`. Link quality of the `study_button` in the same room as the `study_backlight` is 68 and I assume it's connecting to the `study_backlight` router even if the diagram doesn't show that.

Thanks in advance.

Regards,
Michal

Koenkk, 2018-12-04:
The networkmap is indeed still buggy (but it's not high on my priority list).

mihalski, 2018-12-04:
Fair enough, how about router to router connections? Is that a supported feature or is that not part of the spec/implementation?

Koenkk, 2018-12-05:
Some router to router connections are shown, but note that a device can communicated with multiple other devices (something which is not shown in the networkmap).

mihalski, 2018-12-05:
Does that mean an end device can communicate with another end device to relay messages? And does this apply to low power battery powered devices? I'm slowly shifting devices from the Xiaomi gateway to zigbee2mqtt and am hoping this will make the Zigbee network more robust.

Koenkk, 2018-12-05:
No, only routers can communicate with other (multiple) routers. end devices are sleeping most of the time.

danpowell88, 2018-12-17:
Is there a way to view any of this information and confirm what is happening without using the graph?

Koenkk, 2018-12-17:
@danpowell88 the debug version the router firmwares show which childs it has.

danpowell88, 2018-12-17:
Do you have an example of what the output looks like?

milakov, 2019-01-10:
I also observe that the network map generated has a number of online end devices, which are not connected to anything. At the same time 2 routers (Xiaomi smart plug) don't have any end devices connected to them. Could it be that end device to router connections are just not shown?
Having correct network map would help to identify issues with connectivity and reliability, which is especially relevant for CC2531 USB dongle, which seems to have lower signal strength than XIaomi gateway.

Koenkk, 2019-01-16:
@clockbrain looks good, could you make a PR? We can polish the code further there.

clockbrain, 2019-01-17:
@Koenkk i've made a PR for the change to zigbee-shepherd [https://github.com/Koenkk/zigbee-shepherd/pull/9](https://github.com/Koenkk/zigbee-shepherd/pull/9).

With this change I now get 18 links in my raw network map instead of 6 links.

I'm new to github etc and I don't know how to relate my fork of zigbee-shepherd to my fork of zigbee2mqtt so I'm stuck trying to get the zigbee2mqtt network graph to show the extra links.

Koenkk, 2019-01-17:
@clockbrain thanks, I will take care of the zigbee2mqtt update.

Koenkk, 2019-01-18:
@clockbrain merged and updated zigbee2mqtt.

clockbrain, 2019-02-01:
@lolorc regarding the unlinked devices in the map - can you try requesting the map just after triggering those missing end devices. If end devices are asleep when the coordinator issues the lqi scan they may be missed. I also sometimes have unlinked end devices on my map.

@bizziebis does your network map still not work? Make sure you don't have zigbee2mqtt admin panel running as it spams the network with map requests. You could also try requesting a raw network map and counting the returned links but it is more likely that the network is busy during the lqi probe rather than something wrong with the graphviz generation code.

bizziebis, 2019-02-01:
It was looking great when I rebuild the network and had 10 devices connected. Then I added 3 more  including a CC2531 router, it didn't show the connection between routers  anymore. Only between coordinator <-> router <-> end device. 

About the busy network, I was thinking the same, as I see a lot of diag messages from my two routers. I'll try them with the non diag version so the network is more at rest.

milakov, 2019-02-02:
@clockbrain At some point I had a nice map with all devices connected. Not anymore. Using dev branch. It seems like the map shows connections from end devices to coordinator fine but no connections from end devices to routers. Waking up end devices (pressing reset buton shortly) and re-generating the map doesn't help.
[image]

clockbrain, 2019-02-02:
@milakov I see it is only end devices that aren't showing links. Are you sure these devices are all paired and actually working ok? Are all your routers shown on the map?

The online status for end devices doesn't necessarily mean they are actually online. zigbee2mqtt only checks online for routers. End devices are always shown as online regardless.

To gather the network links in response to a map request, each router is polled in turn and their neighbour table is examined. Given that you have reasonable links showing for your coordinator and some routers I guess that either a critical router to which many end devices are paired has gone away or many of these end devices have themselves dropped off your network.

bizziebis, 2019-02-02:
I noticed that when the coordinator is pinging the routers, and get successfull  reply, the network map is OK

`2019-2-2 18:21:01 - debug: Ping 0x00124b00016f2bd5
2019-2-2 18:21:01 - debug: Successfully pinged 0x00124b00016f2bd5
2019-2-2 18:21:01 - debug: Ping 0x00158d0002370efc
2019-2-2 18:21:01 - debug: Successfully pinged 0x00158d0002370efc
2019-2-2 18:21:01 - debug: Ping 0xbfb6775ffeffe79d
2019-2-2 18:21:01 - debug: Successfully pinged 0xbfb6775ffeffe79d
2019-2-2 18:21:01 - debug: Ping 0x00158d00024d8998
2019-2-2 18:21:01 - debug: Successfully pinged 0x00158d00024d8998`

When the ping of a router is not successfull, the network map seems incomplete

`2019-2-2 18:48:35 - error: Failed to ping 0x00158d0002370efc`

It's strange that at first the coordinator was pinging 4 routers with all positive results, and after a restart of Zigbee2MQTT service the coordinator was only pinging 1 router with a negative result. Nothing changed in the network betwee the 5 seconds of restarting.

Koenkk, 2019-02-02:
@bizziebis All xiamoi routers + CC2530/CC2531 routers are pinged.

When setting `availability_timeout` all devices will be pinged (https://koenkk.github.io/zigbee2mqtt/configuration/configuration.html)

bizziebis, 2019-02-03:
I remember that there was only one change I made before my network map was incomplete again. I removed one sensor which was marked as unknown, and re-paired it. After that the network map was incomplete. Before the removal it was very detailed. That also happened the last time I rebuild the network. I don't know if it was a coincidence.

quarcko, 2019-02-13:
I have exactly same story going on, and as my network "heals" and some end devices little by little are getting behind the routers that are themselves behind the routers - network map gets incomplete.
Although, all my devices are giving signals okay, and are operating just fine.
I have 5 routers, all Xiaomi stuff (smart plugs and wall switches), and some 30 end devices.

Could it bee that "router behind a router" answers to some things like ping (or map) differently?
Here is excerpt from logs:

1) Direct router, ping going OK:
`  zigbee2mqtt:debug 2019-2-13 22:46:32 Check online 0x00158d00026eb005 0x00158d00026eb005
2019-02-13T20:46:32.646Z zigbee-shepherd:request REQ --> ZDO:nodeDescReq
2019-02-13T20:46:32.700Z zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp
  zigbee2mqtt:debug 2019-2-13 22:46:32 Successfully pinged 0x00158d00026eb005`

2) This is router behind a router, as you can see in zigbee-shepherd there is something goin on (im no expert on this library) and maybe it needs to be parsed additionally?
`  zigbee2mqtt:debug 2019-2-13 22:46:32 Check online 0x00158d0002b701ae 0x00158d0002b701ae
2019-02-13T20:46:32.870Z zigbee-shepherd:request REQ --> ZDO:nodeDescReq
2019-02-13T20:46:36.834Z zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2019-02-13T20:46:36.844Z zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2019-02-13T20:46:36.846Z zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2019-02-13T20:46:37.888Z zigbee-shepherd:request REQ --> ZDO:nodeDescReq
  zigbee2mqtt:debug 2019-2-13 22:46:42 Failed to ping 0x00158d0002b701ae`

milakov, 2019-04-15:
The map is still very incomplete. I have all devices working properly for me (at least most of the time), but almost all end devices are not connected to anything:
[image]

Koenkk, 2019-04-15:
@milakov this is expected, as during scanning these are sleeping.

milakov, 2019-04-15:
@Koenkk Notice I have a couple of Aqara weather sensors directly connected to the coordinator. I am pretty sure they are sleeping most of the time as well (battery powered) but they are always shown in the map with a link to the coordinator.

ualex73, 2019-04-15:
I have updated Zigbee2MQTT and after this, all my sensors are disconnected in the map (but working fine). I have waited 6 days, but nothing has changed in my map? Is there anything I can do/trigger to fix this?

quarcko, 2019-04-15:
For me it helped to power off all routers for some time and power them back on.

I did it because was doing some maintenance at home and had to shut down whole house power.

After this map was complete.

Then i added 2 more routers and map is again incomplete. Will try same scheme sometime later.

My hassio is on UPS so it is not affected by powering off whole house for an hour.
Coordinator was kept online.

My routers btw are all xiaomi wall switches with neutral.

⁣Sent from BlueMail ​

h4nc, 2019-04-15:
I had the same and wrote down steps that fixed it for me somewhere here on GitHub.

If I remember correctly.

Unplugging the routers and restarting zigbee fixed it for me.

After that plug in your routers again.

Could also be that I had to move the routers nearer to the Coord at the restart.

milakov, 2019-05-01:
Interestingly, when I run map update the coordinator send "Link Quality Request" to one of the routers (Gledopto bulb), it either responds with "Status: Not Supported"or doesn't respond at all, in both cases coordinator doesn't ask any other device about link quality. Could it be a bug?

[image]

lolorc, 2019-05-02:
> @lolorc regarding the unlinked devices in the map - can you try requesting the map just after triggering those missing end devices. If end devices are asleep when the coordinator issues the lqi scan they may be missed. I also sometimes have unlinked end devices on my map.

@clockbrain nope, I've tried with aqara sensors, door sensors and 2 kinds of switches, the devices still appear as unlinked.

clockbrain, 2019-05-13:
@Koenkk I've posted a PR https://github.com/Koenkk/zigbee-shepherd/pull/23 with some changes to lqiscan which make it easier to understand what is going on with map generation.

@mihalski The PR includes a change that swaps the order of map building. Previously it recursed into connected routers before collecting their local links. With this change it collects links before recursing. I don't quite understand how the error handling works with the promise style code but it is possible that previously one error from a router would abort the whole map. This PR may help with that. I don't have very many devices connected in my network so can't test at a large scale.

@lolorc sorry, now that I understand a bit better how the lqiscan builds the map, the advice I gave earlier is wrong. The lqi scan doesn't poll end devices to build the map. It gets all the info it needs from the coordinator and routers. Routers have all the info relating to their associated end devices in neighbor tables so no need to poll end devices.

Koenkk, 2019-05-13:
@clockbrain I will look at it ASAP

clockbrain, 2019-05-19:
@Koenkk  I've had another go at restructuring the network map code. https://github.com/Koenkk/zigbee2mqtt/pull/1543
Testing needed!

Koenkk, 2019-05-19:
@clockbrain  I've merged your PR in the dev branch.

At all: can you test if things have been improved in the dev branch?

milakov, 2019-05-19:
I've just installed the latest version: The map has no connections now. Zero.

bizziebis, 2019-05-19:
Was about to post the same. No connection whatsoever.

clockbrain, 2019-05-19:
@Koenkk looks like its premature to include it in dev. Can you revert that PR for the moment.

@milakov @bizziebis I'll look further into it and then ask you to test again once I add some more debug lines. Got to go to work just now.

clockbrain, 2019-05-20:
@milakov @bizziebis I've updated the PR https://github.com/Koenkk/zigbee2mqtt/pull/1546 with a timeout to handle uncontactable devices which is what I think was preventing your maps from being generated. Are you in a position to test this straight from the code in the PR or do you need @Koenkk to merge it to the dev branch?

milakov, 2019-05-20:
@clockbrain I am using z2m as hass.io add-on, I think the only way for me to check new changes is to have them them in dev branch. I can also sniff zigbee traffic if needed.

clockbrain, 2019-05-20:
@milakov Ok, I guess you will need to wait on dev then. No need to capture zigbee traffic. Running with debug on (DEBUG=zigbee-shepherd* npm start 2>&1 | tee debug.txt) shows the lqi traffic, e.g. this is what I see in my log.
```
2019-05-20T02:02:14.854Z zigbee-shepherd:request REQ --> ZDO:mgmtLqiReq
2019-05-20T02:02:14.858Z zigbee-shepherd:request REQ --> ZDO:mgmtLqiReq
2019-05-20T02:02:14.859Z zigbee-shepherd:request REQ --> ZDO:mgmtLqiReq
2019-05-20T02:02:14.860Z zigbee-shepherd:request REQ --> ZDO:mgmtLqiReq
2019-05-20T02:02:14.899Z zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtLqiRsp
2019-05-20T02:02:14.911Z zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd
2019-05-20T02:02:14.988Z zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtLqiRsp
2019-05-20T02:02:14.995Z zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtLqiRsp
2019-05-20T02:02:15.002Z zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtLqiRsp
2019-05-20T02:02:15.028Z zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd
```
The problem with network map isn't generally the zigbee lqi traffic, its trying to corral all the asynchronous lqi calls back together to build the map. The PR code works fine for me but I do only have a fairly simple network hence asking for broader testing.

lolorc, 2019-05-20:
it is indeed better with the timeout.
with your initial PR, no devices were connected on my cc2652r networkmap (same PR on cc2530 was giving a proper network map)
now it's also ok on cc2652r (more devices)
[image]

clockbrain, 2019-05-21:
@Koenkk thanks for the merge. No, I don't think increasing the timeout is needed. It should return all the links it has gathered before the timeout so at a minimum it would have the direct coordinator links.

If anyone is still having problems with the network map in dev can you add a simple debug line 
```
console.log(result);
```
just before here https://github.com/Koenkk/zigbee2mqtt/blob/e4a50b662d237439bae975422206b38cfbbb868c/lib/zigbee.js#L333
and post the error message.

milakov, 2019-05-22:
@clockbrain Why don't you add debugging in a "normal" way? I updated to the latest dev, zero connections yet.

clockbrain, 2019-05-23:
@Koenkk I've added another PR with extra network map debugging https://github.com/Koenkk/zigbee2mqtt/pull/1559

@milakov yes, I should have included that first time around but I JS isn't my first language (needed to do some quick study). Also, another avenue you could perhaps try is temporarily move your cc2531 and /data to a Windows PC and try to debug from there. zigbee2mqtt runs ok under Windows, See https://github.com/Koenkk/zigbee2mqtt/issues/648

milakov, 2019-05-23:
@clockbrain That's an option! Another one is to modify the file inside the container, could probably work! Will try this evening.

milakov, 2019-05-23:
@clockbrain I added this
```
console.log(result);
```
Where is it logged exactly to? Log doesn't contain any new entries with this.

milakov, 2019-05-23:
@clockbrain Iэму just installed the latest dev with, this is what I see in log:
```
  zigbee2mqtt:info 5/23/2019, 9:20:16 PM Starting network scan...
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Preparing asynch network scan for '0x00124b0019368448'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Scanning device: '0x00124b0019368448'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Preparing asynch network scan for '0x00124b001b50416b'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Scanning device: '0x00124b001b50416b'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Preparing asynch network scan for '0x00124b001b50c0a6'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Scanning device: '0x00124b001b50c0a6'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Preparing asynch network scan for '0x00124b001b505747'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Scanning device: '0x00124b001b505747'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Preparing asynch network scan for '0x00158d00024f2e59'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Scanning device: '0x00158d00024f2e59'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Preparing asynch network scan for '0x00158d0002561266'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM Scanning device: '0x00158d0002561266'
  zigbee2mqtt:debug 5/23/2019, 9:20:16 PM All network map promises created
  zigbee2mqtt:info 5/23/2019, 9:20:16 PM Network scan failed: 'Error: request unsuccess: 132'
  zigbee2mqtt:info 5/23/2019, 9:20:16 PM MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b0019368448" [style="bold", label="{0x00124b0019368448|Coordinator|No model information available|online}"];
  "0x00158d000254cced" [style="rounded, dashed", label="{aqara_double_button_2|EndDevice|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|online}"];
  "0x00158d0001ef8655" [style="rounded, dashed", label="{aqara_button_table|EndDevice|Xiaomi Aqara wireless switch (WXKG11LM)|online}"];
  "0x00158d00024a5018" [style="rounded, dashed", label="{water_leak_bathroom|EndDevice|Xiaomi Aqara water leak sensor (SJCGQ11LM)|online}"];
  "0x00158d0001ef852b" [style="rounded, dashed", label="{aqara_button_cor|EndDevice|Xiaomi Aqara wireless switch (WXKG11LM)|online}"];
  "0x00158d0002720cd1" [style="rounded, dashed", label="{climate_kitchen|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d0002775e09" [style="rounded, dashed", label="{climate_bedroom|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00124b001b50416b" [style="rounded", label="{gledopto_light_bathroom|Router|Gledopto Smart 6W E27 RGB / CW LED bulb (GL-B-007Z)|offline}"];
  "0x00158d000288e3c5" [style="rounded, dashed", label="{door_window_sensor_cor|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x000b57fffe96403a" [style="rounded, dashed", label="{ikea_tradfri_dimmer_2|EndDevice|IKEA TRADFRI wireless dimmer (ICTC-G-1)|offline}"];
  "0x00158d00029c070e" [style="rounded, dashed", label="{motion_sensor_cor|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d00023891e9" [style="rounded, dashed", label="{climate_bathroom|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d000288e3b1" [style="rounded, dashed", label="{door_window_sensor_bedroom|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d000101b949" [style="rounded, dashed", label="{cube_1|EndDevice|Xiaomi Mi/Aqara smart home cube (MFKZQ01LM)|online}"];
  "0x00158d000288e352" [style="rounded, dashed", label="{door_window_sensor_kitchen|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d00024a4e92" [style="rounded, dashed", label="{water_leak_kitchen|EndDevice|Xiaomi Aqara water leak sensor (SJCGQ11LM)|online}"];
  "0x00158d00029c0149" [style="rounded, dashed", label="{motion_sensor_kitchen|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d00027c1b4c" [style="rounded, dashed", label="{aqara_double_button_1|EndDevice|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|online}"];
  "0x00158d00026b76b6" [style="rounded, dashed", label="{xiaomi_button_door|EndDevice|Xiaomi MiJia wireless switch (WXKG01LM)|online}"];
  "0x00124b001b50c0a6" [style="rounded", label="{gledopto_desklamp|Router|Gledopto Smart 6W E27 RGB / CW LED bulb (GL-B-007Z)|offline}"];
  "0x00124b001b505747" [style="rounded", label="{gledopto_light_kitchen|Router|Gledopto Smart 12W E27 RGB / CW LED bulb (GL-B-008Z)|offline}"];
  "0x00158d00026b773d" [style="rounded, dashed", label="{xiaomi_button_kitchen|EndDevice|Xiaomi MiJia wireless switch (WXKG01LM)|online}"];
  "0x00158d000272652f" [style="rounded, dashed", label="{climate_bedroom_outside|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d00024f2e59" [style="rounded", label="{xiaomi_plug_2|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"];
  "0x00158d0002af945e" [style="rounded, dashed", label="{vibration_sensor|EndDevice|Xiaomi Aqara vibration sensor (DJT11LM)|online}"];
  "0x000b57fffe8bb145" [style="rounded, dashed", label="{ikea_tradfri_dimmer_1|EndDevice|IKEA TRADFRI wireless dimmer (ICTC-G-1)|offline}"];
  "0x00158d0002561266" [style="rounded", label="{xiaomi_plug_1|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"];
}'
```

milakov, 2019-05-23:
Here is another one. Please, notice that after some of the routers (here, Gledopto bulb) returned error, z2m generated map with no links, but other routers (Xiaomi Aqara smart plugs) returned links afterwards:
```
  zigbee2mqtt:info 5/23/2019, 9:28:05 PM Starting network scan...
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Preparing asynch network scan for '0x00124b0019368448'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Scanning device: '0x00124b0019368448'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Preparing asynch network scan for '0x00124b001b50416b'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Scanning device: '0x00124b001b50416b'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Preparing asynch network scan for '0x00124b001b50c0a6'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Scanning device: '0x00124b001b50c0a6'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Preparing asynch network scan for '0x00124b001b505747'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Scanning device: '0x00124b001b505747'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Preparing asynch network scan for '0x00158d00024f2e59'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Scanning device: '0x00158d00024f2e59'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Preparing asynch network scan for '0x00158d0002561266'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM Scanning device: '0x00158d0002561266'
  zigbee2mqtt:debug 5/23/2019, 9:28:05 PM All network map promises created
  zigbee2mqtt:info 5/23/2019, 9:28:06 PM Network scan failed: 'Error: request unsuccess: 132'
  zigbee2mqtt:info 5/23/2019, 9:28:06 PM MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b0019368448" [style="bold", label="{0x00124b0019368448|Coordinator|No model information available|online}"];
  "0x00158d000254cced" [style="rounded, dashed", label="{aqara_double_button_2|EndDevice|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|online}"];
  "0x00158d0001ef8655" [style="rounded, dashed", label="{aqara_button_table|EndDevice|Xiaomi Aqara wireless switch (WXKG11LM)|online}"];
  "0x00158d00024a5018" [style="rounded, dashed", label="{water_leak_bathroom|EndDevice|Xiaomi Aqara water leak sensor (SJCGQ11LM)|online}"];
  "0x00158d0001ef852b" [style="rounded, dashed", label="{aqara_button_cor|EndDevice|Xiaomi Aqara wireless switch (WXKG11LM)|online}"];
  "0x00158d0002720cd1" [style="rounded, dashed", label="{climate_kitchen|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d0002775e09" [style="rounded, dashed", label="{climate_bedroom|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00124b001b50416b" [style="rounded", label="{gledopto_light_bathroom|Router|Gledopto Smart 6W E27 RGB / CW LED bulb (GL-B-007Z)|offline}"];
  "0x00158d000288e3c5" [style="rounded, dashed", label="{door_window_sensor_cor|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x000b57fffe96403a" [style="rounded, dashed", label="{ikea_tradfri_dimmer_2|EndDevice|IKEA TRADFRI wireless dimmer (ICTC-G-1)|offline}"];
  "0x00158d00029c070e" [style="rounded, dashed", label="{motion_sensor_cor|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d00023891e9" [style="rounded, dashed", label="{climate_bathroom|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d000288e3b1" [style="rounded, dashed", label="{door_window_sensor_bedroom|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d000101b949" [style="rounded, dashed", label="{cube_1|EndDevice|Xiaomi Mi/Aqara smart home cube (MFKZQ01LM)|online}"];
  "0x00158d000288e352" [style="rounded, dashed", label="{door_window_sensor_kitchen|EndDevice|Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online}"];
  "0x00158d00024a4e92" [style="rounded, dashed", label="{water_leak_kitchen|EndDevice|Xiaomi Aqara water leak sensor (SJCGQ11LM)|online}"];
  "0x00158d00029c0149" [style="rounded, dashed", label="{motion_sensor_kitchen|EndDevice|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online}"];
  "0x00158d00027c1b4c" [style="rounded, dashed", label="{aqara_double_button_1|EndDevice|Xiaomi Aqara double key wireless wall switch (WXKG02LM)|online}"];
  "0x00158d00026b76b6" [style="rounded, dashed", label="{xiaomi_button_door|EndDevice|Xiaomi MiJia wireless switch (WXKG01LM)|online}"];
  "0x00124b001b50c0a6" [style="rounded", label="{gledopto_desklamp|Router|Gledopto Smart 6W E27 RGB / CW LED bulb (GL-B-007Z)|offline}"];
  "0x00124b001b505747" [style="rounded", label="{gledopto_light_kitchen|Router|Gledopto Smart 12W E27 RGB / CW LED bulb (GL-B-008Z)|offline}"];
  "0x00158d00026b773d" [style="rounded, dashed", label="{xiaomi_button_kitchen|EndDevice|Xiaomi MiJia wireless switch (WXKG01LM)|online}"];
  "0x00158d000272652f" [style="rounded, dashed", label="{climate_bedroom_outside|EndDevice|Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online}"];
  "0x00158d00024f2e59" [style="rounded", label="{xiaomi_plug_2|Router|Xiaomi Mi power plug ZigBee (ZNCZ02LM)|online}"];
  "0x00158d0002af945e" [style="rounded, dashed", label="{vibration_sensor|EndDevice|Xiaomi Aqara vibration sensor (DJT11LM)|online}"];
  "0x000b57fffe8bb145" [style="rounded, dashed", label="{ikea_tradfri_dimmer_1|EndDevice|IKEA TRADFRI wireless dimmer (ICTC-G-1)|offline}"];
[... cut ...] 2e59","status":"online"},{"ieeeAddr":"0x00124b001b50c0a6","nwkAddr":58383,"lqi":57,"depth":1,"parent":"0x00158d00024f2e59","status":"offline"},{"ieeeAddr":"0x00124b001b50416b","nwkAddr":30117,"lqi":51,"depth":1,"parent":"0x00158d00024f2e59","status":"offline"},{"ieeeAddr":"0x00158d0002561266","nwkAddr":34561,"lqi":86,"depth":1,"parent":"0x00158d00024f2e59","status":"online"},{"ieeeAddr":"0x00124b001b505747","nwkAddr":52344,"lqi":49,"depth":1,"parent":"0x00158d00024f2e59","status":"offline"}]
```

clockbrain, 2019-05-23:
@milakov thanks, that's really good info.

Regarding logging, when debugging I normally stop the service with sudo systemctl stop zigbee2mqtt then manually run it with DEBUG=* npm start. That shows debug messages on the console so is quicker for me than delving into log files. It also shows richer debug info than just enabling debug logging in configuration.yaml

The problem with your network map is indeed caused by the 'Error: request unsuccess: 132' which means one of the router lqi scans failed and this kills the overall collection of links. I don't know what error 132 means. Will need to add more error handling code to individual scan to intercept the error and turn it back into a fake success. So somewhere around here https://github.com/Koenkk/zigbee2mqtt/blob/c4e9ea208a353112fbbb31b44d03185adf7d9a6e/lib/zigbee.js#L311 or perhaps here https://github.com/Koenkk/zigbee2mqtt/blob/c4e9ea208a353112fbbb31b44d03185adf7d9a6e/lib/zigbee.js#L326 it needs to intercept the error. Will do some more study and testing to see if I can recreate the problem and then deal with it.

In the meantime, if the cause is just one router, or one type of router then you could add temporary code to exclude those from being scanned. I don't have the exact code needed at hand but it would go here as an additional part of the test condition  https://github.com/Koenkk/zigbee2mqtt/blob/c4e9ea208a353112fbbb31b44d03185adf7d9a6e/lib/zigbee.js#L279

Finally, the fact that in your second output some of the link sets are appearing after the map is generated means that the timeout of 2 seconds isn't sufficient. The timeout is triggering and the map generated even though some of the asynch scans are still underway. You could try increasing the timeout - pick a number say 5 seconds. Note that until the other issue with failing scan error 132 is fixed or those problem devices are excluded, an increased timeout won't help.

milakov, 2019-05-24:
@clockbrain sniffer indicates 132 is "Not Supported", that router (Gkedopto  bulb) is a piece of joke. I will try switch ing off that dumb router and trigger map creation.

milakov, 2019-05-24:
Here it is, with failing routers switched off physically:
[image]

clockbrain, 2019-05-24:
@Koenkk I've prepared another PR https://github.com/Koenkk/zigbee2mqtt/pull/1565 with better error handling for the network map. I also increased the timeout to a generous 8 seconds.

@milakov my fingers are crossed that this fixes your problem, i.e. ignores the recalcitrant Gledopto router but builds the map anyway. If not then please share the log extracts again.

milakov, 2019-05-24:
@clockbrain Works fine for me, thank you so much!

milakov, 2019-05-25:
Unfortunately, Xiaomi Aqara smart plugs stop responding to link quality requests. So my map trurn into start with coordinator and some devices (both routers and end-devices) connected directly to the coordinator. I gusss nothing coud be done here.

clockbrain, 2019-05-25:
@milakov No, if they aren't responding the lqi requests then there isn't really any way for their child links to be included in the map. I assume you've tried resetting them and are keeping an eye on other potential routing/firmware issues such as https://github.com/Koenkk/zigbee2mqtt/issues/1536

alpenhase83, 2019-05-26:
I don't know if that helps to catch the issue. But i had a very well working setup with Z2M v1.3.1 and CC2531 firmware 20190223.

After i've upgraded my stick to latest firmware (CC2531_20190425) and Z2M v1.4.0 my OpenHAB2 installation reports many errors for sensors which worked without any issue before.

`Executing the JSONPATH-transformation failed: Invalid path '$.voltage' in '{"illuminance":60,"linkquality":26,"occupancy":true}'`

It seems that the new Z2M v1.4.0 cuts off some of the meta data like voltage, battery, etc.

Z2M v1.4.0 reports this values at a specific intervall but not with every state change.

Reflashing CC2531 firmware 20190223 doesn't work. Is it possible to downgrade to Z2M v1.3.1 without further problems?

clockbrain, 2019-05-29:
@Koenkk seems some of the network scan code transplanted from zigbee-shepherd to zigbee2mqtt isn't behaving. https://github.com/Koenkk/zigbee2mqtt/blob/caff94559ecc942e5b65e7ba3b7f099d1e4eb1cc/lib/zigbee.js#L315
I get this value for every childDev:
```
5/29/2019, 8:28:06 AM - debug: childDev: [Circular]
```
So the status isn't being set properly in the scan. However, for graphviz maps this status value is ignored and replaced when the map is generated from the topology. My proposal would be to simply remove the status from network scan in zigbee.js which would only affect raw maps. Would that be a concern?

Replicating some of the code from graphviz could add the status back into raw maps if it was necessary.

Koenkk, 2019-05-29:
No that's not a concern.

Note that now all network scans are executed simultaneously which can cause a burst of zigbee commands (e.g. when having a network with 20 routers, 20 lqi scans are executed at the same time). I would recommend adding the scan requests to the queue instead (e.g. https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L311). This should make things more reliable.

clockbrain, 2019-05-31:
@Koenkk Ok, I'm starting to have a look at converting the code to use the queue but its tricky as the network scan is built on layers of promise style code and inserting an asynch queue in the middle isn't straightforward. Any chance a random up to 2 second delay for each scan would suffice? That would spread the network requests.

clockbrain, 2019-06-02:
@Koenkk I've posted https://github.com/Koenkk/zigbee2mqtt/pull/1590 using random delayed start times for the network map scans which hopefully avoids network congestion. This is until I can figure out how to properly integrate with the queue system.

dh-harald, 2019-06-07:
With https://github.com/Koenkk/zigbee2mqtt/pull/1590 (dev branch) my network map looks much-much better. Thanks

[25 more replies]

closed 2019-09-15 by stale[bot]