Koenkk/zigbee2mqtt #657: RF4CE profile possible?

TSparkas, 2018-12-05:
Hi,

Is it possible to implement RF4CE profile to control consumer audio/video devices using zigbee2mqtt?
I have Arris IPTV set top box using such RF remote.

https://www.zigbee.org/zigbee-for-developers/network-specifications/zigbeerf4ce/?fbclid=IwAR2oVKwLu13MmqTDUOcC-PUi8KCVGQupGVy4xYTiMycrAW-1nHOG6nYzwTA

Koenkk, 2018-12-05:
Maybe, can you try sniffing the zigbee commands being send? (https://blog.jimmo.id.au/zigbee-packet-capture-724cf9346b47)

TSparkas, 2018-12-05:
I can see packets from remote on channel 20 in the TI packer sniffer program:
[image]

Koenkk, 2018-12-06:
While reading more about RF4CE it seems that RF4CE is a different protocol. 
Currently zigbee-shepherd/zigbee2mqtt is focused towards TI z-stack.

If the current zigbee2mqtt z-stack firmware is able to handle RF4CE it might be possible to add support for it. However, as far as I could find, z-stack is not compatible with RF4CE (http://e2e.ti.com/support/wireless-connectivity/other-wireless/f/667/t/18201)

closed 2019-02-11 by stale[bot]

WilldabeastHA, 2019-03-14:
@TSparkas Were you able to find some work around for sniffing the RF4CE protocol? I want to do the same thing you described. I have an old cable box remote that uses RF4CE, it would be great if I could use that remote to turn on lights and open blinds, etc.

TSparkas, 2019-03-16:
Yes, I was able to sniff RF4CE commands by flashing CC2531 usb stick with TI firmware and using their software. But then I was stuck on how to implement some control over it.