Koenkk/zigbee2mqtt #658: No longer getting any response from Sengled Element bulbs.

dmwyatt, 2018-12-05:
Check out the debug log here: http://termbin.com/p6aw

Right at the end you can see zigbee2mqtt receiving `set` messages but it doesn't do anything with them.  

These worked in the past, but I'm not sure if they stopped working after some update to zigbee2mqtt or not.

Koenkk, 2018-12-06:
Do they work when running the release version (non-dev branch)?

dmwyatt, 2018-12-06:
I kept my data directory and moved to the release version and it's the same problem.

dmwyatt, 2018-12-06:
Should I try with a fresh data directory?

Koenkk, 2018-12-06:
Do they work again when repairing?

dmwyatt, 2018-12-06:
No.

Koenkk, 2018-12-06:
After updating, did you do a

```
rm -rf node_modules
npm i
```

dmwyatt, 2018-12-06:
Well the thing you're not considering is that I'm a huge dummy and I forgot to do that.

After wiping node_modules and doing an `npm i` the latest release version is back to working.  

I'm heading out on vacation shortly but I'll add a reminder to my calendar for when I get back to switch back to the latest version from the repo and double check that things are or are not working there.

closed 2019-02-12 by stale[bot]