Koenkk/zigbee2mqtt #659: is setting via zigbee ensured in any way?

dirstel, 2018-12-06:
I'm using zigbee2mqtt in a Docker-container on an Intel NUC i3 with other containers up and running (mosquitto, node-red, fhem, ...). Most of the time, set-commands (f.e. turn on/off) are executed by a device. Sometimes a device does not executed the specified command. Maybe, my setup (docker, mesh-network, ...) is responsible, but tht's not my point.

Is the execution fo commands ensured in any way by zigbee2mqtt? f.e. set sends zigbee-command and status message is expected, otherwise set command is repeated?

Koenkk, 2018-12-06:
No, there are many reason why a command could fail (e.g. interference, device offline). 

Zigbee already has a built-in retry mechanism, therefore zigbee2mqtt doesn't retry again

dirstel, 2018-12-09:
Ok, I'll have to investigate further on dropped commands. Maybe an other frequency helps. I installed a couple of repeaters (Osram Smart Plug+) and the network maps looks kind of weired. But I assume, that's a problem of the generated map.

closed 2019-02-11 by dirstel

BernieBl22, 2019-07-29:
@dirstel 
I am trying to install Zigbee2MQTT in a Docker container myself, but i have an Intel chip and Ubuntu 18.04, is you setup similar? if so, would you mind sharing a docker-compose of your setup?

dirstel, 2019-08-06:
@BernieBl22 
> I have an Intel chip and Ubuntu 18.04, is you setup similar? if so, would you mind sharing a docker-compose of your setup?

Sorry for late response - holiday season ;-)
I'm using an Intel NUC I3 running Ubuntu 18.04 LTS and Docker CE in the most current version. Most crucial part is assigning the usb-stick a permanent devicename - I did so using an udev-rule

I'll paste some more info when I'm back home.

BernieBl22, 2019-08-06:
Hi Dirk,

Yes, I have an identical setup as yours. I would be very interested in knowing how you were able to run Zigbee2mqtt in a Docker container.  I’ve tried several attempts (for an entire week) and I was unsuccessful.   I ca find my dongle with /dev/serial/by-id/

No rush, I am on vacation myself so will most likely not touch it for 2 weeks

Thanks for writing back 
Bernard

dirstel, 2019-08-06:
I am using a udev rule to create a known device:

```
# Datei: /etc/udev/rules.d/20_usb_cc2531.rules
# Gerät ZigBee-Adapter (cc2531) an USB
SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8", SYMLINK+="ttyUSB_cc2531-1", RUN+="chmod 666 /dev/ttyUSB_cc2531-1"
# Aply low latency only if needed!
#, RUN+="/bin/setserial -q /dev/ttyUSB_cc2531-1 low_latency"
```

This is then used in my docker-compose.yaml:
```
version: '2'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: always
...
    devices:
      - /dev/ttyUSB_cc2531-1
...
```

And referenced in my configuration.yaml (inside the container)
```
...
serial:
  port: /dev/ttyUSB_cc2531-1
...
```

BernieBl22, 2019-08-07:
Thanks Dirk
I will surely give this a shot when I return home from vacation 
Is this your entire docker-compose for Zigbee2mqtt?   No volumes or anything else?

Thanks again
Bernard

dirstel, 2019-08-07:
sure (as indicated by the "..."), but nothing really special. and nothing which will help you find your device. afair:
- timezone settings
- network settings
- volume (zigbee2mqtt-config path) mapped to local path

maybe the user running the (inside) the container is not allowed to access your device?