Koenkk/zigbee2mqtt #660: Time line for next rls ? Osram plug and Ikea e27 crasches!

JLFN, 2018-12-06:
Hey ,
I have big troubles with my routers osram plug an also Ikea e27 i cant use them i think its before of the extra device information. Please help me out with that rows that have been change so i can change on mine 0.20 .
 I already using the newest zigbee-shepherd-converters from devbranch.
Thank you in advance.

Koenkk, 2018-12-06:
Until the next version is released, you can use the dev branch.

JLFN, 2018-12-06:
@Koenkk  Cant i make this False ? 
```.js 
        if (appSettings.mqtt.include_device_information) {
            messagePayload.device = this.getDeviceInfoForMqtt(device);
        }

        this.mqtt.publish(friendlyName, JSON.stringify(messagePayload), options);
    }
```

closed 2018-12-06 by JLFN