Koenkk/zigbee2mqtt #661: Zigbee2mqtt container at 100% cpu

westy1969, 2018-12-06:
Hello i'm trying to get the zigbee2mqtt docker online.

- Raspberry pi 3b+
- latest docker koenkk/zigbee2mqtt:arm32v6
- docker starts, but with 100% cpu, after a while the pi freezes up.

its seems like it has issues with the ttyACM0, the pi uses ttyAMA0

docker-compose.yaml
`  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:arm32v6
    links:
      - mqtt
    volumes:
      - /opt/zigbee2mqtt:/app/data
      - /etc/localtime:/etc/localtime:ro
      - /etc/timezone:/etc/timezone:ro
    devices:
      - /dev/ttyAMA0:/dev/ttyACM0`

configuration.yaml
`# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://10.0.1.4:1883'
  # MQTT server authentication, uncomment if required:
  user: xxx
  password: xxxx

# Serial settings
serial:
  #  Location of CC2531 USB sniffer
  port: /dev/ttyACM0`

any advice is welcome

Koenkk, 2018-12-07:
Does this also happen when running without docker?

westy1969, 2018-12-18:
Thanks for the reply, was my bad, i used the router instead of the coordinator.

closed 2018-12-18 by westy1969