Koenkk/zigbee2mqtt #662: RTCGQ11LM not recogniced

BudBundi, 2018-12-06:
I'm new to zigbee2mqtt have flash the stick and start the docker container on my raspberry.
Then I have tried to pair 2 RTCGQ11LM the log say it is paired but the device is not identified 
New device with address 0x00158d0002b95d67 connected!
MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d0002b95d67"}'
Device incoming...
MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'

((0x00158d0002b95d67): unkown - undefined unknown (EndDevice))

Link to log: https://pastebin.com/QTcXVjAX
Link to the database.db content: https://pastebin.com/t9G1Q9E5

Is there some thing I have overlooked. Some setting or so?

BudBundi, 2018-12-06:
OK after deleting the database and pair again it seems to work

closed 2018-12-06 by BudBundi

phoxy84, 2018-12-10:
I have the poblem again on Hassio 0.83.3 and ZigbeeToMQTT edge addon. Uninstalled the addon and installed again gives the same problem.

`zigbee2mqtt:info 2018-12-10 17:31:19 0x00158d0001a3e25e (0x00158d0001a3e25e): unkown - undefined unknown (EndDevice)`

It is the RTCGQ11LM; Xiaomi Aqara human body movement and illuminance sensor (occupancy and illuminance)