Koenkk/zigbee2mqtt #665: Request: Support TCP Connections to CC2530

Eistee82, 2018-12-07:
Hi is it posible to support TCP Connections to CC2530 so i can connect it to an ESP8266 or an MapleCUN.

ryanbeaton, 2018-12-07:
Info in issue 442

https://github.com/Koenkk/zigbee2mqtt/issues/442#issuecomment-426775612

JLFN, 2018-12-07:
@Eistee82  
Can you mount the port with Serial2net to Socat 
```.bash
/usr/bin/socat -d -d pty,raw,echo=0,link=/dev/ttyACM1 tcp-connect:192.168.1.4:1775
```
https://github.com/Koenkk/zigbee2mqtt/issues/442#issuecomment-426994483
Please give me feedback how the latency is.

JLFN, 2018-12-07:
@Eistee82  

Wiring:
ESP8266 -- CC2530
3v3 -- VCC
GND -- GND
TX -- P02
RX -- P03
GND -- P20
GND -- P04
GND -- P05

Esp8266 flashing
Press the Flashmode button when connection to the computer

Downlad the lastest https://github.com/letscontrolit/ESPEasy/releases
Extract the ESPMega zip file, and open ‘FlashESP8266.exe’
Select the right COM port and select the right Firmware (ESP_Easy_mega-XXXXXXXX_normal_ESP8266_4096.bin)

When it is ready close the flashing tool
The ESP8266 will now emit a wifi signal, connect with it with the following password: ‘configesp’ (more information at https://www.letscontrolit.com/wiki/index.php/ESPEasy#Introduction)
After connection a screen opens in which you can let the ESP8266 connect to your WIFI, if succesfull its new IP address will be shown.
Go to this IP address in your browser and click on devices
or Go to http://192.168.4.1/setup
Click on "Devices" Edit of the first task and select ‘Communication - Serial Server’ from the dropdown list
Fill in the form as following:
a.	Name: ZIGBEE2MQTT
b.	Enabled: checked
c.	TCP Port: a number between 1000 and 9999 "1775"
d.	Baud Rate: 115200
e.	Data bits: 8
f.	Parity: No Parity
g.	Stop bits: 1
h.	Reset target after boot: - none –
i.	RX receive timeout: 0
j.	Event processing: Generic
Press Submit

Then its complete the device will get devicename ESP-Easy-0 check your router for the IP or you will get directed after the setup of the Wifi the first time you connected to the device.

You can also try to use ser2net on Linux then use that port.
https://www.letscontrolit.com/wiki/index.php/Ser2Net
Mount with socat
```.bash
socat -d -d pty,raw,echo=0,link=/dev/ttyACM1 tcp-connect:192.168.1.4:1775
```
**NOTE THIS!** Make sure the zigbee2mqtt is not using the same channel as WIFI.
Please give me feedback how the latency is.

rmounce, 2018-12-10:
I have been using a similar configuration to this for months, but using an OpenWrt AP with USB connected CC2531 instead of ESP8266 with UART connected CC2530.

I am using ser2net on the OpenWrt remote radio side, and socat on the zigbee2mqtt server side. The server side is a bit complicated with an extra systemd unit for socat and dependencies etc. Despite this complexity and to my surprise, it has been working flawlessly.

Allowing direct specification of a TCP endpoint in zigbee2mqtt would greatly simplify this case, which is what I believe Eistee is requesting.

e.g. rather than
```
serial:
  port: /dev/ttyACM0
```
configuration could be like
```
tcp:
  host: 192.168.1.2:5000
```

This would require adding TCP support to zigbee-shepherd (bypassing serialport) and updating zigbee2mqtt to pass through the configuration.

rmounce, 2018-12-10:
@JLFN 

My systemd unit files for socat and zigbee2mqtt
https://gist.github.com/rmounce/16a8625755d4491cdb96a697742f1535
https://gist.github.com/rmounce/021405a0f8f409984c3d6b6f52e69453

And the relevant part of my zigbee2mqtt configuration.yaml
```
serial:
  port: /tmp/ttyVUSB0
advanced:
  rtscts: false
```

JLFN, 2018-12-18:
@rmounce  I think this is what we are looking for https://www.npmjs.com/package/tcp-serial

JLFN, 2019-01-16:
Any update? Its a really nic function i need feedback with the respond time .

JLFN, 2019-02-06:
Solved check docs.

Koenkk, 2019-02-06:
https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_esp8266_with_cc2530.html thanks to @JLFN

closed 2019-02-06 by Koenkk