Koenkk/zigbee2mqtt #668: (BREAKING CHANGE) Zigbee2mqtt 1.0 will break Home Assistant device_ids

Koenkk, 2018-12-07:
**NOTE: Only applies when you used MQTT discovery**

Zigbee2mqtt (to be released) could change the current `device_id` of your Zigbee2mqtt devices discovered via MQTT discovery.

### Why?
This breaking change is required in order to integrate with the [Home Assistant Device Registry](https://developers.home-assistant.io/docs/en/device_registry_index.html). In order to support this, Zigbee2mqtt always has to provide a `unique_id`. In Zigbee2mqtt < 1.0 the `unique_id` was only send when no `friendly_name` was set (which is the case when you initially pair the device). As Home Assistant only determines the `device_id` the first time it is registered, the new `device_id` of your device will look like: `0x00158d0001d82999_occupancy`.

### What are the advantages?
The device registry has several advantages. When a device is registered by the device registry:
- You can change Home Assistants `friendly_name` and `device_id` via the web interface without restarting Home Assistant.
- You can show which entities belong to which device.

[image]
[image]

### What do I have to do?
Fix the `device_id` of your devices via the Home Assistant web interface.

### I'm confused about the different device IDs, names and friendly names
- Home Assistant `device_id`: determined on first discovery of the device, can only be changed via the Home Assistant web interface afterwards. Used to control/read the state from the device (e.g. in automations)
- Zigbee2mqtt `friendly_name`: used to change the MQTT topic where the device listens and publishes to.
- Home Assistant `name`: name shown in the Home Assistant UI (unless overridden via a `friendly_name` in `customize.yaml`). If not changed via the Home Assistant web interface, it is equal to the Zigbee2mqtt `friendly_name`. Is updated if the Zigbee2mqtt `friendly_name` changes (requires restart of Home Assistant)
- Home Assistant `friendly_name` (`customize.yaml`): overrides the name in the Home Assistant web interface.

jwillmer, 2018-12-19:
I'm confused. I had this in my zigbee2mqtt settings:

```
  '0x00158d....9f':
    friendly_name: mijia_socket_switch
```

And now HomeAssistant detects it as:

```
switch.0x00158d....9f_switch | off | friendly_name: mijia_socket_switch_switch
```

Why is the name (id) not `mijia_socket_switch_switch` as before? Is it not possible anymore to define a (unique) name outside of home assistant?

Koenkk, 2018-12-19:
@jwillmer only upon first discover (which is probably done when you pair the device)

bbrendon, 2018-12-21:
So we need to re-pair everything to get the software to adopt using the friendly name in the data/configuration? There isn't another way?

Koenkk, 2018-12-21:
@bbrendon that won't help, as the devices are already known by home assistant. You have to change the names in home assistant itself.

IceEyz, 2018-12-23:
@Koenkk I think it would help if you'd replace `device_id` with `entity_id` in the instruction above.

@MikeVRT, I don't really understand what your issue is, but after the upgrade, all you need to do is fix any `entity_ids` under Integrations->MQTT. Click any entity there, click the cog icon at the top right, and rename the `entity_id` to what it was before the upgrade, i.e. the friendly_name of this entity in the zigbee2mqtt configuration.

pejotigrek, 2018-12-23:
I changed entity_id via integrations, so my sensor showed up again in the dashboard, like before, using my earlier templates.

but now it's "entity unavailable" at the integrations menu, and when I want to change it's entity_id again - I have no option to do that, right? how can I fix it?

MikeVRT, 2018-12-24:
Hm, my core.entity_registry looks like:

>             {
>                 "config_entry_id": null,
>                 "device_id": null,
>                 "disabled_by": null,
>                 "entity_id": "sensor.0x001dkfjkdsfj882c2_temperature",
>                 "name": null,
>                 "platform": "mqtt",
>                 "unique_id": "0x.0x001dkfjkdsfj882c2_temperature_zigbee2mqtt"
>             },

So no capital letters, but Integrations->MQTT is empty now - as I understood, cards in that place from core.device_registry not from core.entity_registry

Koenkk, 2018-12-24:
@IceEyz unique_id != device_id. unique_id is an identifier which doesn't change, device_id something that can be changed.

IceEyz, 2018-12-25:
The IDs that for me needed fixing in the webgui were the entity_ids, not
the device_ids.

Koenkk, 2018-12-26:
@IceEyz thanks, updated the OP.

MikeVRT, 2018-12-27:
Still can't solve problem - Integrations->MQTT is empty
In HASS's log one warning:

> WARNING (MainThread) [hbmqtt.mqtt.protocol.handler] BrokerProtocolHandler Unhandled exception in reader coro: IncompleteReadError('0 bytes read on a total of 1 expected bytes',)

In zigbee2mqtt's log nothing unusual.

jarrah31, 2018-12-27:
Just in case it helps anyone else, I upgraded to 1.0.1 without having to change any entity names within the HA GUI.

- I use MQTT discovery within HA.
- Friendly names set within Z2M configuration.yaml
- A door/window sensor name appears as _binary_sensor.window_study_main_ both before and after the upgrade to 1.0.1

Koenkk, 2018-12-27:
@jwillmer the only wat to do that is to stop homeassistant and remove `.storage/core.entity_registry`

MikeVRT, 2018-12-28:
> only wat to do that is to stop homeassistant and remove `.storage/core.entity_registry`

I removed all '.storage' folder and it's didn't help: Integrations->MQTT This integration has no devices.
After restart HASS in zugbee2mqtt I see this:

> 2018-12-28 19:25:26 - error: Not connected to MQTT server!
> 2018-12-28 19:25:34 - info: Connected to MQTT server
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/bedroom_thp', payload '{"temperature":20.65,"linkquality":13,"humidity":23.11,"pressure":989,"battery":99,"voltage":3015}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/repeater_cc2530', payload '{"state":false,"linkquality":13}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/cabinet_window', payload '{"battery":99,"voltage":3005,"linkquality":0,"contact":true}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/home_door', payload '{"contact":true,"linkquality":42,"battery":100,"voltage":3045}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/hallway_smoke', payload '{"battery":100,"voltage":3055,"linkquality":110}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/bathroom_leak', payload '{"battery":99,"voltage":3015,"linkquality":47}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/cabinet_thp', payload '{"humidity":18.77,"linkquality":0,"pressure":989.54,"temperature":21.6,"battery":99,"voltage":3005}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'zigbee2mqtt/hall_thp', payload '{"temperature":24.55,"linkquality":47,"pressure":989,"humidity":17.45,"battery":99,"voltage":3005}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002322792/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/bedroom_thp","availability_topic":"zigbee2mqtt/bridge/state","name":"bedroom_thp_temperature","unique_id":"0x00158d0002322792_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002322792","name":"bedroom_thp","sw_version":"Zigbee2mqtt 1.0.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"}}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002322792/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/bedroom_thp","availability_topic":"zigbee2mqtt/bridge/state","name":"bedroom_thp_humidity","unique_id":"0x00158d0002322792_humidity_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002322792","name":"bedroom_thp","sw_version":"Zigbee2mqtt 1.0.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"}}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002322792/pressure/config', payload '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/bedroom_thp","availability_topic":"zigbee2mqtt/bridge/state","name":"bedroom_thp_pressure","unique_id":"0x00158d0002322792_pressure_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002322792","name":"bedroom_thp","sw_version":"Zigbee2mqtt 1.0.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"}}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'homeassistant/binary_sensor/0x00124b001b7b18fa/router/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.state }}","device_class":"connectivity","state_topic":"zigbee2mqtt/repeater_cc2530","availability_topic":"zigbee2mqtt/bridge/state","name":"repeater_cc2530_router","unique_id":"0x00124b001b7b18fa_router_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00124b001b7b18fa","name":"repeater_cc2530","sw_version":"Zigbee2mqtt 1.0.1","model":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)","manufacturer":"Custom devices (DiY)"}}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'homeassistant/sensor/0x00124b001b7b18fa/linkquality/config', payload '{"value_template":"{{ value_json.linkquality }}","json_attributes":["description","type","rssi"],"state_topic":"zigbee2mqtt/repeater_cc2530","availability_topic":"zigbee2mqtt/bridge/state","name":"repeater_cc2530_linkquality","unique_id":"0x00124b001b7b18fa_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00124b001b7b18fa","name":"repeater_cc2530","sw_version":"Zigbee2mqtt 1.0.1","model":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)","manufacturer":"Custom devices (DiY)"}}'
> 2018-12-28 19:25:34 - info: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d000236fd86/contact/config', payload '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","state_topic":"zigbee2mqtt/cabinet_window","availability_topic":"zigbee2mqtt/bridge/state","name":"cabinet_window_contact","unique_id":"0x00158d000236fd86_contact_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d000236fd86","name":"cabinet_w [... cut ...]
> 2018-12-28 19:27:01 - info: MQTT publish: topic 'zigbee2mqtt/repeater_cc2530', payload '{"state":false,"linkquality":13}'

What mean topics starts with 'homeassistant/'? If them special for HASS why still "This integration has no devices."?

My zigbee2mqtt config about HASS:

> homeassistant: true
> permit_join: true
> mqtt:
>   base_topic: zigbee2mqtt
>   server: 'mqtt://localhost'

In HASS cfg:

> mqtt:
>   discovery: true
>   birth_message:
>     topic: 'hass/status'
>     payload: 'online'
>   will_message:
>     topic: 'hass/status'
>     payload: 'offline'

MikeVRT, 2018-12-28:
p.s. here was some similar: https://github.com/Koenkk/zigbee2mqtt/issues/468

Koenkk, 2018-12-28:
@MikeVRT are you running hassio?

Koenkk, 2018-12-29:
@mikevrt looking at the log, everything from the zigbee2mqtt side seems to be going well. Do you see anything in the home assistant log?

MikeVRT, 2018-12-29:
@Koenkk only one warning:
`
WARNING (MainThread) [hbmqtt.mqtt.protocol.handler] BrokerProtocolHandler Unhandled exception in reader coro: IncompleteReadError('0 bytes read on a total of 1 expected bytes',)
`
p.s. for sure I deleted '.storage' - HASS recreated files inside but nothing changed - Integrations->MQTT empty with 'This integration has no devices.' message.

MikeVRT, 2018-12-29:
I found issue with device_id = null as in my preview post https://github.com/Koenkk/zigbee2mqtt/issues/668#issuecomment-449677460
But in my case null not only device_id

Koenkk, 2018-12-29:
Does home assistant read anything from MQTT? (e.g. when you create an mqtt sensor manually)

MikeVRT, 2018-12-30:
@Koenkk I made rollback to 0.77.x + 0.2.0 (it's possible because I use virtual machine with snapshots). All works. Than updated again to 0.84.6 + 1.0.1. Befor update I removed all manual sensors temporarily (they was for each sensor's battery and for pressure in mmhg)  and clean customize.yaml from manual friendly_name(s).

After fist start zigbee2mqtt and hass I saw that in integration: https://ibb.co/3SSbHkN
All looks ok. Sensors clickable and I can change name and friendly name. After change each sensor you will got "(entity unavailable)" https://ibb.co/19TJ82p  but it's known issue - page refresh will helps and you see new naming.

**But... big problem.** All it **works after first start**. You see sensors on overview. If you will rename it in gui - you will see it in overview with new names. But after restart whole system/hass sensors will disappear from overview and still visible in intergration as not clickable.

I'm not sure that it's zigbee2mqtt error of cause, @Koenkk - don't worry about it :)

horchi, 2018-12-31:
Since the update it does not work. When I try to change the entity_is I get "save failed: Entity is already registered"
Any idea?

horchi, 2018-12-31:
The other problem is that I can't switch the light anymore, I try to switch it without changing the Name just by clicking the new appeared lights at the web interface I get this in the zigbee2mqtt log:
`2018-12-31 12:50:22 - debug: Received MQTT message on 'zigbee2mqtt/0x000b57fffec03383/set' with data '{"state": "ON"}'` 
and nothing happend

horchi, 2018-12-31:
Now I backuped the whole data of home assistant and zigbee2mqtt and started wit a new plain installation. The zigbee Temp Sensors and the switches are working after paaring but none of the lights are working, the paaring  of the bulbs succeeded but I can't switch them off/on. 
This is still the only masses when try to switch the light by web interface.
`2018-12-31 14:05:42 - debug: Received MQTT message on 'zigbee2mqtt/0x000b57fffec03383/set' with data '{"state": "ON"}'`

MikeVRT, 2018-12-31:
@horchi, read this comment https://github.com/Koenkk/zigbee2mqtt/issues/771#issuecomment-450489804 about lights

MikeVRT, 2018-12-31:
I Reflashed CC2531 with latest fw, and got undefined+unsupported device again (on clear system):

```
2018-12-31 17:17:08 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2018-12-31.17-17-08'
2018-12-31 17:17:20 - info: Starting zigbee2mqtt version 1.0.1 (commit #9380bbc)
2018-12-31 17:17:20 - info: Starting zigbee-shepherd
2018-12-31 17:17:21 - info: zigbee-shepherd started
2018-12-31 17:17:21 - info: Coordinator firmware version: '20181024'
2018-12-31 17:17:21 - info: Currently 0 devices are joined:
2018-12-31 17:17:21 - warn: `permit_join` set to  `true` in configuration.yaml.
2018-12-31 17:17:21 - warn: Allowing new devices to join.
2018-12-31 17:17:21 - warn: Set `permit_join` to `false` once you joined all devices.
2018-12-31 17:17:21 - info: Zigbee: allowing new devices to join.
2018-12-31 17:17:21 - info: Connecting to MQTT server at mqtt://localhost
2018-12-31 17:17:22 - info: zigbee-shepherd ready
2018-12-31 17:17:22 - info: Connected to MQTT server
2018-12-31 17:17:22 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2018-12-31 18:25:25 - info: New device with address 0x00158d00023235c1 connected!
2018-12-31 18:25:25 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d00023235c1"}'
2018-12-31 18:25:25 - warn: Device with modelID 'undefined' is not supported.
2018-12-31 18:25:25 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2018-12-31 18:25:25 - warn: Device with modelID 'undefined' is not supported.
2018-12-31 18:25:25 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2018-12-31 18:25:25 - warn: Device with modelID 'undefined' is not supported.
2018-12-31 18:25:25 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
2018-12-31 18:25:28 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00023235c1', payload '{"temperature":22.95,"linkquality":155}'
2018-12-31 18:25:28 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00023235c1/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d00023235c1","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d00023235c1_temperature","unique_id":"0x00158d00023235c1_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023235c1","name":"0x00158d00023235c1","sw_version":"Zigbee2mqtt 1.0.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"}}'
2018-12-31 18:25:28 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00023235c1/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d00023235c1","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d00023235c1_humidity","unique_id":"0x00158d00023235c1_humidity_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023235c1","name":"0x00158d00023235c1","sw_version":"Zigbee2mqtt 1.0.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"}}'
2018-12-31 18:25:28 - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00023235c1/pressure/config', payload '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","json_attributes":["linkquality","battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d00023235c1","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d00023235c1_pressure","unique_id":"0x00158d00023235c1_pressure_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023235c1","name":"0x00158d00023235c1","sw_version":"Zigbee2mqtt 1.0.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"}}'
2018-12-31 18:25:28 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00023235c1', payload '{"temperature":22.95,"linkquality":155,"humidity":22.11}'
2018-12-31 18:25:28 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00023235c1', payload '{"temperature":22.95,"linkquality":155,"humidity":22.11,"pressure":999}'
```
It's simple WSDCGQ11LM (Aqara Temperature Humidity Sensor) thats works befor without problems on zigbee2mqtt 0.1.x-0.2.x

MikeVRT, 2018-12-31:
Possible I found reason. I don't know why but mqtt's without password caused all this bugs...

Koenkk, 2019-01-01:
@MikeVRT it seems it was an MQTT issue then. A new router firmware is not needed, however when reflashing the coordinator you need to repair all your device, including the CC2530 router, which can only be done reflashing the CC2530 router.

MikeVRT, 2019-01-01:
> which can only be done reflashing the CC2530 router
oh... yes, you are right! I forget about it :(

horchi, 2019-01-02:
It's running again :). After furthermore rebuild (npm) and new pairing of all the bulbs it is working now.

closed 2019-03-03 by Koenkk

stoinov, 2019-03-04:
Sorry I missed the fun discussion last year, but just wanted to understand the reasoning behind having `unique_id` be like `0x00158d0002321010_pressure_zigbee2mqtt`?
Why is the `base_topic` added at the end? Is there a case where you might have overlap of devices that use the same unique ID of the ZigBee device and the specified item that is being reported outside of ZigBee2MQTT? I really can't imagine it.

I'm just asking since I am using somewhat of a complicated naming scheme and the `base_topic` is like `place/location/position` which adds slashes to the name and apart from being unsightly, could cause problems down the road if it's not escaped properly.