Koenkk/zigbee2mqtt #671: Pairing Hue motin sensor

frankiej911, 2018-12-08:
Hello,

Yesterday I've installed the zigbee2mqtt on my server with the recommended hardware. I've successfully connected an Ikea SmartPlug and an Ikea Tradfri Bulb. I've also successfully integrated zigbee2mqtt with my Home Assistant. 

But now I'm trying to pair the Philips Hue Motion Sensors, and I'm stuck right now. First I removed my Motion Sensor with the Hue app and reset the Motion Sensor with a paperclip on the backside, so it's not paired with my Hue Hub anymore. The red led is blinking slowly, but nothing happened in the log of Zigbee2mqtt. Discovery mode is turned on and I've disconnected the power of the Hue Hub and other bulbs paired to my Hue.

Logfile:
https://pastebin.com/4miLXHTH

Thanks!

Koenkk, 2018-12-09:
When running with `DEBUG=* npm start` do you see anything when starting the pairing procedure?

frankiej911, 2018-12-09:
I've tried the pairing process with debugging on. Nothing showed up there: http://oi65.tinypic.com/o56o45.jpg
Also tried another Hue Motion sensor, but unfortunately the same result. Pairing the Motion Sensors with my Hue Hub once again is successful. Green led blinking and the Hue app recognized the device. :(

Hue App:
Type: SML001 - Firmware version; 6.1.0.18912.

Koenkk, 2018-12-10:
@TheCellMC as you added support for this device, could you share the pairing procedure?

TheCellMC, 2018-12-10:
Thats awkward. All I had to do was start discovery mode, reset my sensor until the orange led was blinking and wait for it to connect.

frankiej911, 2018-12-10:
My apologies, I've just solved my issue. It was was related to some wrong permissions for the ttyACM0 interface and the different users who've start the zigbee2mqtt program in the past. The users have broke some permissions with logfiles, so the right user was unable to start the program. With chmod -R and adding the right user to the dialout group solved my issue and the Motion sensor is paired successful right now.

Thanks!

closed 2018-12-10 by Koenkk