Koenkk/zigbee2mqtt #672: basic install /configuration, not able to join devices

bartve66, 2018-12-08:
Yesterday I have installed and configurated a CC2531 USB following the Getting started howto in the wiki. I am running Domoticz on a Raspberry 3B with Raspbian Stretch (that runs well, no issues)
All went pretty good, Zigbee bridge, MQTT server is started on boot.

Now I am trying to join a Xiaomi Aqara motion sensor without any luck. In the configuration file permit_join is set to true, but when I request the status of the service the last line says:
Zigbee: disabling joining new devices.
So something (?) is disabling joining devices *after* it is set to enabled in the configuration file?

The service status:
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-12-08 12:11:47 CET; 11min ago
 Main PID: 412 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─412 npm
           ├─536 sh -c node index.js
           └─538 node index.js

Dec 08 12:13:16 pidomoticz npm[412]:   zigbee2mqtt:info 2018-12-8 12:13:16 zigbee-shepherd ready
Dec 08 12:13:16 pidomoticz npm[412]:   zigbee2mqtt:info 2018-12-8 12:13:16 Connected to MQTT server
Dec 08 12:13:16 pidomoticz npm[412]:   zigbee2mqtt:info 2018-12-8 12:13:16 MQTT publish, topic: 'zigbee2mqtt/bridge/state
Dec 08 12:13:16 pidomoticz npm[412]:   zigbee2mqtt:debug 2018-12-8 12:13:16 Soft reset timeout disabled
Dec 08 12:13:16 pidomoticz npm[412]:   zigbee2mqtt:debug 2018-12-8 12:13:16 Received MQTT message on 'zigbee2mqtt/bridge/
Dec 08 12:13:16 pidomoticz npm[412]:   zigbee2mqtt:info 2018-12-8 12:13:16 MQTT publish, topic: 'zigbee2mqtt/bridge/log',
Dec 08 12:13:16 pidomoticz npm[412]:   zigbee2mqtt:debug 2018-12-8 12:13:16 Received MQTT message on 'zigbee2mqtt/bridge/
Dec 08 12:13:16 pidomoticz npm[412]:   zigbee2mqtt:info 2018-12-8 12:13:16 Zigbee: disabling joining new devices.
Dec 08 12:17:11 pidomoticz npm[412]:   zigbee2mqtt:debug 2018-12-8 12:17:11 Saving state to file /opt/zigbee2mqtt/data/st
Dec 08 12:22:11 pidomoticz npm[412]:   zigbee2mqtt:debug 2018-12-8 12:22:11 Saving state to file /opt/zigbee2mqtt/data/st

Some Zigbee related lines I can find in the Domticz log after a reboot (which I noobly suppose shows there is a connection between Domticz and Zigbee through MQTT?), look at the line before the last line:
2018-12-08 12:11:52.615 (Zigbee MQTT) Connect directive received, action initiation failed.
2018-12-08 12:11:52.615 (Zigbee MQTT) Failed to connect to: 127.0.0.1:1883, Description: resolve: Host not found (authoritative)
2018-12-08 12:11:52.614 Error: Plugin: Connection Exception: 'resolve: Host not found (authoritative)' connecting to '127.0.0.1:1883'
[... non Zigbee related lines deleted]
2018-12-08 12:12:02.243 (Zigbee MQTT) Zigbee2mqtt bridge is offline
[... non Zigbee related lines deleted]
2018-12-08 12:13:16.335 (Zigbee MQTT) Zigbee2mqtt bridge is online
2018-12-08 12:13:16.335 (Zigbee MQTT) Joining new devices is disabled on the zigbee bridge
2018-12-08 12:13:16.386 (Zigbee MQTT) Received available devices list from bridge

Log file: https://pastebin.com/U1SA0Xah

Any idea? Many thanks in advance.

bartve66, 2018-12-08:
And with some help from the Domoticz community I solved my own problem: I forgot to enable pairing in the 'Zigbee MQTT' hardware configuration in Domoticz setup.
Sorry for the fuzz, but I now can go on playing with this great tool!

Koenkk, 2018-12-09:
Great!

closed 2018-12-09 by Koenkk