Koenkk/zigbee2mqtt #677: Trust ZCTS-808 / ZPIR-8000

arcsyst, 2018-12-09:
I'm trying to get the two above devices to work without success. I have an Aqara temp/humidity sensor which works fine. Bothe the trust devices seem to pair correctly but never generate any 'events' when operated. Viewing the log in debug shows:

  zigbee2mqtt:debug 2018-12-9 18:50:28 Received zigbee message of type 'devInterview' with data '"0x00158d0001cc9439"'
  zigbee2mqtt:debug 2018-12-9 18:50:28 Received zigbee message of type 'devIncoming' with data '"0x00158d0001cc9439"' of device 'VMS_ADUROLIGHT' (0x00158d0001cc9439)
  zigbee2mqtt:info 2018-12-9 18:50:28 Device incoming...
  zigbee2mqtt:info 2018-12-9 18:50:28 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 2018-12-9 18:50:28 Received zigbee message of type 'devStatus' with data '"online"' of device 'VMS_ADUROLIGHT' (0x00158d0001cc9439)

then nothing.  
Any help appreciated.

arcsyst, 2018-12-09:
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/2661024/database.zip)

Koenkk, 2018-12-12:
Can you try adding this to devices.js:
```js
    {
        zigbeeModel: ['VMS_ADUROLIGHT'],
        model: 'ZPIR-8000',
        vendor: 'Trust',
        description: 'Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [ ],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
            execute(device, actions, callback);
        },
    },
```

arcsyst, 2018-12-13:
Thank you for your reply, now after detection I get the following output:

  zigbee2mqtt:debug 2018-12-13 16:37:53 Received zigbee message of type 'devInterview' with data '"0x00158d0001cc9439"'
  zigbee2mqtt:debug 2018-12-13 16:37:53 Received zigbee message of type 'devIncoming' with data '"0x00158d0001cc9439"' of device 'VMS_ADUROLIGHT' (0x00158d0001cc9439)
  zigbee2mqtt:info 2018-12-13 16:37:53 Device incoming...
  zigbee2mqtt:info 2018-12-13 16:37:53 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 2018-12-13 16:37:53 Received zigbee message of type 'devStatus' with data '"online"' of device 'VMS_ADUROLIGHT' (0x00158d0001cc9439)
  zigbee2mqtt:info 2018-12-13 16:37:54 Succesfully configured 0x00158d0001cc9439 0x00158d0001cc9439
  zigbee2mqtt:debug 2018-12-13 16:37:54 Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneStatus":32,"iasCieAddr":"0x00124b0018ed35a7","zoneId":23}}' of device 'VMS_ADUROLIGHT' (0x00158d0001cc9439)
  zigbee2mqtt:warn 2018-12-13 16:37:54 No converter available for 'ZPIR-8000' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneStatus":32,"iasCieAddr":"0x00124b0018ed35a7","zoneId":23}}'
  zigbee2mqtt:warn 2018-12-13 16:37:54 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-13 16:37:55 Received zigbee message of type 'devChange' with data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneStatus":32,"iasCieAddr":"0x00124b0018ed35a7","zoneId":23}}' of device 'VMS_ADUROLIGHT' (0x00158d0001cc9439)
  zigbee2mqtt:warn 2018-12-13 16:37:55 No converter available for 'ZPIR-8000' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneStatus":32,"iasCieAddr":"0x00124b0018ed35a7","zoneId":23}}'
  zigbee2mqtt:warn 2018-12-13 16:37:55 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-12-13 16:37:56 Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":34}' of device 'VMS_ADUROLIGHT' (0x00158d0001cc9439)
  zigbee2mqtt:warn 2018-12-13 16:37:56 No converter available for 'ZPIR-8000' with cid 'ssIasZone', type 'statusChange' and data '{"cid":"ssIasZone","zoneStatus":34}'
  zigbee2mqtt:warn 2018-12-13 16:37:56 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.

ekaats, 2018-12-13:
Awhile ago I've also tried to add this sensor, and am giving it a shot once again.
With Koen's suggestion I am able to connect to it, but I don' t get any meaningful information from the device:

```
Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":32}' of device 'VMS_ADUROLIGHT' (0x00158d0001cc708f)
No converter available for 'ZPIR-8000' with cid 'ssIasZone', type 'statusChange' and data '{"cid":"ssIasZone","zoneStatus":34}'
```
If I add gz.generic_occupancy to fromZigbee, I get a ' failed to configure'  error from Zigbee2mqtt.

The device does seem to be connected as the LED on the device is periodically flashing. Is there any way to have it send a dump or something?

Koenkk, 2018-12-14:
Looks good, this should be it:
```js
    {
        zigbeeModel: ['VMS_ADUROLIGHT'],
        model: 'ZPIR-8000',
        vendor: 'Trust',
        description: 'Motion Sensor',
        supports: 'occupancy',
        fromZigbee: [fz.ias_zone_motion_dev_change, fz.ias_zone_motion_status_change],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
            execute(device, actions, callback);
        },
    },
```

Make sure you have the `ias_zone_motion_dev_change` and `ias_zone_motion_status_change` in `fromZigbee.js` (they have been added quite recently: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L1019)

ekaats, 2018-12-14:
Thanks, it seems to call the function, but something is still wrong:

```
Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":32}' of device 'VMS_ADUROLIGHT' (0x00158d0001cc708f)
/zigbee2mqtt-0.2.0/lib/controller.js:262
                return c.cid === cid && c.type === message.type;
                         ^
TypeError: Cannot read property 'cid' of undefined
    at mappedModel.fromZigbee.filter (/zigbee2mqtt-0.2.0/lib/controller.js:262:26)
    at Array.filter (<anonymous>)
    at Controller.handleZigbeeMessage (/zigbee2mqtt-0.2.0/lib/controller.js:260:51)
    at Zigbee.handleMessage (/zigbee2mqtt-0.2.0/lib/zigbee.js:172:18)
    at emitOne (events.js:116:13)
    at ZShepherd.emit (events.js:211:7)
    at ZShepherd.<anonymous> (/zigbee2mqtt-0.2.0/node_modules/zigbee-shepherd/lib/shepherd.js:138:14)
    at emitMany (events.js:147:13)
    at ZShepherd.emit (events.js:224:7)
    at dispatchIncomingMsg (/zigbee2mqtt-0.2.0/node_modules/zigbee-shepherd/lib/components/af.js:667:45)
```
Perhaps the cid should be a property of the message instead of a key in the data? I can't find a comparable example in devices.js to copy this from. Also, are those values like the zoneid just placeholders for initial setup?

arcsyst, 2018-12-14:
Thanks Koen that seems to work fine for the PIR, I also gained an insight into the bit functions of the zoneStatus value. With that I have modified your code for the Trust ZCTS-808 window contact as below, maybe you could review?

I still get these log messages for the PIR and door contact though it doesn’t seem to affect anything:

  zigbee2mqtt:error 2018-12-14 10:40:55 Failed to configure Trust_Contact_1 0x00158d000202bb30
  zigbee2mqtt:error 2018-12-14 10:40:55 Failed to configure Trust_PIR_1 0x00158d0001cc9439

Devices.js:
    {
      zigbeeModel: ['CSW_ADUROLIGHT'],
      model: 'ZCTS-808',
      vendor: 'Trust',
      description: 'Door & Window Contact',
      supports: 'contact',
      fromZigbee: [fz.ias_contact_dev_change, fz.ias_contact_status_change],
      toZigbee: [],
      configure: (ieeeAddr, shepherd, coordinator, callback) => {
          const device = shepherd.find(ieeeAddr, 1);
          const actions = [
              (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
              (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
          ];
          execute(device, actions, callback);
      },
    },

fromZigbee.js:
    ias_contact_dev_change: {
        cid: 'ssIasZone',
        type: 'devChange',
        convert: (model, msg, publish, options) => {
            if (msg.data.data.zoneType === 0x000D) { // type 0x000D = motion sensor
                const zoneStatus = msg.data.data.zoneStatus;
                return {
                    contact: (zoneStatus & 1) == 0, // Bit 1 = Alarm 2: Presence Indication
                    // tamper: (zoneStatus & 1<<2) > 0, // Bit 2 = Tamper status
                    // battery_low: (zoneStatus & 1<<3) > 0, // Bit 3 = Battery LOW indicator (trips around 2.4V)
                };
            }
        },
    },
    ias_contact_status_change: {
        cid: 'ssIasZone',
        type: 'statusChange',
        convert: (model, msg, publish, options) => {
            const zoneStatus = msg.data.zoneStatus;
            return {
                contact: (zoneStatus & 1) == 0, // Bit 1 = Alarm 2: Presence Indication
                // tamper: (zoneStatus & 1<<2) > 0, // Bit 2 = Tamper status
                // battery_low: (zoneStatus & 1<<3) > 0, // Bit 3 = Battery LOW indicator (trips around 2.4V)
            };
        },
    },

Regards
Brian Clarke.

Koenkk, 2018-12-14:
@ekaats please update to the latest dev branch.

Koenkk, 2018-12-14:
@arcsyst how did the no converter available message look like, I think we can reuse https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L988

ekaats, 2018-12-14:
Thanks, that seems to do the job.
Now I get information from the sensor:
```
MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc708f', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":131}'
```
occupancy was false before I triggered the sensor, so the sensor relays information just fine. However, the error is also still popping up in the logs:
```
Failed to configure 0x00158d0001cc708f 0x00158d0001cc708f
```

Koenkk, 2018-12-14:
The failed to configure happen because the device is probably sleeping (and zigbee2mqtt tries to configure on every startup, you can ignore this).

arcsyst, 2018-12-15:
Contact Closed:
  zigbee2mqtt:debug 2018-12-15 10:51:30 Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":32}' of device 'CSW_ADUROLIGHT' (0x00158d000202bb30)
'

Contact Opened:
  zigbee2mqtt:debug 2018-12-15 10:51:33 Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d000202bb30)

I think we need :
return {contact: msg.data.zoneStatus === 33};

To return true when opened which seems the logical way round for compatibility, not sure if there’s battery info in there, I’ll dig out a dead battery to test!!

arcsyst, 2018-12-15:
Contact Closed:
  zigbee2mqtt:debug 2018-12-15 10:51:30 Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":32}' of device 'CSW_ADUROLIGHT' (0x00158d000202bb30)
'

Contact Opened:
  zigbee2mqtt:debug 2018-12-15 10:51:33 Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d000202bb30)

I think we need :
return {contact: msg.data.zoneStatus === 33};

I have tried reducing the battery voltage to the point of failure, there doesn't appear to be any change in the status so I guess there's no battery info.

Koenkk, 2018-12-15:
Can you check if everything work in the latest: https://github.com/Koenkk/zigbee-shepherd-converters ?

Koenkk, 2018-12-15:
Also supported in the dev branch now.

arcsyst, 2018-12-16:
Thanks, works for me too, just get:
  zigbee2mqtt:error 2018-12-16 13:49:31 Failed to configure Trust_PIR_1 0x00158d0001cc9439
  zigbee2mqtt:error 2018-12-16 13:49:36 Failed to configure Trust_Contact_1 0x00158d000202bb30
  zigbee2mqtt:error 2018-12-16 13:49:47 Cannot get the Node Descriptor of the Device: 0x00158d0001cc9439 (Error: Timed out after 10000 ms)

But I guess we can ignore these.

Koenkk, 2018-12-16:
Ok, great :)

closed 2018-12-16 by Koenkk