Koenkk/zigbee2mqtt #678: Ikea Bulb LED1624G9 "normal" hue after color change

djusHa, 2018-12-10:
Hi everyone!

How can i return to a "normal" hue on my Ikea bulb, after a color change?
What i want is the same warm white color tone that i get if i turn the Bulb off and on again ( bulb was currentless ).

Koenkk, 2018-12-10:
This can be done by setting the `color_temperature` to the default one.

djusHa, 2018-12-10:
thanks for prompt replay. Any suggestions which these value is?

djusHa, 2018-12-10:
Send various values with
`zigbee2mqtt/0xd0cf5efffe28ba53/set {"state":"ON","color_temp":255}`
or
``zigbee2mqtt/0xd0cf5efffe28ba53/set {"color_temp":255}``
no Luck

Feedback from zigbee2mqtt:
`zigbee2mqtt:info 2018-12-10 23:56:05 MQTT publish, topic: 'zigbee2mqtt/0xd0cf5efffe28ba53', payload: '{"state":"ON","brightness":254,"color":{"x":0.14,"y":0.052}}'`

There is no color_temp attribute...

Koenkk, 2018-12-11:
Can you provide more logging?

djusHa, 2018-12-11:
Found something interesting, part of data from db:
`"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 CWS opal 600lm"
`

`"lightingColorCtrl":{"dir":{"value":1},"attrs":{"15":0,"16397":0,"16398":0,"65533":1,"remainingTime":0,"currentX":30015,"currentY":26870,"colorMode":2,"numPrimaries":0,"enhancedColorMode":1,"colorCapabilities":8,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}}`

and here from debug log:
`zigbee2mqtt:debug 2018-12-11 21:36:48 Received zigbee message of type 'devChange' with data '{"cid":"lightingColorCtrl","data":{"currentX":30499,"currentY":24765,"colorMode":1}}' of device 'TRADFRI bulb E27 CWS opal 600lm' (0xd0cf5efffe28ba53)
  zigbee2mqtt:info 2018-12-11 21:36:48 MQTT publish, topic: 'zigbee2mqtt/0xd0cf5efffe28ba53', payload: '{"state":"ON","brightness":254,"color":{"x":0.465,"y":0.378}}'`

Device seems to support color_temp.
What is "colorMode" and "enhancedColorMode"?

djusHa, 2018-12-12:
What about this:
e0426b58a302d265415f97d919d83871954409f5

Koenkk, 2018-12-12:
Can you checkout the dev branch and check if `color_temp` work for this bulb? (should be supported: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L416)

djusHa, 2018-12-12:
What about this commit: [e0426b5](https://github.com/Koenkk/zigbee-shepherd-converters/commit/e0426b58a302d265415f97d919d83871954409f5)

Koenkk, 2018-12-13:
I don't know, does setting the color temperature work?

sergiejL, 2018-12-27:
Hi, 
I have the same problem.
On "release" docker image command is not supported, on latest develop docker image command supported and processed (according to log sent to zigbee network "ToZigbee"). Unfortunately it doesn't work.

Workaround with setting xy color according CIEXYZ gives 20% light less then bulb after power cycle.

jvdpas, 2019-02-09:
> I don't know, does setting the color temperature work?

No setting { color_temp: 304 } does not work on this bulb.

Koenkk, 2019-02-09:
@jvdpas thanks for confirming, this is also what I observed with my LED1624G9, removed the colortemp support for this device.

sergiejL, 2019-02-11:
Hi Koenkk,
but how restore "normal" color. This bulb has 4 different LED (warm white, red, yellow, blue): 
https://github.com/ffleurey/ThingML-Tradfri

I've tried to set xy values but light level is ~20% lower (measure using RGB sensor in my smartphone)
I'm new in zigbee world but is it possible to sniff decrypted data between Ikea remote (or as a last resort Ikea gateway) and bulb?

Koenkk, 2019-02-11:
@sergiejL yes that's possible, please see https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_sniff_zigbee_traffic.html

djusHa, 2019-02-11:
@sergiejL:

Can you try this Values, especially X,Y:
// Color / Temperature related, these are independent of brightness, i.e. do not change if brightness does
	public static final String COLOR = "5706";
	public static final String COLOR_X = "5709";
	public static final String COLOR_Y = "5710";
	public static final String COLOR_COLD = "f5faf6";
	public static final String COLOR_COLD_X = "24930";
	public static final String COLOR_COLD_Y = "24694";
	public static final String COLOR_NORMAL = "f1e0b5";
	public static final String COLOR_NORMAL_X = "30140";
	public static final String COLOR_NORMAL_Y = "26909";
	public static final String COLOR_WARM = "efd275";
	public static final String COLOR_WARM_X = "33135";
public static final String COLOR_WARM_Y = "27211";

It's a part from src of ThingML / Tradfri.
https://github.com/ffleurey/ThingML-Tradfri/blob/master/tradfri-java/src/main/java/org/thingml/tradfri/TradfriConstants.java

sergiejL, 2019-02-12:
> public static final String COLOR_NORMAL_X = "30140";
> public static final String COLOR_NORMAL_Y = "26909";

unfortunately color is almost the same (naked eye) but illuminance drops from ~160 lx to 130 lx (on table placed 1m bellow single bulb).
I sent {"state": "ON", "color": {"x": 0.45989990234, "y": 0.41059875488}} which gives log message:
> Zigbee publish to device '******************', lightingColorCtrl - moveToColor - {"transtime":0,"colorx":30140,"colory":26909} - {"manufSpec":0,"disDefaultRsp":0} - null

I'll buy ikea remote and capture traffic

closed 2019-04-21 by stale[bot]

keteflips, 2020-06-14:
Same issue

https://github.com/Koenkk/zigbee2mqtt/issues/3738