Koenkk/zigbee2mqtt #680: Hue Dimmer Switch stopped working but still works with Hue Bridge

sascha777, 2018-12-11:
Hi guys,

I had two Hue dimmer switches running perfectly on zigbee2mqtt 0.2.0 and a CC2530_CC2591 coordinator for a few weeks.

All of the sudden one of them stopped working yesterday. Since then I am not able to get it running anymore. 

That's what I tried:
- Replaced battery
- Removed via mqtt/conf and then tried to pair again
- Updated zigbee2mqtt
- NOT reflashed firmware yet

Resetting the dimmer switch by pressing button on the back for 10 seconds / all four button on the front for 2-3 seconds sets the dimmer switch in pairing mode (starting blink red/green/red/green followed by a sequence of periodically orange blinking) does not produce any messages in zigbee2mqtt anymore, not even in debug mode.

The dimmer hardware is the same model as the other dimmer switch that is stills running smoothly. Firmware is up-to-date (by means of Hue bridge).

I thought of a hardware defect but the strange thing is that the dimmer switch can still be added to the original Hue bridge and is fully working with that bridge.

What might be worth mentioning is that I ran 16 devices before (without errors) on the coordinator (not exactly sure, what firmware is running) and the error occured on the first reboot after adding the 16th device.

Had anyone experience a similar problem?

Kind regards,
Sascha

ryanbeaton, 2018-12-11:
Doesn’t the CC2530_CC2591 firmware have a 15 device limit. Do you have any routers in the zigbee network.

ryanbeaton, 2018-12-11:
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2530_CC2591/README.md

Shows NWK_MAX_DEVICE_LIST=15

sascha777, 2018-12-11:
I don't have routers in the network. Strange thing is that it ran with 16 device for a couple of days. Even removing device down to 14 doesn't allow of re-adding the dimmer switch.

ryanbeaton, 2018-12-11:
Yeah that what others have said too. I think you have to flash the stick again and re-pair everything now. It’s the stick that’s corrupted

ryanbeaton, 2018-12-11:
No reason to put the router in the Pi, it only draws power. Just plug it into a USB phone charger somewhere else in the house

sascha777, 2018-12-11:
Can't believe that it's easy like this! You made my day, thanks a lot!

closed 2018-12-11 by sascha777