Koenkk/zigbee2mqtt #682: GLEDOPTO (GL-S-007Z)  modelID 'undefined'

DJTerentjev, 2018-12-11:
Thank you to developers who use to works on this project hard. I’ve seen how project was developing and decided to try. I order CC2531 and several bulbs RGB GLEDOPTO (GL-S-007Z). It was only bulb with GU10 available.  GLEDOPTO weren’t in table of supported devises, but support by HUE was promising. As soon I received it and install zigbee2mqtt + HA I found bulb don’t provide modelID. Just 'undefined'. I use last versions of zigbee2mqtt and firmware of CC2531. I reset bulb several times, tried all of them, flush CC2531 and reinstall zigbee2mqtt. I tried iobroker as well. Same result. I was lucky to connect some Xiaomi devises, but not GLEDOPTO.
Deeply searching I found devises with such behavior can’t be used with zigbee2mqtt (than I can’t use this bulbs at last). Sorry if my questions are stupid, but let me ask you one more time. As I understood, modelID needed to select correct converter. If it possible to use something like default controller (for any devise correct controller was not found):
{ 
zigbeeModel: ['undefined'], 
..
},
or maybe converter can be selected by devise ID , assigned manually etc?  Any chance I can connect and use these bulbs? For now bulb light led green , so totally unusable even as ‘stupid’ bulb. 

[Zigbee-ZLL-dimmer-LED-RGBW](https://www.ebay.com/itm/Zigbee-ZLL-dimmer-LED-RGBW-WW-CW-spotlight-link-LED-GU10-bulb-with-Echo-Plus/183359903703?ssPageName=STRK%3AMEBIDX%3AIT&var=690747103065&_trksid=p2057872.m2749.l2648) 

Thank you for support!

DJTerentjev, 2018-12-11:
[Start-up log](https://www.hastebin.com/okilowapiw.pl)
[Pairing log](https://www.hastebin.com/mixaqenequ.py)
[database.db](https://www.hastebin.com/egejocoxuc.rb)

khzd, 2018-12-12:
Hi
Can you post your configuration.yaml ?
if you see Model "undefined" you need to retry again and again the pairing process until you see GLEDOPTO in the database.db and Pairing log

give a try

DJTerentjev, 2018-12-12:
> Can you post your configuration.yaml ?

khzd,
Thank you for response. Please see: [configuration.yaml ](https://www.hastebin.com/amakenelol.js)
Could you please advise if I just reset devise and pair it again or I should repeat all the process including  flush CC2531 firmware.

khzd, 2018-12-12:
Hi
you have to retry pairing process until undefined became the name of the GLEDOPTO device.

some time with a fresh reboot i observe a faster pairing

DJTerentjev, 2018-12-12:
hello,

please see [database.db ](https://pastebin.com/STLqfjJa)

I named them in configuration.yaml and see the names while starting npm:
GLEDOPTO_2 (0x00124b001b6aea81): unkown - undefined unknown (Router)
GLEDOPTO_1 (0x00124b001b6aec74): unkown - undefined unknown (Router)
but I don't see the names in database.db
As I understood, zigbee2mqt will not pair devise again, until I clean-up configuration.yaml, correct?

khzd, 2018-12-13:
Hi 
You may to delete those 2 lignes in dB and restart zigbee2mqtt
Retry the pairing and try to put the device as close as possible to the router 

As sois as the device are not undefined , you may add some code in devices , tozigbee and fromzigbee.js

First try to get it not undefined

It happened to me once and I forced in dB undefined to dummy and start next steps, later the pairing cleared my dummy to the good name :) 
Pairing is touchy

DJTerentjev, 2018-12-14:
Hi,

Looks like it's not working for me. I've done more the 20 attempts (then I stopped counting) (tried to change zigbee channel and flush CC2531, reinstall Hassio). Every time  "undefined". All the times bulbs were next to RP with stick (just some mm away). Is anything else I can try?

khzd, 2018-12-15:
Hi
I m fare from my pc...
What I did once is to complete the dB manually. 
I will give you an example to test ASAP

DJTerentjev, 2018-12-15:
hello,

Thank you. Appreciate it.

ArJay60, 2018-12-18:
I have the same problem but with GLEDOPTO GL-S-008Z. No Model. Did you get yours working. Here the line in database.db after coupling:

{"id":5,"type":"Router","ieeeAddr":"0x00124b001b6ae333","nwkAddr":12119,"manufId":0,"epList":[11,12,13],"status":"online","joinTime":1545169776,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{}}},"_id":"bjI8WVVvB8ZwqWon"}

DJTerentjev, 2018-12-18:
ArJay60 ,
I didn't. Stoped trying for now.  
Hope for help from khzd (see above).

khzd, 2018-12-18:
Hi guys
Always far from my pc
I had a exchange with devs team for another gledopto and we solve it. Can you see the issue and adapt the solution as if your device is a GLEDOPTO as in the wiki?
You need to add the model I’d by hand in the dB like in my post 
( I m on a iPhone...)
Not sure it work,  gledopto have different  release for the same item :(

khzd, 2018-12-20:
hi back to my pc

@ArJay60 please replace 
{"id":5,"type":"Router","ieeeAddr":"0x00124b001b6ae333","nwkAddr":12119,"manufId":0,"epList":[11,12,13],"status":"online","joinTime":1545169776,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{}}},"_id":"bjI8WVVvB8ZwqWon"}
by
{"id":5,"type":"Router","ieeeAddr":"0x00124b001b6ae333","nwkAddr":12119,"manufId":0, **"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO"**,"epList":[11,12,13],"status":"online","joinTime":1545169776,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{}}},"_id":"bjI8WVVvB8ZwqWon"}
and give a try 
for @DJTerentjev give a try with our record. 

here is a copy of my GL-C-008
{"id":50,"type":"Router","ieeeAddr":"0x00124b0019c606cd","nwkAddr":10828,"manufId":0,**"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO"**,"epList":[11,13],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"geCEMkRqlaMe6muE"}

ArJay60, 2018-12-20:
Hi hkzd. Manually adaption works although my GL-S-008Z is now displayed as GL-C-008. Do they have the same set of commands (as the entry in devices.js)?

DJTerentjev, 2018-12-20:
Hi 
Looks like it's doesn't work for me. Device isn't responding. 
 
  zigbee2mqtt:info 2018-12-20 23:45:07 Zigbee publish to '0x00124b001b6aea81', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:error 2018-12-20 23:45:37 Zigbee publish to '0x00124b001b6aea81', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms

Although there is different immediate response when devise is off:

  zigbee2mqtt:error 2018-12-20 23:48:32 Zigbee publish to '0x00124b001b6aea81', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.

khzd, 2018-12-21:
@ArJay60 > Hi hkzd. Manually adaption works although my GL-S-008Z is now displayed as GL-C-008. Do they have the same set of commands (as the entry in devices.js)?

I do not have a GL-S-008Z, but you need to compare the "epList":[11,12,13], this is a main difference. "devId":528 is a second main one :(

khzd, 2018-12-21:
@DJTerentjev that's bad :(
please post your DB record, you probaly have different eplist, on guys had a different combiniton and dev team recommend to hard code the ep in devices.js...

ArJay60, 2018-12-21:
> @ArJay60 > Hi hkzd. Manually adaption works although my GL-S-008Z is now displayed as GL-C-008. Do they have the same set of commands (as the entry in devices.js)?
> 
> Obviously, if you re-pair your device , you have to redo this operation
> you may open a **PR** to have a official support, something is wrong during pairing, so you miss a main part of DB record.
> 
> what are the main functionalities of GL-S-008Z ? RGB + CCT or only RGB ? i see on ali that it's not a cct.
> 
> I do not have a GL-S-008Z, but you need to compare the "epList":[11,12,13], this is a main difference. "devId":528 is a second main one :(
> 
> only one ep is used in my case, the others are ignored, see the code of devices.js

Hi khzd,
full record is (this is a second bulb therefore another ieeeAddr)

{"id":5,"type":"Router","ieeeAddr":"0x00124b001b6aeae5","nwkAddr":12119,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO GL-S-008Z","epList":[11,12,13],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"w48AFN78O9JS7M6x"}

Created a device in devices.js referencing GLEDOPTO GL-S-008Z (copy of the original GLEDOPTO record with only name changes).

Device is a RGBWW PAR16 bulb. Trying to find out what to put directly in MQTT message to check wether of not I have a good connection. Maybe only on/of command. Do you now the JSON structure to put in MQTT to do that?

DJTerentjev, 2018-12-21:
Hi. Thank you for help.
Please see DB: https://pastebin.com/xmDfwGgf
and log: https://pastebin.com/An5D5JUt
My epList":[11,12,13]" looks same as ArJay60's
Firs time when I start using zigbee2mqtt I saw two different devises of  GLEDOPTO in devices.js, but after I update zigbee2mqtt I have only one.

dcsg, 2018-12-22:
@khzd do you know all mqtt state_topics/command_topics for brigthness, rgb, etc?

dcsg, 2018-12-22:
Working now with this configuration:

```

  - platform: "mqtt"
    name: "Gledopto"
    state_topic: "zigbee2mqtt/0x00124b001ad74041"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    color_temp: true
    xy: true
    schema: "json"
    command_topic: "zigbee2mqtt/0x00124b001ad74041/set"
```

ArJay60, 2018-12-22:
> {"id":3,"type":"Router","ieeeAddr":"0x00124b001b6aea81","nwkAddr":42382,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO","epList":[11,12,13],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"s44UV7vu3VnD7VSz"}
> 
> > > @ArJay60 > Hi hkzd. Manually adaption works although my GL-S-008Z is now displayed as GL-C-008. Do they have the same set of commands (as the entry in devices.js)?
> > > Obviously, if you re-pair your device , you have to redo this operation
> > > you may open a **PR** to have a official support, something is wrong during pairing, so you miss a main part of DB record.
> > > what are the main functionalities of GL-S-008Z ? RGB + CCT or only RGB ? i see on ali that it's not a cct.
> > > I do not have a GL-S-008Z, but you need to compare the "epList":[11,12,13], this is a main difference. "devId":528 is a second main one :(
> > > only one ep is used in my case, the others are ignored, see the code of devices.js
> > 
> > 
> > Hi khzd,
> > full record is (this is a second bulb therefore another ieeeAddr)
> > {"id":5,"type":"Router","ieeeAddr":"0x00124b001b6aeae5","nwkAddr":12119,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO GL-S-008Z","epList":[11,12,13],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"12":{"profId":260,"epId":12,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"w48AFN78O9JS7M6x"}
> > Created a device in devices.js referencing GLEDOPTO GL-S-008Z (copy of the original GLEDOPTO record with only name changes).
> > Device is a RGBWW PAR16 bulb. Trying to find out what to put directly in MQTT message to check wether of not I have a good connection. Maybe only on/of command. Do you now the JSON structure to put in MQTT to do that?
> 
> try {"state":"OFF","color":{"x":0.701,"y":0.299},"color_temp":311,"brightness":92}

@hkzd publishing {"state":"OFF","color":{"x":0.701,"y":0.299},"brightness":92} or {"state":"OFF","color_temp":311,"brightness":92} in zigbee2mqtt/<NAME OF DEVICE/set works for me.
Thks. Going further to make a connection to Domoticz through stas-demydiuk/domoticz-zigbee2mqtt-plugin.

khzd, 2018-12-22:
@ArJay60 welcome

khzd, 2018-12-22:
@DJTerentjev 
what is your integration status?

DJTerentjev, 2018-12-22:
Hi khzd ,
After install Hassbian, mosquito and zigbee2mqtt. Adjust DB and device.js according your recommendations. Devise start working, but almost after every command it get hanged-up. I can see errors in the log:

     zigbee2mqtt:info 2018-12-22 17:09:58 Zigbee publish to '0x00124b001b6aec74', lightingColorCtrl - moveToColor - {"colorx":43712,"colory":18612,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 12
  zigbee2mqtt:error 2018-12-22 17:09:59 Zigbee publish to '0x00124b001b6aec74', lightingColorCtrl - moveToColor - {"colorx":43712,"colory":18612,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 12 failed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:info 2018-12-22 17:10:05 Zigbee publish to '0x00124b001b6aec74', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 12
  zigbee2mqtt:error 2018-12-22 17:10:05 Zigbee publish to '0x00124b001b6aec74', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 12 failed with error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:info 2018-12-22 17:11:26 MQTT publish: topic 'zigbee2mqtt/0x00124b001b6aec74', payload '{"state":"ON"}'

DJTerentjev, 2018-12-22:
Hi khzd , I see a lot of people have the same error (233. MAC no ack). It is working OK if change status not very often (lets say every 10 second). So I is way better then nothing. Thank you very much for you help.

DJTerentjev, 2018-12-22:
After I re-paired device finally it is working fine. Thanks again.
So to connect GL-S-007Z:
1. Follow [getting-started](https://koenkk.github.io/zigbee2mqtt/#getting-started) and make sure you have last version of zigbee2mqtt (1.0.1 in the moment)
2. While pairing keep your devise less then one miter (you can put it away after pairing)
3. As you got 'undefine' you have to manualy add to /opt/zigbee2mqtt/data/database.db  bold part:

{"id":50,"type":"Router","ieeeAddr":"0x00124b0019c606cd","nwkAddr":10828,"manufId":0,**"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GLEDOPTO"**,"epList":[11,13],"status":"offline","joinTime":null,"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}},"13":{"profId":49246,"epId":13,"devId":57694,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"geCEMkRqlaMe6muE"}

DJTerentjev, 2018-12-22:
I think it is good I idea, to add GL-S-007Z  to supported devices and please add instruction to wiki.
Please close this issues. Thanks a lot.

khzd, 2018-12-22:
you are welcome.

DJTerentjev, 2018-12-24:
Hi 
I was too optimistic when I decided that bulb is working correctly. It was working properly only on the maximum distance around 1 miter. A little further and a lot of mistakes in the log.  Re-pairing did't help. Only one I could think about was Zigbee channel. From beginning I was trying with channel 22. So I decided to give very last try with channel 11.  I had to start over again, but this time both my bulbs started working correctly. 
What do you think, GLEDOPTO can't support all zigbee channels (but how it get paired...)? Or I had some other issue? Please post you experience if any one use GLEDOPTO products with zigbee channels rather then 11. Appreciated .

khzd, 2018-12-26:
Hi
I have a pair of gl-c-008 working at channel 26, I chooses this channel because we think that livolo device is working only on channel 26.
The gledopto are fare from the gateway.
What is the device link quality?

Do you have others routers ? I peraonnqly add a pair of cc2530 to extend the link quality
Cheers

DJTerentjev, 2018-12-26:
Hi Khaled,
Good to know. After I reinstalled Hassio, Mosquito and ZigbeeToMQTT and change channel everything are working well and I satisfied with range (before, it was less then 1 miter).  I'm not sure what caused problem before, maybe something else.
I don't filling to play with it now and change the channel again. I'll do it later.
I don't know about link quality. How I can check it? Log don't provide such information:

```
Dec 26 21:27:47 hassbian npm[491]:   zigbee2mqtt:info 2018-12-26 21:27:47 Zigbee publish to '0x00124b001b6aec74', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 12
Dec 26 21:27:47 hassbian npm[491]:   zigbee2mqtt:info 2018-12-26 21:27:47 MQTT publish: topic 'zigbee2mqtt/0x00124b001b6aec74', payload '{"state":"OFF","brightness":254,"color":{"x":0.136,"y":0.04}}'
````

khzd, 2018-12-26:
ho sorry, the GLEDOPTO doesn't have linkquality attribute!
all my xiaomi devices have and i'm able to see if they are far or out of the network.

closed 2018-12-28 by DJTerentjev