Koenkk/zigbee2mqtt #685: Network not secure -> request: add configuration of encryption key

d-EScape, 2018-12-12:
Both the Trust Center key and the default network key are published on the internet. Using a sniffer i can decrypt and read all traffic using those keys, without joining the network(!) I have been reading code and trying to understand how to add a configurable network key as a PR bit this seems te be beyond my capabilities.

Can someone come up with a way to make the network key user configurable in the zigbee2mqtt config file? (not hardcoding it into config files of submodules, which seems the only option now).

I deliberately do not mention the locations and methods I found... don't want te make it to easy until a fix is in the making ;-)

JLFN, 2018-12-18:
@Koenkk have you seen this?

Koenkk, 2018-12-19:
This seems to be a firmware settings, setting to 0x00 will produce a random key (http://dev.ti.com/tirex/content/simplelink_zigbee_sdk_plugin_2_20_00_06/docs/zigbee_user_guide/html/zigbee/developing_zigbee_applications/network_configuration.html)

d-EScape, 2018-12-20:
Wouldn't the precfgkey constant in Zigbee-shepherd just overwrite it again?
I found out (the hard way) that changing the key and changing it back will still require all devices to repair. So to make it a random/configurable key the software should set it once and not touch it after the first install.

sehraf, 2018-12-24:
Nice, thank you!

Koenkk, 2018-12-24:
Thank you too!

Bruceforce, 2019-01-26:
@Koenkk this might be a stupid question but what type do the values in `network_key:` have? Are those single hex-values in int format? If so, I thought that the zigbee network_key has a 128 Bit key. The example value `network_key: [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13],` can be seen as 64 Bit when 1=0x1, 3=0x3 ... or would something like `network_key: [107, 243, 142, 198, 20, 216, 136, 183, 10, 23, 195, 214, 209, 247, 225, 104], ` be valid?

Sorry if I'm mixing something up. I'm quite new to this.