Koenkk/zigbee2mqtt #687: Bad Latency

cjsteward, 2018-12-12:
After I updated from 0.1.7 to 0.2.0 it seems like zigbee2mqtt has gotten a lot slower. It definitely takes a lot longer to stop and start and now my Xiaomi sensors seem to also react a lot slower/miss messages. I have a door sensor that I use to test the latency/accuracy of my zigbee network and before I updated I could "open and close" my test sensor as fast as I could and it would always end up reporting correctly. Now when I "open and close" the sensor, it misses a lot of messages and ends up reporting it as open when the door sensor is physically closed or vice versa. I'm seeing a lot of double messages as well, such as two "contact":false" in a row. QOS settings don't seem to make a difference either. I can see that the messages can take a few seconds to update in the zigbee2mqtt addon logs now too. I can open and close a door and it will randomly take a couple seconds to show up in the logs whereas before it was always instant. I'm running hassio on a raspberry pi 3b+ with a CC2530 coordinator and a CC2530 router.

Any help would be greatly appreciated.

Koenkk, 2018-12-13:
Do you see a high load when running `htop` ?

cjsteward, 2018-12-13:
@Koenkk No, that's what's weird, my resources seem to be fine. CPU hovers around 6-10% and RAM is at ~60%. I experience the random delays after fresh reboots too.

Koenkk, 2018-12-14:
Does downgrading back to 0.1.7 fix the problem?

cjsteward, 2018-12-14:
@koenkk I'm using the add-on by Daniel Welch, how do I manually downgrade to 0.1.7. I haven't been able to find any answers online.

Thanks again for the help too man, I appreciate you taking the time to respond

Koenkk, 2018-12-14:
@danielwelch can you help on how to downgrade?

danielwelch, 2018-12-14:
@Koenkk interestingly, I’m not sure if downgrading is even supported. You can downgrade Hassio via an API call (using the upgrade method and supplying the desired version as a parameter). But in reviewing the source code for the add ons API, I see that the analogous method for add-ons only checks against the latest version. I’ll investigate if there’s another way.

closed 2019-02-19 by stale[bot]