Koenkk/zigbee2mqtt #688: How does one use toZigbee.js to set device parameters

mcsSolutions, 2018-12-12:
I have DJT11LM that reports accelerometer-based information.  toZigbee.js implies that it can be set to high/low/medium sensitivity.  What do I do so that zigbee2mqtt sends the desired sensitivity to the device?

/* Note when send the command to set sensitivity, press button on the device to make it wakeup*/
    DJT11LM_vibration_sensitivity: {
        key: 'sensitivity',
        convert: (value, message, type) => {
            const cid = 'genBasic';
            const attrId = 0xFF0D;

            if (type === 'set') {
                const lookup = {
                    'low': 0x15,
                    'medium': 0x0B,
                    'high': 0x01,
                };

mcsSolutions, 2018-12-13:
I believe my question is more basic.  What is the general mechanism by which toZigbee.js is used?  It is clear that fromZigbee.js is used by zigbee2mqtt when it needs to process information from a device.  When and how is toZigbee.js used?  Do I set something upon in configuration.yaml or is there another way to communicate with a running zigbee2mqtt to tell it to configure the zigbee device?

I am not a home assistant user so do not recognize how the pasted text is used with it.

Ton1965, 2018-12-13:
In general you need to send an MQTT message to zigbee2mqtt with a special payload. It is described in detail [here](https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttdevice_idset).

mcsSolutions, 2018-12-13:
Excellent.  

I was able send the 'set' and 'get' topics.  With low setting for vibration level it seems to be hard to make it report vibration.  High setting was easier to induce vibration, but still needed a deliberate degree of movement.

I can also see that the 'get' topic was recognized, but I do not see a response related to sensitivity that was programmed.  Is it undecoded in the 'unknown_data' parameter?

zigbee2mqtt:info 2018-12-12 22:55:59 Zigbee publish to '0x00158d0002b23edc', genBasic - read - [{"attrId":65293}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null
  zigbee2mqtt:info 2018-12-12 22:55:59 MQTT publish, topic: 'zigbeeWA/0x00158d0002b23edc', payload: '{"angle_x":3,"angle_y":82,"angle_z":8,"angle_x_absolute":87,"angle_y_absolute":8,"linkquality":73,"unknown_data":7733248,"battery":99,"voltage":3025}'

zigbeeWA/0x00158d0002b23edc/get = {"sensitivity":"high"}
zigbeeWA/0x00158d0002b23edc = {"angle_x":3,"angle_y":82,"angle_z":8,"angle_x_absolute":87,"angle_y_absolute":8,"linkquality":73,"unknown_data":7733248,"battery":99,"voltage":3025}

closed 2019-02-18 by stale[bot]