Koenkk/zigbee2mqtt #690: IKEA TRADFRI wireless dimmer (ICTC-G-1) How to get the angle and direction of move

andreas-nn, 2018-12-12:
The device is supported! OK

It is creating one payload {"brightness":226}.
But this information is not reliable. Sometime it changes the value with high intense.
Is it possible to add the output with "angle" and "direction"? With this information I could attenuate the change of the value.
Thanks

Koenkk, 2018-12-13:
This converted is implemented here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L116 

Can you try modifying it according to your needs?

andreas-nn, 2018-12-13:
Thank you. This is what I want to change.
But there is a problem for me:
- how to publish "delta" from line 128 as a second element?

Koenkk, 2018-12-14:
New values can be passed by changing: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L147 like: `s.publish({brightness: s.value, newvalue: 1});`

jbaumann, 2019-01-15:
In the file fromZigbee.js change the function ictcg1:
`const ictcg1 = (model, msg, publish, options, action) => {
    const deviceID = msg.endpoints[0].device.ieeeAddr;

    if (!store[deviceID]) {
        const _publish = debounce((msg) => publish(msg), 250);
        store[deviceID] = {since: false, direction: false, value: 255, publish: _publish};
    }

    const s = store[deviceID];
    var delta = 0;
    if (s.since && s.direction) {
        // Update value
        const duration = Date.now() - s.since;
        delta = Math.round((duration / 10) * (s.direction === 'left' ? -1 : 1));
        const newValue = s.value + delta;
        if (newValue >= 0 && newValue <= 255) {
            s.value = newValue;
        }
    }

    if (action === 'move') {
        s.since = Date.now();
        s.direction = msg.data.data.movemode === 1 ? 'left' : 'right';
    } else if (action === 'stop' || action === 'level') {
        if (action === 'level') {
            s.value = msg.data.data.level;
        }

        s.since = false;
        s.direction = false;
    }

    s.publish({brightness: s.value, delta: delta});
};`

I've posted the whole function solely because delta was declared in an inner block and thus not available for the call to publish().

shadow974, 2019-01-15:
Does this work for you? Is the reading more stable than the brightness-level?
Can you create a pull-request, please? This should be included in the main release, imho.

jbaumann, 2019-01-15:
The function works without problems, but the deltas are kind of strange. Sometimes there's nothing even though I turn the knob a quarter, other times it is the wrong leading sign (but that is seldom) and quite a number of times I get a 0 (Zero) as delta.
So there might be an underlying general problem with the communication with the dimmer...

shadow974, 2019-01-16:
Maybe somehow related to #542 ?

andreas-nn, 2019-01-16:
The code from jbaumann is good. thanks.
But I have the same problem with the communication to the dimmer. Sometime it works and sometime not. That is why I want to get the delta. In the home automation system (symcon) I want to smooth it by cutting the peaks and integration.
Ikea-dimmer is looking nice, but working ...
I would like to see the code in the next release.

jbaumann, 2019-01-17:
Removing the debounce()-function seems to help a bit:
store[deviceID] = {since: false, direction: false, value: 255, publish: publish};

But the devices still is quite erratic in its behavior...

closed 2019-03-18 by andreas-nn