Koenkk/zigbee2mqtt #691: Cube doesnt work anymore?

1981nick1982, 2018-12-12:
Since today my cube doesnt  switch my lights anymore. In the zigbee2mqtt log is see the action but lights dont go on our off.

Code in automation.yaml
` - id: flip180
    alias: flip180
    trigger:
      platform: mqtt
      topic: 'zigbee2mqtt/0x00158d0002486e28'
    condition:
      condition: template
      value_template: "{{ 'flip180' == trigger.payload_json.action }}"
    action:
    - service: light.toggle
      data:
        entity_id: group.eetkamer
`
Log:
` zigbee2mqtt:info 2018-12-12 23:09:03 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002486e28', payload: '{"battery":37,"voltage":2975,"linkquality":102,"action":"flip180","side":3}'
 `

1981nick1982, 2018-12-15:
I first had another problem (see https://github.com/danielwelch/hassio-zigbee2mqtt/issues/71). This problem has been solved by using a username + password in both MQTT and Zigbee2MQTT. Now I see in the log of Zigbee2MQTT all status updates, however home Assistant does nothing with this. Also my temperature sensors from Aqara have disappeared and are no longer shown in my home screen. Have no idea how I can see that the info arrives in Home Assistant. Have read about deleting the Mosquito DB however I can not find it in HASSIO or via SHH.

Koenkk, 2018-12-15:
@1981nick1982 what version are you running, if you updated to edge or 0.1.0 you likely run to https://github.com/Koenkk/zigbee2mqtt/issues/668

1981nick1982, 2018-12-15:
After removing MQTT: + discovery from the configuration.yaml. and restart the thermostats return a value. The Aqara cube, however, does nothing with the set automation.

Zigbee2MQTT version 0.2.0 (Danielwelsh hassio addon)
MQTT version 4

1981nick1982, 2018-12-16:
@koenkk In the history panel of home assistant I see the various movements that I do with the cube. Could it be that my automation.yaml script is no longer correct? Or that I need to set up a friendly name? Do not know exactly where to do this?

Koenkk, 2018-12-17:
@1981nick1982 probably your device ID changed, see https://github.com/Koenkk/zigbee2mqtt/issues/668

1981nick1982, 2018-12-25:
@Koenkk I have reisstalles my cube but it has no effect. Also my device id is correct and in the history panel of Hassio I see my last action. When I look at the state.json file in the zigbee2mqtt map, I don't see the actions of the cube. Can this be the problem? In my automation i link to trigger.payload_json.action ?

Koenkk, 2018-12-26:
@1981nick1982 seems that this is something in your home assistant configuration. 

`action` is one of the attribute that get's not cached by zigbee2mqtt as you don't want to resend those. You only want to send them when the action happens.