Koenkk/zigbee2mqtt #693: Unable to connect to MQTT server

doppiej, 2018-12-13:
Hello,

I am unable to get zigbee2mqtt working. I want it on a raspberry pi 2 with Domoticz.
This is the log:
`  zigbee2mqtt:info 2018-12-13 11:10:10 Logging to directory: '/opt/zigbee2mqtt/d                                       ata/log/2018-12-13.11-10-10'
  zigbee2mqtt:info 2018-12-13 11:10:12 Starting zigbee2mqtt version 0.2.0 (commi                                       t #044db31)
  zigbee2mqtt:info 2018-12-13 11:10:12 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-12-13 11:10:13 zigbee-shepherd started
  zigbee2mqtt:info 2018-12-13 11:10:13 Coordinator firmware version: '20180815'
  zigbee2mqtt:info 2018-12-13 11:10:13 Currently 0 devices are joined:
  zigbee2mqtt:warn 2018-12-13 11:10:13 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2018-12-13 11:10:13 Allowing new devices to join.
  zigbee2mqtt:warn 2018-12-13 11:10:13 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2018-12-13 11:10:13 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-12-13 11:10:13 Connecting to MQTT server at mqtt://localhost:1883
  zigbee2mqtt:info 2018-12-13 11:10:13 zigbee-shepherd ready
  zigbee2mqtt:error 2018-12-13 11:10:23 Not connected to MQTT server!
  zigbee2mqtt:error 2018-12-13 11:10:33 Not connected to MQTT server!
  zigbee2mqtt:error 2018-12-13 11:10:43 Not connected to MQTT server!
`
Is there someone who can help me fixing this?
Thanks in advance!

doppiej, 2018-12-13:
I tried running it through a VPN connection, however no luck. It says unable to connect to server.
I added the ip address of the pi and the port 1883 and generated a client_id. I left the remaining settings unchanged.

Koenkk, 2018-12-13:
You should first debug why you can't connect via mqtt.fx.

millerce, 2018-12-15:
But mine is on danielwelch/hassio-zigbee2mqtt, zigbee2mqtt-edge

Koenkk, 2018-12-15:
@millerce I think you experiencing the same as: https://github.com/danielwelch/hassio-zigbee2mqtt/issues/71

cbrherms, 2018-12-19:
@millerce you're using a different platform to the OP of this issue so may be a different cause entirely.

Make a new issue and if you can, include your config for both your mqtt addon and zigbee2mqtt addon (though redact sensitive passwords)

closed 2019-02-24 by stale[bot]

whizkid79, 2019-03-01:
I had the same issue. The values for the mqtt broker weren't updated properly in /share/zigbee2mqtt/configuration.yaml
Fixed that manually and it worked.

Btje1977, 2019-03-21:
Still the case, when you change it in hassio it won't change in the configuration.yaml