Koenkk/zigbee2mqtt #695: Installation Issue

NEA1971, 2018-12-13:
Followed the installation instructions - when starting zigbee2mqtt (npm start) I get an error

> TypeError: Cannot read property 'port' of null

the complete output as follows:

`> zigbee2mqtt@0.2.0 start /opt/zigbee2mqtt
> node index.js

		Zigbee2mqtt requires node version >=8.11 10, you are running v8.14.0!

  zigbee2mqtt:info 2018-12-13 13:47:09 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-12-13.13-47-09'
  zigbee2mqtt:info 2018-12-13 13:47:09 Starting zigbee2mqtt version 0.2.0 (commit #044db31)
  zigbee2mqtt:info 2018-12-13 13:47:09 Starting zigbee-shepherd
/opt/zigbee2mqtt/lib/zigbee.js:35
        this.shepherd = new ZShepherd(settings.get().serial.port, shepherdSettings);
                                                           ^

TypeError: Cannot read property 'port' of null
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.js:35:60)
    at startupLogVersion (/opt/zigbee2mqtt/lib/controller.js:45:25)
    at git.getLastCommit (/opt/zigbee2mqtt/lib/controller.js:496:13)
    at /opt/zigbee2mqtt/node_modules/git-last-commit/source/index.js:55:4
    at /opt/zigbee2mqtt/node_modules/git-last-commit/source/index.js:23:3
    at ChildProcess.exithandler (child_process.js:268:7)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:915:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.2.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.2.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/administrator/.npm/_logs/2018-12-13T12_47_10_000Z-debug.log`

Additionally content of the config-file:

`# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://homeserver'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
port:   /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E1A42D-if00`

Koenkk, 2018-12-13:
Add 2 spaces before port:

```yaml
# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port:   /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E1A42D-if00
```

NEA1971, 2018-12-13:
works now - thank you!

closed 2018-12-13 by NEA1971