Koenkk/zigbee2mqtt #696: Unable to add new devices

Imperial-Guard, 2018-12-13:
Somehow I can't add new devices anymore.

I'm trying to add a Philips Hue bulb, the bulb is reset to factory settings. But I won't add to my devices.

Tried already to restart Zigbee2MQTT. 

See version and info below: 

`[Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2018-12-13T17:36:15: PM2 log: Launching in no daemon mode
2018-12-13T17:36:16: PM2 log: App [npm:0] starting in -fork mode-
2018-12-13T17:36:16: PM2 log: App [npm:0] online
> zigbee2mqtt@0.2.0 start /zigbee2mqtt-0.2.0
> node index.js
		Zigbee2mqtt requires node version >=8.11 10, you are running v8.11.4!
  zigbee2mqtt:info 2018-12-13 17:36:26 Logging to directory: '/share/zigbee2mqtt/log/2018-12-13.17-36-26'
  zigbee2mqtt:info 2018-12-13 17:36:30 Starting zigbee2mqtt version 0.2.0 (commit #unknown)
  zigbee2mqtt:info 2018-12-13 17:36:30 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-12-13 17:36:31 zigbee-shepherd started
  zigbee2mqtt:info 2018-12-13 17:36:31 Coordinator firmware version: '20181024'
  zigbee2mqtt:info 2018-12-13 17:36:31 Currently 15 devices are joined:
  zigbee2mqtt:info 2018-12-13 17:36:31 hallway/sensor/0004/doorsensor (0x00158d000251b6f2): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 scullery/sensor/0003/motionsensor (0x00158d0002b45f88): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 kitchen/sensor/0001/motionsensor (0x00158d0002732adb): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 living_room/sensor/0004/doorsensor (0x00158d00022d2d2c): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 hallway/sensor/0003/motionsensor (0x00158d0002b4863d): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 hallway_up/sensor/0002/smokedetector (0x00158d00028ac409): JTYJ-GD-01LM/BW - Xiaomi MiJia Honeywell smoke detector (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 hallway_up/sensor/0001/motionsensor (0x00158d0002753dde): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 kitchen/sensor/0002/doorsensor (0x00158d00025316d3): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 scullery/sensor/0002/smokedetector (0x00158d000282298f): JTYJ-GD-01LM/BW - Xiaomi MiJia Honeywell smoke detector (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 scullery/sensor/0001/gardendoorsensor (0x00158d000243753d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 toilet/sensor/0001/windowsensor (0x00158d00024e548d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 living_room/light/0006/leddriver (0x90fd9ffffe010c68): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
  zigbee2mqtt:info 2018-12-13 17:36:31 living_room/sensor/0003/gardendoorsensor (0x00158d000232c194): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
  zigbee2mqtt:info 2018-12-13 17:36:31 living_room/light/0002/boatlight (0xd0cf5efffe2ed08a): E1603 - IKEA TRADFRI control outlet (Router)
  zigbee2mqtt:info 2018-12-13 17:36:31 0x001788011032d314 (0x001788011032d314): 324131092621 - Philips Hue dimmer switch (EndDevice)
  zigbee2mqtt:warn 2018-12-13 17:36:31 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2018-12-13 17:36:31 Allowing new devices to join.
  zigbee2mqtt:warn 2018-12-13 17:36:31 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2018-12-13 17:36:31 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-12-13 17:36:31 Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 2018-12-13 17:36:32 zigbee-shepherd ready
  zigbee2mqtt:info 2018-12-13 17:36:32 Connected to MQTT server
  zigbee2mqtt:info 2018-12-13 17:36:32 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'`

UnrealKazu, 2018-12-14:
It's typical that you hit a bump at the exact limit of the network size ([15 devices](https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size)), even though you do have routers in your network.

Can you inspect your network map and see if any of your end devices are using the routers? See the [wiki](https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttbridgenetworkmap) on how to retrieve the network map for your setup. You can display a graphviz map using [webgraphviz](http://www.webgraphviz.com/). I'm curious if it happens to be that all your end devices are still directly connected to the coordinator.

jonathanmusto, 2018-12-14:
I'm hitting a similar issue, except I'm at 19 devices.  Anything obviously out of the ordinary with my network map:
https://pastebin.com/1erQgrve
I seemed to start hitting issues after adding my first Tradfri Outlet (which i've since removed).

Koenkk, 2018-12-14:
Can you still add non-hue devices?

jonathanmusto, 2018-12-14:
Sorry, I hijacked the thread as my issue is similar. To the OP. I don't have any hue devices, but I can't add anymore of my Xaoimi devices either.

Imperial-Guard, 2018-12-15:
@koenk, since I don’t have another devices left Phillips Hue, I can’t try it with other devices.

Koenkk, 2018-12-15:
@Imperial-guard I guess the problem is the Hue bulb itself. I also have a Hue bulb which I cannot get paired. How did you factory reset the bulb?

Imperial-Guard, 2018-12-15:
@Koenkk I did the factory reset on the Hue light bulb with the remote, according to the video what's on YouTube. Hold the on and off button for certain time, etc.

jonathanmusto, 2018-12-16:
Ok, I did some tinkering tonight and the Tradfri Outlets are definitely the problem. I removed all my devices, reflashed the stick and started again. As soon as I add one Tradfri outlet, I cant add any other device. Even a restart of the zigbee2mqtt container or reboot of the host server doesn't solve it. Removing the device doesn't fix the issue either. I have to reflash the stick and start over. I guess I'll be sending them back for now.

Imperial-Guard, 2018-12-19:
> Ok, I did some tinkering tonight and the Tradfri Outlets are definitely the problem. I removed all my devices, reflashed the stick and started again. As soon as I add one Tradfri outlet, I cant add [...]

@Koenkk could this be reported as a bug?

Koenkk, 2018-12-19:
@Imperial-Guard if this happens with more tradfri outlets, yes, However more people have these device and this has never been reported before

jonathanmusto, 2018-12-19:
Is any data actually stored on the USB sniffer when you add devices?  

I wonder if my reflashing of the stick is a red herring and actually just pulling the stick out, rebooting and trying to add devices again is what gets it working.  I'll do some tinkering over the weekend, for £9 i'm determined to get these outlets to work!

Koenkk, 2018-12-19:
@johnnyletrois yes, the USB sniffer (coordinator) keeps track of all joined devices. The `database.db` is just for zigbee-shepherd.

Imperial-Guard, 2018-12-20:
I just re-installed my complete Zigbee network with all my devices. Yesterday I recieved my Osram Smart+ Plug, set allowed devices to join and directly the device was connected. 

So why I couldn't add my Hue lights the last time ..... I have still the feeling but maybe I'm wrong that the Hue lights don't respond and react that good compare with Tradfri lights.

jonathanmusto, 2018-12-21:
Ok, so I've made some progress with mine. 

It does look like the Ikea Outlets do cause issues and stop any other devices being added from time to time, but what resolves it for me is unplugging the USB sniffer, plugging it back in, then rebooting the host machine. After this, I'm able to pair more devices, including the outlets themselves. Some other things that may have been a factor too:
- I also cleaned up my database.db file from zigbee shepherd, by removing the duplicate rows
- I paired the plugs from a different location to my first attempt, so maybe they paired via a Router rather than directly on the coordinator??

I've now added 2x Mijia motion sensors and 3x Tradfri Outlets. Everything is working nicely.

I'm now rocking 23 devices on my network and will be added more over the coming weeks. It's been a little painful at times, but I've now succeeded in my goal of replacing SmartThings!!  Thanks @Koenkk for all the hard work, not only in writing the code but monitoring these forums! I'm looking forward to see what the future holds for this project.

RIMperator, 2019-01-08:
> It's typical that you hit a bump at the exact limit of the network size ([15 devices](https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size)), even though you do have routers in your network.

What's the typical behaviour when no additional devices can be added? I thought the limit of the network size was 20 devices (at least according to the documentation)?

In my case when trying to add devices, nothing happens and the log shows nothing as well, even after setting it to debug mode.

Koenkk, 2019-01-11:
> What's the typical behaviour when no additional devices can be added? I thought the limit of the network size was 20 devices (at least according to the documentation)?
>
> In my case when trying to add [...]

The limit of the current firmware is 15.
When this is reached, you will see behavior exactly as you described, nothing happens when pairing devices.

Perhaps this can also help you: https://github.com/Koenkk/Z-Stack-firmware/issues/23#issuecomment-452839123

gurbyz, 2019-02-10:
> unplugging the USB sniffer, plugging it back in, then rebooting the host machine

This fixed it for me. (I wanted to add 4 Ikea Tradfri E1603 switches. The first two succeeded without issues. The third and fourth wouldn't succeed without the suggestion of @jonathanmusto )

MrBramme, 2019-02-16:
I've had the exact same issue after adding Ikea outlets. the fix as suggested by @jonathanmusto fixed it for me as well

iFloris, 2019-02-24:
After being stumped as to why I was unable to add any devices at all, I found this thread. I started with adding an Ikea outlet, because I thought that would help range. This seems to have failed, and I have been unable to add any subsequent device at all. I am going to try the fix suggested by @jonathanmusto now!

PedroLamas, 2019-03-06:
Seeing the same issue reported here with my IKEA TRADFRI outlet: after adding one, I can't add anything else... I'm not home right now to check on @jonathanmusto proposed solution, but will try that later today and update here accordingly.

I only bought the IKEA outlets, so I didn't get an IKEA Hub but this made me think that maybe there's some firmware update to fix this issue?

I can't find any information on IKEA firmware updates, so can anyone with both the IKEA Hub and outlets confirm if there's any firmware updates?

PedroLamas, 2019-03-06:
Confirming now, the workaround solution presented fixed the problem of not being able to register new devices (unplug usb -> plug again -> reboot).

Had to turn off and on the IKEA tradfri outlets to get them online (really!) and I'm now waiting for the whole mesh to rebuild... offtopic, but anyone knows a way to force this?

rachetfoot, 2019-03-17:
I've got this issue too, but reconnecting the dongle and restarting doesn't help. I managed to connect one outlet, but after that nothing will pair, even devices that have connected before.

MrBramme, 2019-03-17:
I tried it again yesterday and it worked fine on an Intel NUC running Ubuntu. I had the issue on a Raspberry Pi 3 running raspbian

rachetfoot, 2019-03-18:
After another restart cycle or two I got things working, although I still have a Aqari switch that unjoined and refuses to join again.

RIMperator, 2019-04-14:
> After another restart cycle or two I got things working, although I still have a Aqari switch that unjoined and refuses to join again.

Also have approx. 3 Aqara devices that lost connection and I haven't found a way to get them back. Were you able to solve this issue?

Damme, 2019-05-17:
I'm using this issue ticket instead of creating a new one.

I am running into a problem that after a few days I am totally unable to add new devices, I don't get any log output at all then trying to pair. My map also locks a bit weird imo so I'll attach that here.

I recently upgraded all controller and router firmware to latest, I'm running latest dev zigbee2mqtt.
I removed all devices the other day and started re-adding, beginning with router and a few ikea outlets.
Most devices are not added directly to the controller but other routers/outlets/lamps.

enter https://pastebin.com/CW2ECbrw into http://www.webgraphviz.com/

closed 2019-07-23 by stale[bot]

rubdos, 2019-11-01:
I just bumped on the Trådfri outlet issue too. Shouldn't a note be added to [https://www.zigbee2mqtt.io/devices/E1603_E1702.html](the zigbee2mqtt Supported Devices page)?

Otah, 2020-01-25:
I observe a very same thing after I have added IKEA Symfonisk volume control. I tried all the tricks except of flashing the stick again.

My total number of paired devices is just 9, according to the network map majority of them going through routers, not the coordinator.

It seems more and more as some IKEA-related issue. @Koenkk isn't there a way how to debug this? There is not a single character written in the zigbee2mqtt log (even in debug mode) when I try to pair something now. Is there any way how to get to the underlying zigbee messages, for example?

Koenkk, 2020-01-26:
@otah that means the cc2531 does even pick it up, please check https://www.zigbee2mqtt.io/information/FAQ.html#no-logging-is-shown-at-all

bjschuit, 2020-01-26:
Same issue here. Cannot add either light bulbs (Gledopto) or switches (Ikea and Aqara Opple). Already restarted Zigbee2MQTT multiple times, reinserted the dongle, updated to 1.9.0 from 1.8.0, all to no avail yet.

Otah, 2020-01-26:
@Koenkk thanks for the link, however I already have/tried everything in the list except of flashing again, so far.

The point is that I don’t understand how I could hit the limit if I only have and ever had 9 devices and as I said they are even connected through routers in most cases.

I will try flashing and see, but I am still kinda suspicious whether some ikea signals cannot get the entire system to some inconsistent state.

bjschuit, 2020-01-26:
> 
> 
> Same issue here. Cannot add either light bulbs (Gledopto) or switches (Ikea and Aqara Opple). Already restarted Zigbee2MQTT multiple times, reinserted the dongle, updated to 1.9.0 from 1.8.0, all to no avail yet.

I swapped CC2531 stick with same firmware. Now it does work... So probably a reflash would do the trick.

marcetad, 2020-01-29:
> @Koenkk thanks for the link, however I already have/tried everything in the list except of flashing again, so far.
>
> The point is that I don’t understand how I could hit the limit if I only have and [...]

I have the same issue after adding 1 ikea device(E1766). Added +-23 xiaomi devices. All without problems. After adding ikea device I experience problems. Cannot add more xiaomi devices or other ikea devices. Interview doesn't start..

rosscullen, 2020-02-08:
> Ok, so I've made some progress with mine.
>
> It does look like the Ikea Outlets do cause issues and stop any other devices being added from time to time, but what resolves it for me is unplugging the [...]

Thanks @jonathanmusto this followed your instructions and this appears to have done the trick for me too!

marcetad, 2020-02-11:
> I have the same issue after adding 1 ikea device(E1766). Added +-23 xiaomi devices. All without problems. After adding ikea device I experience problems. Cannot add more xiaomi devices or other ikea [...]

Issue fixed, nothing to do with ikea devices.
One of the devices failed the interview. After that interviewing didn't start anymore.
After deleting the specific device from the database interviewing started again.

MixelF, 2022-10-02:
same issues here. After adding a tradfri outlet, joining stops.