Koenkk/zigbee2mqtt #698: Recommendation of a remote control

tringler, 2018-12-13:
Hello,

I know this is not an issue, but I didn't find a forum or similar community platform. Can anyone recommend a remote, which is or can be supported and is able to dim lights.

I would like to have the IKEA or Hue remotes, but as far as I seen the IKEA remote can only bound to an IKEA bulb and for the Hue remote only on/off is supported.

Can anyone share experience with me? 

What about Trust Zigbee Remote Control ZYCT-202? Did anyone tested it?

Thanks, Thomas

johnnyletrois, 2018-12-13:
I use the Xiaomi round buttons that support single, double, triple, quadruple, long press, and release. Single tap toggles light on/off. Release after press and hold sets brightness level, ie a long hold makes light brighter and a short hold makes it dimmer. 

The brightness is not exact or ideal.  I tried changing brightness real time based on length of hold but too many messages flooded zogbee2mqtt and it wasn’t reliable.

khzd, 2018-12-13:
What about Xiaomi cube
https://m.youtube.com/watch?v=5YtqG1wEnng
I have a couple of them, use ha to control them

tringler, 2018-12-13:
I have the cube and the round buttons - I don't like the haptic of the buttons, but they are OK. I cannot imagine to use the cube in "production" - For me it's more a tech demo. I'm not really using the cube.

I like these remotes which I can mount on the wall, so I have my wallswitch and my remote control in one device. Currently I'm using HomeMatic wall switches, but I really like the idea of a removable wall switch.

clockbrain, 2018-12-14:
@tringler  all the buttons on the Hue remote work fine. You would need to set up home assistant automations to map the button mqtt messages to specific on/off/dim functions but that isn't too hard. See [https://github.com/Koenkk/zigbee2mqtt/issues/344#issuecomment-443482463](https://github.com/Koenkk/zigbee2mqtt/issues/344#issuecomment-443482463) or [https://community.home-assistant.io/t/zigbee2mqtt-automation-help/79402](https://community.home-assistant.io/t/zigbee2mqtt-automation-help/79402)

tringler, 2018-12-14:
Oh! Great news! Should we update the wiki then? For me it looks like that the switch is just supporting on/off, but indeed it looks like it works completely.

mcsSolutions, 2018-12-20:
I have been working with RGBgenie which has 2 all on/off buttons, 4 on/off/dim and 2 scene select.  The on/off/dim are functional via zigbee2mqtt, but still working the others.  There is a recent thread on this forum for thi device.

closed 2019-02-25 by stale[bot]

james-fry, 2019-03-28:
> @tringler all the buttons on the Hue remote work fine. You would need to set up home assistant automations to map the button mqtt messages to specific on/off/dim functions but that isn't too hard. See [#344 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/344#issuecomment-443482463) or https://community.home-assistant.io/t/zigbee2mqtt-automation-help/79402

Hue dimmer is even better since the change to support long click, long release and multi-clicks.