Koenkk/zigbee2mqtt #699: Binding MCCGQ01LM - Xiaomi MiJia door & window contact sensor

fishpepper, 2018-12-13:
First of all thanks for your great work on the zigbee stuff!
I was wondering what the correct way to bind the (old rounded) Xiaomi  door and window sensors.

The readme says:
> Most MiJia devices: press and hold the reset button on the device for +- 5 seconds (until the blue light starts blinking). IMPORTANT: Release and start pressing the reset button every second. This keeps the device awake, otherwise pairing will fail!.

However this does not work. I managed to bind 8 sensors after minutes of random button pressing. The first 5s seems to be important. Pressing every second seems to be wrong.
I am still struggling with the last two sensors...
I could not find out what the correct button press frequency is. Any hints?
Do I have to be close to the gateway (cc2531 stick)?

> 2018-12-13 21:04:49 - info: Coordinator firmware version: '20180815'
> 2018-12-13 21:04:49 - debug: zigbee-shepherd info: {"enabled":true,"net": "state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00xxxxxxxxxxxxxxxx","nwkAddr":0},"firmware": "transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1544731489,"joinTimeLeft":0}

fishpepper, 2018-12-13:
Update: I managed to bind one more device after one hour of waiting.
Sequence: 5s long press, every second a short press:

```
2018-12-13 22:04:35 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_magnet"}}' of device 'undefined' (0x00158d0002c52fc8)
2018-12-13 22:04:35 - info: New device with address 0x00158d0002c52fc8 connected!
2018-12-13 22:04:35 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d0002c52fc8"}'
2018-12-13 22:04:38 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_magnet"}}' of device 'lumi.sensor_magnet' (0x00158d0002c52fc8)
2018-12-13 22:04:44 - debug: Received zigbee message of type 'devInterview' with data '"0x00158d0002c52fc8"'
2018-12-13 22:04:48 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"modelId":"lumi.sensor_magnet"}}' of device 'lumi.sensor_magnet' (0x00158d0002c52fc8)
2018-12-13 22:04:48 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2018-12-13 22:04:49 - debug: Received zigbee message of type 'devInterview' with data '"0x00158d0002c52fc8"'
2018-12-13 22:04:53 - debug: Received zigbee message of type 'devInterview' with data '"0x00158d0002c52fc8"'
2018-12-13 22:04:57 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'lumi.sensor_magnet' (0x00158d0002c52fc8)
```

The second missing device timed out again and does not trigger the initial incoming device message any more. I will try again tomorrow. It seems that something is in an undefined state after an unsuccessful try and the timeout (?) is quite long.

```
2018-12-13 22:07:01 - error: Cannot get the Node Descriptor of the Device: 0x00158d0002c66e52 (Error: Timed out after 10000 ms)
2018-12-13 22:07:21 - error: Cannot get the Node Descriptor of the Device: 0x00158d0002c66e52 (Error: Timed out after 10000 ms)
2018-12-13 22:07:41 - error: Cannot get the Node Descriptor of the Device: 0x00158d0002c66e52 (Error: Timed out after 10000 ms)
2018-12-13 22:08:03 - error: Cannot get the Node Descriptor of the Device: 0x00158d0002c66e52 (Error: Timed out after 10000 ms)
```
There was no device incoming message for this one

Koenkk, 2018-12-14:
`Cannot get the Node Descriptor of the Device: 0x00158d0002c66e52` is solved by retarting zigbee2mqtt and trying again.

fishpepper, 2018-12-14:
I still can not get the last one to bind. No more incoming device, no more Cannot get Node descriptor messages.
Currently I have 18 devices connected. Is there a limit i might run into?

Do you know what the other blinkcodes mean?
I can get the device to emit  ~25really short pulses. Sometimes I also get a ~1s long led light. All by random button pressing. 
I can do the same with all Window/Door sensors... So it's unlikely that the sensor is bad.
It was really annoying to bind the other door sensors. Try'n'Error and sometimes it works.

The other sensors (Temp+Hum and Motion) always work on the first try. Only the Window/Door sensors seem to be really hard to bind...

Any ideas?

jonathanmusto, 2018-12-14:
Hey, so this is really weird.  I'm currently stuck at around 18 devices too and can't add any more either. 
My network is a mixture of Xiaomi Door sensors (Mijia and Aqara) (x7), Xiaomi Mijia Buttons (x4) and IKEA Tradfri bulbs (x8).  I just bought 4x IKEA Outlets, I was able to add 1 of them, to get me up to 20, but then, no more. During debugging, I removed the 1  Outlet I added, but now I can no longer re-add it, or add any other Xiaomi Sensors. It's like joining is is no longer allowed, nothing is showing up debug and allow_join is on.

I hit the same problem on the CC2531ZNP-Prod.hex firmware, so i'm now trying the CC2531ZNP-Prod-44-devices.hex firmware but have hit the same limit.

Any tips for debugging and increasing the size of the network?  It seems like lots of people have been able to reach in excess of 40 devices inc various routers.

Koenkk, 2018-12-14:
@fishpepper can you show your startup log? To check what kind of devices you have in your network.

@jonathanmusto do you even run into this problem with the 44 devices firmware?

jonathanmusto, 2018-12-14:
Hey @Koenkk, yes same issue with the 44 device firmware. I'm going to do some more playing around tomorrow, try removing some devices and then adding new ones, see if I can figure anything out.

Koenkk, 2018-12-14:
if you run into issues after adding only 20 devices using the 44 devices firmware its definitely not a network size issue.

jonathanmusto, 2018-12-15:
Yeah, that's what I thought. My problems began after adding my first Tradfri  Outlet (UK) to the system. It added fine, but then I couldn't add any more of them, or any other devices. So I removed it, but then couldn't add it again or any other devices.

I think tomorrow I'll try reflash my stick again, then try adding the Tradfri outlets first and see what happens.

fishpepper, 2018-12-15:
I think my problem is not related to the plain number of devices but binding the MCCGQ01LM itself.
As said before, it was a mess and really annoying to add the first sensors already. I had to do random presses, the 5s and then every 1s did not always work.
For some it did, for others not. Interestingly the original Xiaomi hub connects to them directly after pressing the button for 5s.

Anyway, here is my startup log:
`2018-12-15 13:29:59 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2018-12-15.13-29-59'
2018-12-15 13:29:59 - debug: Removing old log directory '/opt/zigbee2mqtt/data/log/2018-12-14.19-54-54'
2018-12-15 13:30:00 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
2018-12-15 13:30:00 - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2018-12-15 13:30:00 - info: Starting zigbee2mqtt version 0.2.0 (commit #b0d3c2f)
2018-12-15 13:30:00 - info: Starting zigbee-shepherd
2018-12-15 13:30:01 - info: zigbee-shepherd started
2018-12-15 13:30:01 - info: Coordinator firmware version: '20180815'
2018-12-15 13:30:01 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0009eefc1d","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1544877001,"joinTimeLeft":0}
2018-12-15 13:30:01 - info: Currently 18 devices are joined:
2018-12-15 13:30:01 - info: 0x00158d00015ad043 (0x00158d00015ad043): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d0001268e77 (0x00158d0001268e77): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d0001dcc7b4 (0x00158d0001dcc7b4): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d000156de37 (0x00158d000156de37): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d0001b91228 (0x00158d0001b91228): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d0001b79107 (0x00158d0001b79107): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d00014d4fd1 (0x00158d00014d4fd1): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d0001537644 (0x00158d0001537644): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d0001a21c9e (0x00158d0001a21c9e): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d00013097da (0x00158d00013097da): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d00016c7b25 (0x00158d00016c7b25): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d00012cd7d1 (0x00158d00012cd7d1): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d00012d9844 (0x00158d00012d9844): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-12-15 13:30:01 - info: 0x00178801034476bf (0x00178801034476bf): 8718696449691 - Philips Hue White Single bulb B22 (Router)
2018-12-15 13:30:01 - info: 0x00178801035e5ef0 (0x00178801035e5ef0): 8718696449691 - Philips Hue White Single bulb B22 (Router)
2018-12-15 13:30:01 - info: 0x00158d000205a2be (0x00158d000205a2be): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d0002c52fc8 (0x00158d0002c52fc8): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-12-15 13:30:01 - info: 0x00158d0002c6684b (0x00158d0002c6684b): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
2018-12-15 13:30:01 - warn: `permit_join` set to  `true` in configuration.yaml.
2018-12-15 13:30:01 - warn: Allowing new devices to join.
2018-12-15 13:30:01 - warn: Set `permit_join` to `false` once you joined all devices.
2018-12-15 13:30:01 - info: Zigbee: allowing new devices to join.
2018-12-15 13:30:01 - info: Connecting to MQTT server at mqtt://localhost
2018-12-15 13:30:01 - info: zigbee-shepherd ready
2018-12-15 13:30:01 - info: Connected to MQTT server
2018-12-15 13:30:01 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
`

Koenkk, 2018-12-15:
is the xiaomi hub still powered on? Having multiple zigbee networks is known to cause pairing problems.

jonathanmusto, 2018-12-17:
I did some troubleshooting on my Network, it turns out that adding just one Ikea Tradfri Outlet causes my entire network to no longer pair with any new devices. @fishpepper do you have any of these Tradfri outlets?

As far a pairing the Mijia devices, I never really had any problems with them. Just hold the reset pin until you get the 3 flashes then keep hitting the reset pin every 0.5s until it starts to pair, i keep hitting reset until i get the final pairing message in the log. I only have contact sensors and buttons though.

fishpepper, 2018-12-19:
No, no tradfi devices on my net.

I just got two of the new rectangular window and door sensors from xioami (MCCGQ11LM).
Binding was no problem and done within some seconds. Interestingly I was able to bind the old one that
I unsucessfully tried to bind for the last few days afterwards! 
Really strange.

This could explain why I might always have had a bad time adding the last device. When I was adding multiple devices I simply switched when one was not working and tried another one.

Really strange. I have 21 devices connected now.

AlexBoosh, 2019-01-03:
> I just got two of the new rectangular window and door sensors from xioami (MCCGQ11LM).
> Binding was no problem and done within some seconds.

Hello. I have a rectangular sensor (MCCGQ11LM) too and... I don't can pairing it. This is the first and only sensor. Please, tell me the sequence of your actions for the pairing. Thank you!

jonathanmusto, 2019-01-03:
> > I just got two of the new rectangular window and door sensors from xioami (MCCGQ11LM).
> > Binding was no problem and done within some seconds.
> 
> Hello. I have a rectangular sensor (MCCGQ11LM) too and... I can't pairing it. This is the first and only one my sensor. Please, tell me the sequence of your actions for the pairing. Thank you!

Aqara? You just need to push and hold the reset button for 3-4s, once the light starts blinking let go. It should add within 20-30s or so.  I had some problems getting anything to pair initially after first setup, I did the following to troubleshoot:
1. Remove and reinsert the USB sniffer, reboot the host PC
2. Press the reset button in the USB sniffer (once nearest to the USB end), reboot the host PC
3. Try the sniffer in a different USB socket on the host PC, reboot the host PC

AlexBoosh, 2019-01-03:
> Aqara?
Yes. Aqara.
I try your solution - it's miracle!!! It's work!
My PC is Rock64 with 2 USB 2.0 and 1 USB 3.0 ports.
CC2531 was connected on one of USB 2.0.
First - I rebooted my Rock64. No result.
Second - I changed USB 2.0 port one to another via shutdown my Rock64 - no result.
And last - I changed USB to USB 3.0 via shutdown - and ta-dam!

```
  zigbee2mqtt:info 1/3/2019, 3:52:55 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:error 1/3/2019, 3:53:18 PM Cannot get the Node Descriptor of the Device: 0x00158d000275b83e (Error: Timed out after 10000 ms)
  zigbee2mqtt:info 1/3/2019, 3:53:34 PM Connecting with device...
  zigbee2mqtt:info 1/3/2019, 3:53:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 1/3/2019, 3:53:34 PM Message without device!
  zigbee2mqtt:info 1/3/2019, 3:53:37 PM Connecting with device...
  zigbee2mqtt:info 1/3/2019, 3:53:37 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 1/3/2019, 3:53:37 PM Message without device!
  zigbee2mqtt:info 1/3/2019, 3:53:41 PM Connecting with device...
  zigbee2mqtt:info 1/3/2019, 3:53:41 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 1/3/2019, 3:53:41 PM Message without device!
  zigbee2mqtt:info 1/3/2019, 3:53:44 PM Connecting with device...
  zigbee2mqtt:info 1/3/2019, 3:53:44 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 1/3/2019, 3:53:44 PM Message without device!
  zigbee2mqtt:info 1/3/2019, 3:53:47 PM Connecting with device...
  zigbee2mqtt:info 1/3/2019, 3:53:47 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:warn 1/3/2019, 3:53:47 PM Message without device!
  zigbee2mqtt:info 1/3/2019, 3:53:47 PM Device incoming...
  zigbee2mqtt:info 1/3/2019, 3:53:47 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 1/3/2019, 3:53:47 PM New device with address 0x00158d000275b83e connected!
  zigbee2mqtt:info 1/3/2019, 3:53:47 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d000275b83e"}'
  zigbee2mqtt:info 1/3/2019, 4:03:21 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000275b83e', payload '{"contact":true,"linkquality":86}'
  zigbee2mqtt:info 1/3/2019, 4:03:24 PM MQTT publish: topic 'zigbee2mqtt/0x00158d000275b83e', payload '{"contact":false,"linkquality":94}'

```
Many thanks! I tryed pairing 4 days...)

After that I change USB back to USB 2.0 port - and all work.

jonathanmusto, 2019-01-03:
> Many thanks! I tryed pairing 4 days...)
> 
> After that I change USB back to USB 2.0 port - and all work.

Great, i'm glad you got it all working! 

I'm currently at around 25 devices and still adding. I have a mixture of Ikea Tradfri Bulbs, Ikea Tradfri Outlets, Xioami Sensors (Contact (Aqara, Mijia), Motion (Mijia)). Everything is working well in HA and all very stable!

alinelena, 2019-02-10:
i doubt the usb 2.0 vs 3.0 may have too many issues. I have also doubt the tradfiri outlets are the issue.
Once network grows in size seems to be more difficult to pair devices, especially battery powered ones.
i tried once and left the motion sensors last to pair.. a nightmare, reflashed and tried again... this time i left the contact sensors last... again nightmare to pair... still have 4 unpaired.
my network is 12 tradfiri bulbs, 1 tradfiri socker, 1 xiaomi socker, 4 motions sensors, 5 contact, 3 temperature, 3 button switches.
I tried to add an explicit router cc2531 but somehow the network looks sluggish once the router is in.

jossie67, 2019-03-14:
I had the same issue.
I solved it with  5s long press, blue led is flashing and then every second a short press:

way2many, 2019-04-11:
I seem to have a problem with one of my door contact sensors. I bought 3 at the same time, all from the same production date according to the sticker on the box. It is stuck in "device incoming" and I have tried everything suggested here but no dice. I'm running latest docker dev on max stability.
I think it has a fault and might try to get a refund for it.

vogler, 2019-04-13:
Same. I had to copy `,"manufName":"LUMI", "powerSource":"Battery","modelId":"lumi.sensor_magnet.aq2"`  from another sensor in `database.db` to make it work.

closed 2019-06-19 by stale[bot]

mimik13, 2023-01-03:
> I had the same issue. I solved it with 5s long press, blue led is flashing and then every second a short press:

Same thing worked for me. 👍