Koenkk/zigbee2mqtt #700: Docker container does not support creating the network graph

cynicated, 2018-12-14:
I created a Dockerfile that would pull in graphviz, imagemagick and mosquitto-clients with the hope that I could have the container automatically provide the Zigbee network graph every 10 minutes or so. I figured this would be an ideal solution and could even be done out of the box if anyone so desired.

However, I ran into an issue where the graphs were not spaced correctly. Digging in to the error logs, I saw this:

`Error: remove_overlap: Graphviz not built with triangulation library`

After doing some research, it appears that the Alpine Linux graphviz package was not compiled with the --with-gts configuration. It also appears that the appropriate library does not exist in Alpine Linux anyway (libgts-dev).

Koenkk, 2018-12-14:
Are you able to compile the packages yourself (in the docker image?)

UnrealKazu, 2018-12-14:
I've always been able to generate correct graphviz maps in my docker setup ever since the map functionality was implemented, so this does work for me in Docker. Could it be that the problem is specific for your setup (your paired devices I mean, not your hardware setup)?

cynicated, 2018-12-15:
@UnrealKazu - to clarify, I can correctly create the graphviz maps raw code, but converting that using Graphviz doesn't work for me since the Alpine base that zigbee2MQTT is built upon doesn't have graphviz installed. So I went ahead and installed it, as well as imagemagick and mosquitto-clients. Which all went fine. Until I actually went to run the command to convert the graphviz map (provided by request) into a .png. At that point, the error was thrown.

What docker image are you using? I'm using the version for the Raspberry Pi (arm32v6-dev).

cynicated, 2018-12-15:
I've demonstrated here:

https://gist.github.com/cynicated/969155dcb322820162eeb2fb2d2d0d4e

And here's the resulting image

https://pasteboard.co/HRMYRPU.png

danpowell88, 2018-12-17:
Is there a good way to use the existing dockerfile but somehow get the graphwiz stuff included, dont really want to have to maintain my own docker file if i can help it

closed 2019-02-22 by stale[bot]

epicurean70, 2019-06-13:
Hi,
Does anyone have a docker link they can share that provides the graphivz link ability, to a zigbee2mqtt (also installed as a docker container)?