Koenkk/zigbee2mqtt #701: ikea tradfri dimmer

h4nc, 2018-12-14:
My tradfri dimmer worked fine for about a week, but it stopped working yesterday.

I tried reconnecting it, but after is is connected again I get an error like this:

zigbee2mqtt:error 2018-12-14 12:55:45 Failed to configure Zigbee Dimmer WZ 0x00XXXXX.

I also changed the battery.
I don't know what else I can do. Any suggestions welcome!

tutis51, 2018-12-14:
I have the same issue with this dimmer, though it works (with HA automation), but gives the same error:

Dec 07 22:03:34 orangepizero npm[7391]: { groupid: 0,
Dec 07 22:03:34 orangepizero npm[7391]:   clusterid: 8,
Dec 07 22:03:34 orangepizero npm[7391]:   srcaddr: 12800,
Dec 07 22:03:34 orangepizero npm[7391]:   srcendpoint: 1,
Dec 07 22:03:34 orangepizero npm[7391]:   dstendpoint: 1,
Dec 07 22:03:34 orangepizero npm[7391]:   wasbroadcast: 0,
Dec 07 22:03:34 orangepizero npm[7391]:   linkquality: 47,
Dec 07 22:03:34 orangepizero npm[7391]:   securityuse: 0,
Dec 07 22:03:34 orangepizero npm[7391]:   timestamp: 2702552,
Dec 07 22:03:34 orangepizero npm[7391]:   transseqnumber: 0,
Dec 07 22:03:34 orangepizero npm[7391]:   len: 5,
Dec 07 22:03:34 orangepizero npm[7391]:   data: { '0': 17, '1': 70, '2': 1, '3': 1, '4': 195 },
Dec 07 22:03:34 orangepizero npm[7391]:   zclMsg:
Dec 07 22:03:34 orangepizero npm[7391]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Dec 07 22:03:34 orangepizero npm[7391]:      manufCode: 0,
Dec 07 22:03:34 orangepizero npm[7391]:      seqNum: 70,
Dec 07 22:03:34 orangepizero npm[7391]:      cmdId: 'move',
Dec 07 22:03:34 orangepizero npm[7391]:      payload: { movemode: 1, rate: 195 } } }
Dec 07 22:03:34 orangepizero npm[7391]: cmd { groupid: 0,
Dec 07 22:03:34 orangepizero npm[7391]:   clusterid: 8,
Dec 07 22:03:34 orangepizero npm[7391]:   srcaddr: 12800,
Dec 07 22:03:34 orangepizero npm[7391]:   srcendpoint: 1,
Dec 07 22:03:34 orangepizero npm[7391]:   dstendpoint: 1,
Dec 07 22:03:34 orangepizero npm[7391]:   wasbroadcast: 0,
Dec 07 22:03:34 orangepizero npm[7391]:   linkquality: 47,
Dec 07 22:03:34 orangepizero npm[7391]:   securityuse: 0,
Dec 07 22:03:34 orangepizero npm[7391]:   timestamp: 2702552,
Dec 07 22:03:34 orangepizero npm[7391]:   transseqnumber: 0,
Dec 07 22:03:34 orangepizero npm[7391]:   len: 5,
Dec 07 22:03:34 orangepizero npm[7391]:   data: { '0': 17, '1': 70, '2': 1, '3': 1, '4': 195 },
Dec 07 22:03:34 orangepizero npm[7391]:   zclMsg:
Dec 07 22:03:34 orangepizero npm[7391]:    { frameCntl: { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 1 },
Dec 07 22:03:34 orangepizero npm[7391]:      manufCode: 0,
Dec 07 22:03:34 orangepizero npm[7391]:      seqNum: 70,
Dec 07 22:03:34 orangepizero npm[7391]:      cmdId: 'move',
Dec 07 22:03:34 orangepizero npm[7391]:      payload: { movemode: 1, rate: 195 } } }
Dec 07 22:03:34 orangepizero npm[7391]:   zigbee2mqtt:warn 2018-12-7 22:03:34 No converter available for 'ICTC-G-1' with cid 'genLevelCtrl', type 'cmdMove' and data '{"cid":"genLevelCtrl","data":{"movemode":1,"rate":195}}'
Dec 07 22:03:34 orangepizero npm[7391]:   zigbee2mqtt:warn 2018-12-7 22:03:34 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
Dec 07 22:03:34 orangepizero npm[7391]:   zigbee2mqtt:info 2018-12-7 22:03:34 MQTT publish, topic: 'zigbee2mqtt/Ikea_dimmer', payload: '{"brightness":255}'
Dec 07 22:03:34 orangepizero npm[7391]:   zigbee2mqtt:info 2018-12-7 22:03:34 MQTT publish, topic: 'zigbee2mqtt/Ikea_dimmer', payload: '{"brightness":223}'

Koenkk, 2018-12-14:
Is this issue still present in the dev branch? Probably has already been fixed in https://github.com/Koenkk/zigbee2mqtt/issues/553

h4nc, 2018-12-14:
I use zigbee2mqtt as an addon for hass.io. Do I have an option to run the dev branch with that setup? Is it zigbee2mqtt-edge?

Koenkk, 2018-12-14:
Yes it's edge.

h4nc, 2018-12-14:
thanks. Do I have to reconnect all my devices agaon for edge, or is it possible to migrate?

h4nc, 2018-12-15:
I did not try edge yet, because it might be a lot of work to set up all my sernsors again. Also in the linked post they talk about "no converter available"
But I think what causes by problems is this error:

zigbee2mqtt:error 2018-12-14 12:55:45 Failed to configure Zigbee Dimmer WZ 0x00XXXXX.

I tried to reconnect it serveral times. Sometimes it works for a few mintues or hours but ends with this error.

phoxy84, 2018-12-15:
I thought that Indid not have to setup sensors again after i went to zigbee-edge. Only copied the config. It’s not guaranteed....

Koenkk, 2018-12-15:
@h4nc should be fixed in 1.0.0/dev branch. `Failed to configure` happens because the device is probably sleeping. However, succeeding one time is enough.

h4nc, 2018-12-15:
@Koenkk 
> Failed to configure happens because the device is probably sleeping.
So I think it goes to sleep when it's not used for some time. The problem is that it doesn't seem to wake up, because I could not see events. 
Could it be a connection issue? (It worked for some days, don't understand why it stopped working)

> However, succeeding one time is enough.
What do you mean? Succeeding one time is enough too?

Do you know when the fix will come to the stable version?

Koenkk, 2018-12-15:
It could be that the device drops of the network due to bad connection.

The no converter fix is included in 1.0.0 (released today).

The device needs some initial configuration when paired which only has to be done once (however I doens't harm if this is done multiple times).

h4nc, 2018-12-15:
I have it as addon in hassio. It says version 0.2.0
Where can I find the real version number to knwo which version I do run
.
> The no converter fix is included in 1.0.0 (released today).

Just to avoid misunderstanding... you don't mean the dev version, you mean the "official" one, right?

> The device needs some initial configuration when paired which only has to be done once (however I doens't harm if this is done multiple times).
I did that and it worked perfectly.

Koenkk, 2018-12-16:
1.0.0 for hassio isn't out yet, just use the edge for now.

h4nc, 2018-12-24:
I have 1.0.1 running. I still have issues with my ikea dimmer. 

I deleted it with` zigbee2mqtt/bridge/config/remove` and I tried to reconnect it. I took a new battery. 
Now I only get the warning "Message without device" and it won't connect. All my other devices (11) work.

Morphy99, 2019-01-16:
I can't get the dimmer connected, running latest firmware and zigbee edge freshly installed today. Pressing the button 4 times did nothing but this: 

zigbee2mqtt:error 2019-1-16 22:51:00 Cannot get the Node Descriptor of the Device: 0x000b57fffe208e8f (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 2019-1-16 22:51:20 Cannot get the Node Descriptor of the Device: 0x000b57fffe208e8f (Error: Timed out after 10000 ms)

pressing and holding for 10 secs did this:

  zigbee2mqtt:debug 2019-1-16 22:51:36 Received zigbee message of type 'devLeaving' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:warn 2019-1-16 22:51:36 Message without device!
  zigbee2mqtt:debug 2019-1-16 22:51:51 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:51 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:51 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:52 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:52 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:52 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:53 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:53 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:53 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:54 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:54 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:54 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:54 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:54 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:54 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:55 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:55 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:55 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:56 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:56 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:56 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:56 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:56 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:56 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:57 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:57 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:57 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:58 Received zigbee message of type 'devInterview' with data '"0x000b57fffe208e8f"'
  zigbee2mqtt:info 2019-1-16 22:51:58 Connecting with device...
  zigbee2mqtt:info 2019-1-16 22:51:58 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:debug 2019-1-16 22:51:58 Received zigbee message of type 'devIncoming' with data '"0x000b57fffe208e8f"' of device 'TRADFRI wireless dimmer' (0x000b57fffe208e8f)
  zigbee2mqtt:info 2019-1-16 22:51:58 Device incoming...
  zigbee2mqtt:info 2019-1-16 22:51:58 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2019-1-16 22:51:58 New device with address 0x000b57fffe208e8f connected!
  zigbee2mqtt:info 2019-1-16 22:51:58 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000b57fffe208e8f"}'
  zigbee2mqtt:debug 2019-1-16 22:51:58 Configuring 0x000b57fffe208e8f (0x000b57fffe208e8f) ...
  zigbee2mqtt:error 2019-1-16 22:51:58 Failed to configure 0x000b57fffe208e8f (0x000b57fffe208e8f) ('No device')
  zigbee2mqtt:info 2019-1-16 22:51:58 MQTT publish: topic 'homeassistant/sensor/0x000b57fffe208e8f/brightness/config', payload '{"unit_of_measurement":"brightness","icon":"mdi:brightness-5","value_template":"{{ value_json.brightness }}","state_topic":"zigbee2mqtt/0x000b57fffe208e8f","json_attributes_topic":"zigbee2mqtt/0x000b57fffe208e8f","name":"0x000b57fffe208e8f_brightness","unique_id":"0x000b57fffe208e8f_brightness_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000b57fffe208e8f","name":"0x000b57fffe208e8f","sw_version":"Zigbee2mqtt 1.0.1","model":"TRADFRI wireless dimmer (ICTC-G-1)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[... cut ...]
  zigbee2mqtt:debug 2019-1-16 22:52:37 Received zigbee message of type 'devIncoming' with data '"0x000b57fffe208e8f"' of device 'TRADFRI wireless dimmer' (0x000b57fffe208e8f)
  zigbee2mqtt:info 2019-1-16 22:52:37 Device incoming...
  zigbee2mqtt:info 2019-1-16 22:52:37 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 2019-1-16 22:52:38 Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI wireless dimmer' (0x000b57fffe208e8f)
  zigbee2mqtt:debug 2019-1-16 22:53:10 Saving state to file /share/zigbee2mqtt/state.json

Morphy99, 2019-01-19:
> `zigbee2mqtt:debug 2019-1-16 22:52:38 Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI wireless dimmer' (0x000b57fffe208e8f)` looks like a successful pair.
> 
> When starting zigbee2mqtt, do you see a ` Successfully configured 0x000b57fffe208e8f...`, if you see a failed, rotate the dimmer while starting zigbee2mqtt.

Ah ok I wasn't seeing any activity but bizarrely after popping the battery back in a few days later it's working!

I do see ` zigbee2mqtt:error 2019-1-19 12:09:00 Failed to configure Dimmer (0x000b57fffe208e8f) ('Error: Timed out after 10000 ms')` and don't see `Successfully configured 0x000b57fffe208e8f...` however it is working:
`zigbee2mqtt:info 2019-1-19 12:09:43 MQTT publish: topic 'zigbee2mqtt/Dimmer', payload '{"brightness":208,"linkquality":49,"last_seen":"2019-01-19T12:09:43.963Z"}'
  zigbee2mqtt:info 2019-1-19 12:09:44 Zigbee publish to device '0xd0cf5efffe2fb85b', genLevelCtrl - moveToLevelWithOnOff - {"level":208,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2019-1-19 12:09:44 MQTT publish: topic 'zigbee2mqtt/Living Room', payload '{"state":"ON","last_seen":"2019-01-19T11:44:24.107Z","brightness":254}'
  zigbee2mqtt:info 2019-1-19 12:09:44 MQTT publish: topic 'zigbee2mqtt/Living Room', payload '{"state":"ON","last_seen":"2019-01-19T12:09:44.790Z","brightness":208}'`

Koenkk, 2019-01-19:
The configure only has to succeed once

h4nc, 2019-01-19:
I'd like to add that I also get that "Failed to configure" error sometimes but only for the ikea dimmer.

Whether or not I get that error in both cases the dimmer works.

wizzor, 2019-03-26:
Hi, I still have this issue (or I assume it's the same one), I get the
"PM Failed to configure" 
message in the logs, running 1.2.1 on Hass.io.

closed 2019-06-01 by stale[bot]