Koenkk/zigbee2mqtt #702: Power 'value' missing reporting 0.00 (For xiaomi plugs, wall switches same)

quarcko, 2018-12-14:
My lumi.plug's and wall switches sometimes do not report power values.
But in the debug log i clearly can see these values are there. 
Acoording to the log file, zigbee2mqtt first 'posts' a MQTT message with values
and only after posting it receives them from device? here is relevant part of log:

  zigbee2mqtt:info 2018-12-14 17:00:05 MQTT publish: topic 'zigbee2mqtt/0x00158d000249790c', payload '{"state_left":"ON","linkquality":89,"state_right":"ON","power":0,"consumption":0.06,"temperature":26}'
  zigbee2mqtt:debug 2018-12-14 17:00:05 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"61440":117440517,"onOff":1}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d000249790c)
  zigbee2mqtt:debug 2018-12-14 17:00:06 Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d000249790c)
  zigbee2mqtt:debug 2018-12-14 17:00:16 Received zigbee message of type 'attReport' with data '{"cid":"genAnalogInput","data":{"presentValue":6.329999923706055}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d000249790c)
  zigbee2mqtt:debug 2018-12-14 17:00:16 Received zigbee message of type 'devChange' with data '{"cid":"genAnalogInput","data":{"presentValue":6.329999923706055}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d000249790c)

As you can see it posts "power":0 but after a second there is a debug entry showing 
"presentValue":6.329999923706055 <- which is a power level.

Maybe there is needed some delay to wait for data before posting MQTT message?

quarcko, 2018-12-14:
Here is the moment when it finally catches something:

  zigbee2mqtt:debug 2018-12-14 17:25:27 Received zigbee message of type 'attReport' with data '{"cid":"genAnalogInput","data":{"presentValue":0.6299999952316284}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d000249790c)
  zigbee2mqtt:debug 2018-12-14 17:25:27 Received zigbee message of type 'devChange' with data '{"cid":"genAnalogInput","data":{"presentValue":0.6299999952316284}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d000249790c)
  zigbee2mqtt:debug 2018-12-14 17:25:36 Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"3":26,"5":15,"7":"0x0000000000000000","8":4383,"9":768,"100":1,"101":0,"149":0.06812231242656708,"152":0.6299999952316284,"154":16}}}' of device 'lumi.ctrl_ln2.aq1' (0x00158d000249790c)
  zigbee2mqtt:info 2018-12-14 17:25:36 MQTT publish: topic 'zigbee2mqtt/0x00158d000249790c', payload '{"state_left":"ON","linkquality":81,"state_right":"OFF","power":0.63,"consumption":0.07,"temperature":26}'

Now power is 0.63 (after floodlight went automatically off, and is in sleep mode)

It is clearly seen that
"presentValue":0.6299999952316284 in 'attReport' with data '{"cid":"genAnalogInput"
and
"152":0.6299999952316284 in 'attReport' with data '{"cid":"genBasic"

is same value. So device sometimes posts power level in separate 'genAnalogInput' message
but it seems that zigbee2mqtt is discarding it, and only reacts to 'genBasic' message.

So we have lost data, which is still there...

Koenkk, 2018-12-16:
see https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices_on_hassio.html

Koenkk, 2018-12-18:
Nice, could you perhaps make a PR? https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js