Koenkk/zigbee2mqtt #703: Co-ordinator not showing as connected

danpowell88, 2018-12-14:
I've got 2 cc2531s one as a coordinator and the other as the router

Pushing to home assistant it always shows the router to be disconnected.

The router is logging the following from the COM port:

> 0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
> 50610/0x001788010356
> 50610/0x00178801035609F4,NEIGHBOR
> 0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
> 50610/0x00178801035609F4,NEIGHBOR
> 0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
> 50610/0x00178801035609F4,NEIGHBOR
> 0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
> 50610/0x00178801035609F4,NEIGHBOR
> 0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
> 50610/0x00178801035609F4,NEIGHBOR
> 0/0x00124B0014BDEB58,COORD,Depth:0,Assoc:0
> 50610/0x00178801035609F4,NEIGHBOR

and the docker logs show

> [32m zigbee2mqtt:info[39m 2018-12-15 01:53:58 MQTT publish, topic: 'zigbee2mqtt/0x00124b001202345c', payload: '{"state":false,"linkquality":55,"description":"50610/0x00178801035609F4","type":"NEIGHBOR","rssi":80}'
> [32m zigbee2mqtt:info[39m 2018-12-15 01:53:58 MQTT publish, topic: 'zigbee2mqtt/0x00124b001202345c', payload: '{"state":false,"linkquality":55,"description":"0/0x00124B0014BDEB58","type":"COORD","rssi":51}'
> [32m zigbee2mqtt:info[39m 2018-12-15 01:53:58 MQTT publish, topic: 'zigbee2mqtt/0x00124b001202345c', payload: '{"state":false,"linkquality":55,"description":"50610/0x00178801035609F4","type":"NEIGHBOR","rssi":47}'
> [32m zigbee2mqtt:info[39m 2018-12-15 01:54:58 MQTT publish, topic: 'zigbee2mqtt/0x00124b001202345c', payload: '{"state":false,"linkquality":63,"description":"50610/0x00178801035609F4","type":"NEIGHBOR","rssi":47}'
> [32m zigbee2mqtt:info[39m 2018-12-15 01:54:58 MQTT publish, topic: 'zigbee2mqtt/0x00124b001202345c', payload: '{"state":false,"linkquality":63,"description":"0/0x00124B0014BDEB58","type":"COORD","rssi":60}'
> [32m zigbee2mqtt:info[39m 2018-12-15 01:54:58 MQTT publish, topic: 'zigbee2mqtt/0x00124b001202345c', payload: '{"state":false,"linkquality":63,"description":"50610/0x00178801035609F4","type":"NEIGHBOR","rssi":54}'

I'm guessing it is the fact that the state value is false, is my router not actually connected here or it is a bug?

Koenkk, 2018-12-14:
@ptvoinfo could you clarify what the `state` means?

danpowell88, 2018-12-14:
Sorry just making an assumption about the payload state attribute

payload: `'{"state":false,"linkquality":63,"description":"50610/0x00178801035609F4","type":"NEIGHBOR","rssi":54}'`

ptvoinfo, 2018-12-15:
@danpowell88 It means the state of the green light. If you press the SW button on the board the state should change.

danpowell88, 2018-12-15:
Ah ok, still doesnt ever seem to show as connected in HASS

Koenkk, 2018-12-15:
What makes you think its not connected?

danpowell88, 2018-12-15:
Hmm it may have actually not been connected,

Is there an explanation as to what the buttons do on the cc2531 router, i hit the one furthest from the usb end and it seemed to make the light stay on solid and now it shows as connected