Koenkk/zigbee2mqtt #708: Is there any way to set the level of precision for temp/humidity sensors?

mihalski, 2018-12-15:
I normally use one decimal point of precision for my temp/humidity sensors and I searched but could not find any way to do this with zigbee2mqtt.

I know I could manually do it in a Home Assistant template but this would prevent the use of MQTT auto-discovery.

Thanks in advance.

Koenkk, 2018-12-15:
Implemented, see [documentation](https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/configuration/device_specific_configuration.md) on how to use it.

mihalski, 2018-12-15:
Thank you!

Would it also make sense to have a global config equivalent to the device specific config in the main section of the configuration.yaml for when you always want parameters measured with a particular level of precision (including all future sensors)?

mihalski, 2018-12-15:
I also noticed that even with precision set to 1 if the value returned by the sensor happens to be a whole number it returns an int rather than a float (ie missing the .0)

justmvg, 2018-12-16:
I've installed 1.0.0 yesterday and made changes to the configuration.yaml to set the precision to 1 decimal instead of the default of 2. Restarted zigbee2mqt but still receiving data with 2 decimals.
Is there anything else I should change?

Koenkk, 2018-12-19:
- Precision for whole numbers is not possible in javascript (see https://www.w3schools.com/js/js_numbers.asp)
- Device type specific default can now be set via `device_options` (see https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/configuration/device_specific_configuration.md#changing-device-type-specific-defaults)

justmvg, 2018-12-19:
> @justmvg checkout the dev branch

Thank you. I managed to make it work with the dev branch.

mihalski, 2018-12-20:
Thank you so much for implementing device type specific defaults!
Granted, I don't really know what I am talking about and this is based on a quick google search, but couldn't you just use `.toFixed(precision)` where `precision` is a variable containing the number of decimal points of precision desired?

Koenkk, 2018-12-20:
The problem of `toFixed` is that it returns a string, the values should be kept numerical.

mihalski, 2018-12-20:
Does it matter when the intended output is a JSON string? Internally it can be numerical, but when being reported as part of a MQTT JSON string it could be considered seperate.

mihalski, 2018-12-20:
My understanding of MQTT was that value types were inferred from the presentation of the JSON string. But I am about as far from an expert in the field as is possible.

Koenkk, 2018-12-20:
Yes that's true, but this JSON string can contains string and numerical values.

mihalski, 2018-12-20:
I guess trying it and seeing what happens is the only ay to be sure.

mateuszdrab, 2019-01-07:
Hey guys, I'm running the latest version of zigbee2mqtt and after adding the following, I'm still seeing the messages get sent with default precision settings.

`device_options:
  temperature_precision: 1
  humidity_precision: 1
  pressure_precision: 1`

Koenkk, 2019-01-07:
@mateuszdrab are you using the dev branch?

mateuszdrab, 2019-01-07:
> @mateuszdrab are you using the dev branch?

I'm on master, makes sense.

closed 2019-03-08 by Koenkk

zerofruchtshake, 2019-03-15:
> * Device type specific default can now be set via `device_options` (see https://github.com/Koenkk/zigbee2mqtt/blob/dev/docs/configuration/device_specific_configuration.md#changing-device-type-specific-defaults)
The link is not working any more :(

Koenkk, 2019-03-15:
@zerofruchtshake http://www.zigbee2mqtt.io/configuration/device_specific_configuration.html#changing-device-type-specific-defaults

cigas4, 2019-09-02:
> 
> 
> @zerofruchtshake http://www.zigbee2mqtt.io/configuration/device_specific_configuration.html#changing-device-type-specific-defaults

Where should I put this:

device_options:
  occupancy_timeout: 130
  temperature_precision: 1

If I put this to HA configuration.yaml, I get error "Component error: device_options - Integration device_options not found."

Also tried HA Configuration - > Customization -> Entitty: Humidity sensor -> add other attribute -> name: humidity_precision value: 1 -> Reboot
But this also does not work.

KuChar91, 2020-07-11:
> Where should I put this:
>
> device_options:
> occupancy_timeout: 130
> temperature_precision: 1
>
> If I put this to HA configuration.yaml, I get error "Component error: device_options - Integration [...]

Put it in devices.yaml or configuration.yaml (add-on) not HA.

settler-als, 2021-02-12:
Hello,
i try use this options in configuration zigbee2mqtt (stable and edge):

device_options:
  retain: true
  temperature_precision: 1
  humidity_precision: 1

 temperature_precision: 1 - the option works and it can be seen in the interface lovelace or z2m web ui 
humidity_precision: 1 - this option does not work globally for all sensors.

When i check config in web ui -> Settings -> Raw  :
when I checked the config file in web ui -> Settings -> Raw , i see only:

 "device_options": {
            "temperature_precision": 1
        },

no  "retain: true" and "humidity_precision: 1".

Why?