Koenkk/zigbee2mqtt #712: Using transition from Home Assistant crashes with "cannot read 'cid' of undefined"

ashwinvandijk, 2018-12-16:
When I try to set an Ikea E14 bulb with a transition, zigbee2mqtt crashes. This problem came up after I updated to the latest version (1.0.0)

**Reproduce**
Send the following command with the `homeassistant.turn_on` service using Home Assistant:
```
{
  "entity_id": "light.zigbee_e14_1_light",
"brightness": 140,
"transition": 1 
}
```

**Logs**
```
zigbee2mqtt:debug 2018-12-16 15:32:33 Received MQTT message on 'zigbee/zigbee_e14_1/set' with data '{"state": "ON", "transition": 1, "brightness": 140}'
  zigbee2mqtt:debug 2018-12-16 15:32:33 Skipping 'state' because of 'brightness'
  zigbee2mqtt:info 2018-12-16 15:32:33 Zigbee publish to '0x000b57fffed8cd13', genLevelCtrl - moveToLevelWithOnOff - {"level":140,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 2018-12-16 15:32:34 MQTT publish: topic 'zigbee/zigbee_e14_1', payload '{"state":"ON"}'
/app/lib/extension/devicePublish.js:163

                            ieeeAddr, getConverted.cid, getConverted.cmd, getConverted.cmdType,

TypeError: Cannot read property 'cid' of undefined
    at queue.push (/app/lib/extension/devicePublish.js:163:52)
    at Queue.start (/app/node_modules/queue/index.js:143:17)
    at Queue.(anonymous function) [as push] (/app/node_modules/queue/index.js:59:12)
    at Timeout.setTimeout [as _onTimeout] (/app/lib/extension/devicePublish.js:161:32)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5)

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.0 start: `node index.js`
npm ERR! Exit status 1

npm ERR! Failed at the zigbee2mqtt@1.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-12-16T15_32_34_650Z-debug.log
Using '/app/data' as data directory
```

Koenkk, 2018-12-16:
After upgrading, did you do a `rm -rf node_modules && npm install`?

ashwinvandijk, 2018-12-16:
Forgot to mention, I'm running inside docker. I don't think I need to run these commands inside a container, right?

Aejoh, 2018-12-16:
I have the same issue, and I also use Docker.

Maximus48p, 2019-03-04:
Having similar issues. Running on a RBPi2.

`/zigbee2mqtt-1.1.1/lib/extension/deviceReceive.js:71
                return c.cid === cid && c.type === message.type;
                         ^
TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/zigbee2mqtt-1.1.1/lib/extension/deviceReceive.js:71:26)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/zigbee2mqtt-1.1.1/lib/extension/deviceReceive.js:69:52)
    at extensions.filter.forEach (/zigbee2mqtt-1.1.1/lib/controller.js:125:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/zigbee2mqtt-1.1.1/lib/controller.js:125:14)
    at Zigbee.onMessage (/zigbee2mqtt-1.1.1/lib/zigbee.js:179:18)
    at ZShepherd.emit (events.js:182:13)
    at ZShepherd.<anonymous> (/zigbee2mqtt-1.1.1/node_modules/zigbee-shepherd/lib/shepherd.js:158:14)
    at ZShepherd.emit (events.js:182:13)
npm
 ERR! code ELIFECYCLE
npm ERR!
 errno 1
npm ERR! zigbee2mqtt@1.1.1 start: `node index.js`
npm 
ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
npm ERR!
 This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-04T09_51_50_323Z-debug.log
2019-03-04T10:51:54: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-03-04T10:51:54: PM2 log: App [npm:0] starting in -fork mode-
2019-03-04T10:51:54: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /zigbee2mqtt-1.1.1
> node index.js
  zigbee2mqtt:info 3/4/2019, 10:52:02 AM Logging to directory: '/share/zigbee2mqtt/log/2019-03-04.10-52-02'
  zigbee2mqtt:info 3/4/2019, 10:52:05 AM Starting zigbee2mqtt version 1.1.1 (commit #unknown)
  zigbee2mqtt:info 3/4/2019, 10:52:05 AM Starting zigbee-shepherd
  zigbee2mqtt:info 3/4/2019, 10:52:07 AM zigbee-shepherd started
  zigbee2mqtt:info 3/4/2019, 10:52:07 AM Coordinator firmware version: '20190109'
  zigbee2mqtt:info 3/4/2019, 10:52:07 AM Currently 6 devices are joined:`

Koenkk, 2019-03-04:
@Maximus48p can you do a `rm -rf node_modules && npm install`?

Maximus48p, 2019-03-04:
Found out that it has something to do with my own personal file 'devices.js'. 

Therefore my question; What is the best way to update new shepherd devices from site [https://github.com/Koenkk/zigbee-shepherd-converters](url) for a raspberry pi?
ssh in the root of the pi and execute the command `npm i zigbee-shepherd-converters`?

Maximus48p, 2019-03-04:
> @Maximus48p can you do a `rm -rf node_modules && npm install`?

Should i do that in the root of my raspberry?

Koenkk, 2019-03-04:
You need to do that in the directory where you installed zigbee2mqtt, if you are using hassio, remove the personal devices.js.

Maximus48p, 2019-03-04:
Hi Koenkk, i appreciate your help very much, but it is not totally clear to me. Prob me, so sorry for that.
I have done a clean hassio install on my raspberry pi according [https://www.home-assistant.io/hassio/installation/](url). 
added hassio add-on 'Mosquitto broker' and 'zigbee2mqtt'.
I removed my personal file 'devices.js' and it al seems working again. Accept for a new device i want to add; 'Heiman Smart Carbon Monoxide Sensor', 'HS1CA-E', zigbee device name: 'COSensor-EM'.

I have now also added hassio add-on 'SSH & Web Terminal' to be able to login with SSH. But when i try to execute both commands, i am getting the following error;
[image]

What am i doing wrong? 
I also like to update the zigbee-shepherd files, because i notices that new devices are submitted which i also own. I can't find the procedure on the internet.

Thanks for your help again.
Max.

Koenkk, 2019-03-04:
AFAIK, using a custom `devices.js` will only allow you to use devices specified in there.

What you should do here is:
- Make sure your HS1CA-E' works 
- Create a pull request, in this way the device can be officially supported by zigbee2mqtt
- Remove your custom devices.js and use hassio edge (to get the latest zigbee-shepherd-converters after I merged it)

It seems that you currently have a syntax error in your personal `devices.js`, can you post it here?

james-fry, 2019-05-14:
> Accept for a new device i want to add; 'Heiman Smart Carbon Monoxide Sensor', 'HS1CA-E', zigbee device name: 'COSensor-EM'.

Hi @Maximus48p 
Did you manage to get the Heiman HS1CA-E CO detector working?
I'd like to add this device, too.